Ysurac/openmptcprouter #3902: After enabling aggregation, the connection is disconnected without any signs

lhc1993, 2025-04-06:
## Expected Behavior

After enabling aggregation, the connection is disconnected without any signs
## Current Behavior

Apr  6 03:39:39 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 
Apr  6 03:39:39 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 Watch
Apr  6 03:39:52 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 
Apr  6 03:39:52 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 Watch
Apr  6 03:40:04 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 
Apr  6 03:40:04 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 Watch
Apr  6 03:40:08 OpenMPTCProuter daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 77.88.55.77)
Apr  6 03:40:10 OpenMPTCProuter daemon.info omr-tracker-v2ray: Reload V2Ray rules
Apr  6 03:40:13 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr  6 03:40:27 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 
Apr  6 03:40:27 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 Watch
Apr  6 03:41:01 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
Apr  6 03:41:21 OpenMPTCProuter daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 1.1.1.1, 74.82.42.42, 198.41.212.162)
Apr  6 03:41:26 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr  6 03:41:56 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 
Apr  6 03:41:56 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 Watch
Apr  6 03:42:01 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
Apr  6 03:42:08 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 
Apr  6 03:42:08 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 Watch
Apr  6 03:43:58 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 
Apr  6 03:43:58 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.9.236 f6:70:9b:eb:d3:d9 Watch
## Specifications

  - OpenMPTCProuter version: v0.62
  - OpenMPTCProuter VPS version: 0.1032
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2025-04-06:
I'm not sure to understand what you mean and would be better with more information like platform, proxy,...

lhc1993, 2025-04-06:
Before this line of log, there is no prompt, the aggregation proxy is down, The log is as follows
Apr 6 03:41:21 OpenMPTCProuter daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 1.1.1.1, 74.82.42.42, 198.41.212.162)

Ysurac, 2025-04-06:
It's only one time or more ? You can increase proxy check in OMR-Tracker Manager on router and check v2ray log on the vps ```journalctl -u v2ray```. Did you also try XRay ?