Ysurac/openmptcprouter #3907: Filesystem becomes Read-Only

sm4ck82, 2025-04-09:
Okay, here is a draft of an issue report in English based on the problems you described:
Subject: Filesystem becomes Read-Only and Issues with TP-Link USB Network Adapter on Raspberry Pi 5
Description:
I am experiencing issues with OpenMPTCProuter on a Raspberry Pi 5. I have tested two different versions of the software: the latest available version and version 0.61 (the first version to support the Raspberry Pi 5). In both versions, after some time of use (the exact timing is not consistent), the filesystem becomes read-only, preventing any further configuration changes.
Additionally, I encountered a specific problem with version 0.61 while trying to add a TP-Link USB network adapter.
Steps to Reproduce (Version 0.61):
 * Flash the OpenMPTCProuter version 0.61 image onto an SD card and boot the Raspberry Pi 5.
 * Perform basic configuration settings via the web interface. This was successful.
 * Successfully connect a mobile phone via USB tethering. This was recognized and worked correctly.
 * Connect a TP-Link USB network adapter to the Raspberry Pi 5.
 * Navigate to the "Status" page. The TP-Link adapter is listed but shows a MAC address of 00:00:00:00:00:00.
 * Observe that the main router on the network correctly recognizes the TP-Link adapter and assigns it an IP address.
 * Attempt to configure the TP-Link adapter under "Devices" and then "Network Interfaces" in the OpenMPTCProuter web interface.
 * Try to save the configuration. The system fails to save the changes.
 * Upon further investigation (e.g., via SSH), it is found that the /etc/config/network file is now read-only.
Expected Behavior (Version 0.61):
 * The TP-Link USB network adapter should be recognized with its correct MAC address in the OpenMPTCProuter status.
 * It should be possible to configure the TP-Link adapter as a network interface via the web interface and save the configuration successfully.
 * The filesystem should remain writable, allowing for configuration changes to be saved.
Observed Behavior:
 * In both the latest and version 0.61 of OpenMPTCProuter, the filesystem eventually becomes read-only.
 * In version 0.61, the TP-Link USB network adapter is shown with an incorrect MAC address (00:00:00:00:00:00).
 * Configuration changes for the TP-Link adapter could not be saved due to the /etc/config/network file becoming read-only.
Hardware:
 * Raspberry Pi 5
Software:
 * OpenMPTCProuter (latest version at the time of testing)
 * OpenMPTCProuter version 0.61
Additional Information:
 * I have tried multiple SD cards, so the issue does not seem to be related to a specific SD card.
 * The issue of the filesystem becoming read-only occurs in both tested software versions.
I hope this information is helpful in diagnosing these issues. Please let me know if you require any further details.
Note: You can copy and paste this text into a GitHub issue report or forum post. Remember to replace "(latest version at the time of testing)" with the actual version number you used.

sm4ck82, 2025-04-09:
On the last build for raspberry pi 5 i have instantly the read only file system on first boot after flash.
I even cant change my root password.

I tested both ext4 and the other, i dont know the name noe

mrkaffeine92, 2025-04-16:
For RPI, always use ext4.