Ysurac/openmptcprouter #3909: Issues Bonding Two WAN Connections – Only One WAN Active for Download (Upload Works Fine)

iWeeDev, 2025-04-11:
## Expected Behavior  
Both WAN connections (`wan_1` and `wan_2`) should be bonded and used simultaneously to increase overall bandwidth, as configured in OpenMPTCPRouter. Each WAN line offers approximately 200 Mbps of download speed.

## Current Behavior  
Only one of the WAN connections seems to be active. Even though both `wan_1` and `wan_2` are properly configured, the total bandwidth I get is consistent with just a single connection being used (around 200 Mbps). It seems that the upload speed is correctly bonded, but there is an issue with the download speed

This screenshot, showing the achieved speed, was taken with only the wan_1 Ethernet cable connected (the wan_2 cable was unplugged):

[image]

This one was taken with only the wan_2 Ethernet cable connected (the wan_1 cable was unplugged):

[image]

Finally, both wan_1 and wan_2 are connected:

[image]

It seems that the upload speed bonding is working correctly, but the download speed is only reflecting the speed of one of the WAN connections.

What could be happening?

## Specifications  
- **OpenMPTCProuter version:** openmptcprouter v0.62-6.6 r0+28431-92e020b50f  
- **OpenMPTCProuter VPS version:** Showed on screenshots  
- **OpenMPTCProuter VPS provider:** Amazon EC2  
- **OpenMPTCProuter platform:** Banana pi BPI-R4

Ysurac, 2025-04-11:
There is something strange with your WANs. Can you try a ```ping -B -I lan1 <vpsip>``` ?

iWeeDev, 2025-04-11:
> There is something strange with your WANs. Can you try a `ping -B -I lan1 <vpsip>` ?

I’d like to clarify that in my setup, both WAN connections are using SIM cards from the same provider (Vodafone), but on separate accounts. I have access to dozens of active SIMs with truly unlimited data. My goal is to deploy a device capable of bonding as many WAN interfaces as possible. From what I understand, OpenMPTCPRouter supports up to 10 WANs, but I’m open to testing beyond that limit if feasible.

Based on your comment I saw in other threads "If you use same SIMs, so same provider, the bandwidth is shared between modems, never a good idea." This makes me think the issue may not be related to hardware or OpenMPTCPRouter, but rather due to the provider applying some sort of bandwidth or limitation across both SIMs.

Do you think that could be the reason download bonding isn't working properly?
Is there any known way to bypass or work around this limitation?

Ysurac, 2025-04-11:
OpenMPTCProuter can support up to 8 WANs.
When using same provider it's often the issue. Bandwidth is shared between users/SIM card.
Can you also give me the result of the ping command ?

iWeeDev, 2025-04-11:
> Can you also give me the result of the ping command ?

I ran the command as you suggested and here are the results:

- **Ping to 8.8.8.8:**  
  I received replies without any packet loss:
  ```
  root@OpenMPTCProuter:~# ping -B -I lan1 8.8.8.8
  PING 8.8.8.8 (8.8.8.8) from 192.168.3.101 lan1: 56(84) bytes of data.
  64 bytes from 8.8.8.8: icmp_seq=1 ttl=113 time=43.4 ms
  64 bytes from 8.8.8.8: icmp_seq=2 ttl=113 time=49.9 ms
  64 bytes from 8.8.8.8: icmp_seq=3 ttl=113 time=52.3 ms
  64 bytes from 8.8.8.8: icmp_seq=4 ttl=113 time=47.9 ms
  64 bytes from 8.8.8.8: icmp_seq=5 ttl=113 time=47.9 ms
  64 bytes from 8.8.8.8: icmp_seq=6 ttl=113 time=46.9 ms
  --- 8.8.8.8 ping statistics ---
  6 packets transmitted, 6 received, 0% packet loss, time 5006ms
  ```
- **Ping to my VPS (3.10.xxx.xxx):**  
  No replies were received, resulting in 100% packet loss:
  ```
  root@OpenMPTCProuter:~# ping -B -I lan1 3.10.xxx.xxx
  PING 3.10.xxx.xxx (3.10.xxx.xxx) from 192.168.3.101 lan1: 56(84) bytes of data.
  --- 3.10.xxx.xxx ping statistics ---
  24 packets transmitted, 0 received, 100% packet loss, time 23915ms
  ```

It appears that while the WAN connection via `lan1` can successfully reach external servers like 8.8.8.8, it is unable to reach my VPS at 3.10.xxx.xxx. Could this be due to ICMP being blocked or rate-limited on the VPS (hosted on Amazon EC2), or is it possibly related to the WAN bonding configuration?

Any insights on this would be greatly appreciated.

iWeeDev, 2025-04-11:
I've fixed the ping issue from the VPS. I'm now getting a stable response via the `lan1` interface:

```bash
root@OpenMPTCProuter:~# ping -B -I lan1 3.10.x.x
PING 3.10.x.x (3.10.x.x) from 192.168.3.101 lan1: 56(84) bytes of data.
64 bytes from 3.10.x.x: icmp_seq=1 ttl=53 time=42.7 ms  
64 bytes from 3.10.x.x: icmp_seq=2 ttl=53 time=50.9 ms  
64 bytes from 3.10.x.x: icmp_seq=3 ttl=53 time=51.7 ms  
64 bytes from 3.10.x.x: icmp_seq=4 ttl=53 time=50.0 ms  
64 bytes from 3.10.x.x: icmp_seq=5 ttl=53 time=45.9 ms  
64 bytes from 3.10.x.x: icmp_seq=6 ttl=53 time=47.9 ms  
64 bytes from 3.10.x.x: icmp_seq=7 ttl=53 time=47.0 ms  
^C
--- 3.10.x.x ping statistics ---
7 packets transmitted, 7 received, 0% packet loss, time 6007ms  
rtt min/avg/max/mdev = 42.704/48.029/51.734/2.917 ms
```

[image]

Bonding still doesn't work for downloads, only for uploads.

Ysurac, 2025-04-11:
To check if aggregation is working, via SSH on the router run ```omr-test-speed``` and at the same time look at Network->MPTCP, Bandwidth tab

iWeeDev, 2025-04-11:
> To check if aggregation is working, via SSH on the router run `omr-test-speed` and at the same time look at Network->MPTCP, Bandwidth tab

After running the test you suggested (`omr-test-speed` while monitoring the *Network → MPTCP → Bandwidth* tab), I noticed that the bandwidth usage between `wan_1` and `wan_2` is highly unbalanced. Specifically:

- `wan_1` peaked at **~185 Mbps**
- `wan_2` peaked at **~35 Mbps**
- Total aggregated throughput reached **~200 Mbps**

This is roughly the maximum expected throughput of either individual WAN connection, which suggests that bonding is not effectively aggregating both links, or that one link is being significantly underutilized.

[image]

---

To further investigate whether my 4G/5G ISP is performing traffic shaping or bandwidth limiting when multiple simultaneous connections are made to the same destination, I’m planning an external test:

### Test Plan:

1. **Host a large file on my VPS (Debian-based):**
   I’ll create a 1–2 GB file filled with zeros on my VPS and serve it via HTTP.

2. **Connect two separate PCs** each using one of the SIMs via 4G/5G (i.e., `wan_1` and `wan_2`).

3. **Simultaneously download** the same file from both PCs.

4. **Compare the throughput** of each download individually, and observe whether the speeds are significantly reduced when both SIMs are downloading from the same source.

This should help determine whether the provider is throttling based on destination IP or if the issue lies in OMR’s ability to properly perform bonding over mobile networks.

Ysurac, 2025-04-11:
This suggest both links are aggregated. And you have a limit somewhere (maybe provider).
To test both wans you can open 2 SSH terminal to the router and run ```omr-test-speed wan_1``` and ```omr-test-speed wan_2``` at the same time (You can stop after 2 minutes with Ctrl+C). And then compare results.

closed 2025-07-15 by github-actions[bot]