Ysurac/openmptcprouter #3910: Speed is too low after aggregation

sanfweb, 2025-04-11:
**Expected Behavior**
Three WAN connections (WAN1, WAN2, WAN3) should be bonded and used simultaneously to increase the overall bandwidth, as configured in OpenMPTCPRouter. Each WAN line provides approximately 100 Mbps of download speed. I have 3 modems from 3 different providers.

**Current Behavior**
All modems are successfully connected, and the status shows everything is fine. However, when I run a speed test using speedtest, the result is very low.

[image]

[image]
[image]

To verify, I opened 3 separate SSH sessions and ran omr-test-speed wanXX in each one. After that, I checked the MPTCP -> Bandwidth page, and there the results looked as expected, showing the proper speeds for each connection.

[image]

I also tested the aggregation using a VPS from a different provider, and the results were the same. So I believe the issue might be related to some configuration settings. I would really appreciate any help or suggestions on what to look into.

Thank you in advance.

**Specifications**
  - OpenMPTCProuter version: v0.61-6.6 r0+27346-c7ba5574f5
  - OpenMPTCProuter VPS version: v0.1032 6.12.15-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B

sanfweb, 2025-04-11:
This gateway status for mobile providers is a common occurrence and should not affect the aggregation

Running the omr-test-speed command on the server:

[image]

Running the omr-test-speed command without interfaces on the device:

[image]

sanfweb, 2025-04-11:
I have updated to version **v0.62-6.6 r0+28431-92e020b50f**

I checked the speed after updating the version, all modems are active and connected.

[image]

All speed tests do not exceed 100 Mbps, regardless of which server is used on speedtest. It feels like there is a threshold value set somewhere, beyond which the device does not go.

Running the omr-test-speed command on the server:

[image]

Running the omr-test-speed command on the device:

[image]

Ysurac, 2025-04-11:
The graph and the omr-test-speed command give higher speed, 23.4*8=187Mb/s. Can you test with another speedtest like https://www.nperf.com ?

sanfweb, 2025-04-12:
Unfortunately, I had some issues with one of the modems. As soon as I fix the problem, I will take the measurement right away.

sanfweb, 2025-04-15:
I managed to restore the device's operation, and I've taken the measurements again

[image]

**Running the omr-test-speed command without interfaces on the device:**

[image]

**Tests were done using Speedtest and nPerf services:**

[image]

[image]

And once again, I couldn't get past the 100 Mbps threshold, so I think there might be some kind of limitation in the settings (

Ysurac, 2025-04-15:
From where are you running speedtest ? As results from the box are ok. Did you check switch, port configuration (negociation,...),... ? Seems like something is limiting to 100Mb/s outside of the router.

sanfweb, 2025-04-15:
I swapped the cable, and that solved the issue — looks like the first one was faulty. Thanks a lot for your help! You have a great product, and I really appreciate your support with troubleshooting

[image]

closed 2025-04-15 by sanfweb

Ysurac, 2025-04-15:
Thanks for the feedback.