Ysurac/openmptcprouter #3912: OMR bypass behind UDM Pro

Nicras, 2025-04-12:
Hello,

I'm using OMR behind a UDM Pro. 
My question is, would it be possible to use OMR bypass based on client IP?
What is required for bypass to work?

Thank you

Ysurac, 2025-04-12:
Yes and for IP only require to set it in OMR-ByPass, for domains you need to use OMR IP as DNS.

Nicras, 2025-04-12:
I'm sorry but it didn't work. Neither IP, Local LAN IP nor MAC Address works. UDM does masquerading to the WAN connection. I guess that is why OMR doesn't recognize.

Ysurac, 2025-04-13:
Yes in this case it's not possible as OMR only know UDM IP.

darkman1983, 2025-04-13:
If you're using Network 9.1.x (9.1.118) on the UDM Pro, then you can deactivate NAT for specific networks, but then don't forget to add a static route on OMR for your UDM, so portforwarding works correctly from OMR <-> UDM Pro.

[image]

Nicras, 2025-04-13:
Good to know, thank you. I'll try but this way I have to reconfigure a few things.

darkman1983, 2025-04-14:
You only have to add two firewall rules for OMR on UDM, if you wanna ping from OMR to clients in UDM Subnet.
I'm still forwarding all ports from OMR to UDM and open ports on UDM as needed, all others get blocked.
I just switched over and deactivated the NAT for Default network on UDM, works as expected, no more double NATing.
My OMR is on another subnet at UDM, the only ceavat is: MAC/ARP (L2) isn't visible, but IP's are. So you can still configure DSCP, Firewall, etc. on IP-Level.
Network feels faster now 😄

darkman1983, 2025-04-14:
Ah, i've looked at UDM cmd, seems that exclude isn't working, NAT still active for that network. Searching for Workarounds. Maybe you need an on_boot script, to deactivate NAT for this interface. I wonder why unifi brings out UI improvements, when they're not workin ^^

darkman1983, 2025-04-14:
Ok, there's defnitly a bug, which prevents those excludes to work.
Maybe you have to use something like: Example "ipset add UBIOS4ALL_ADDRv4_eth8 192.168.1.0/24" (replace interface, network and IPSET with your own values)
But as this would get overwritten on provisioning, reboot, firmware update...
You would also have to use the on_boot script or something else.
Could also be a script on another machine which checks over ssh, if NAT is disabled and otherwise disable it.

closed 2025-07-18 by github-actions[bot]

ioogithub, 2025-10-23:
[darkman1983](https://github.com/darkman1983) and [Nicras](https://github.com/Nicras)  were you able to get omr-bypass working with any of the latest shapshots or any other versions? If so could you please advise on which version you got working? I have OMR-bypass working as expected in version 0.61rc2-6.6 however no version past this are working. I have tested all of the fresh images between 0.61 up to the latest 0.64-snapshot-6.12 r0+31388-17badf6099 without success.

OMr-bypass will temporarily work on all of these versions for a short time only after the Save & Apply button is clicked on the wizard page but after a reboot or after some time a script or some other action runs and breaks it again. Are you aware of which scripts are likely involved in setting these routes? I am willing to look into the issue more deeply but I don't really know where to start.