Ysurac/openmptcprouter #3914: Cant get public IP address from Xray, Cant contact server admin script (No result)

orakult, 2025-04-15:
I use OpenMPTСPRouter in a virtual machine. I transferred the OVA image to another hardware, after which I noticed that port forwarding stopped working. What was entered on the first hardware (80, 443, etc.) works, but the new rules do not work. In the status section I now see this picture.

[image]

Channel aggregation works perfectly.

## Specifications

  - OpenMPTCProuter version: 6.12.15-x64v3-xanmod1 0.1032
  - OpenMPTCProuter VPS version: 0.1032 6.12.15-x64v3xanmod1
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86-64

orakult, 2025-04-16:
Yes, the previous PC is no longer working. The OVA image has been transferred to another PC, where the problem is observed.
The high load is caused by the aggregation of 8 wireguard channels.
How can I fix this problem? Thank you!
P.S. Maybe there are ways to forward ports (XRay) in the VPS settings?

Ysurac, 2025-04-16:
Try to check in System->OpenMPTCProuter, "Advanced settings" the "Force retrieve settings" box and save.
Else check Status->System log.

orakult, 2025-04-16:
```
> Wed Apr 16 16:42:15 2025 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on *.*.*.*, no server API answer on *.*.*.*)
> Wed Apr 16 16:42:16 2025 daemon.err omr-tracker-xray[24094]: /bin/omr-tracker-xray: line 105: curl: not found
> Wed Apr 16 16:42:16 2025 daemon.err omr-tracker-xray[24094]: /bin/omr-tracker-xray: line 63: curl: not found
> Wed Apr 16 16:42:16 2025 daemon.err omr-tracker-xray[24094]: /bin/omr-tracker-xray: line 70: curl: not found
```

This message is repeated endlessly.

 

> System->OpenMPTCProuter, "Advanced settings" the "Force retrieve settings"

This box doesn't seem to work.
If I delete the VPS server from "Server settings" and then add it again, "Proxy settings" are not filled.
"Server key" and "server username" are specified correctly. I copy from "openmptcprouter_config.txt".

orakult, 2025-04-16:
> journalctl -u omr-admin

```
Apr 16 20:05:57 "ServerName" omr-admin.py[1899]: INFO:     127.0.0.1:28845 - "GET / HTTP/1.1" 200 OK
Apr 16 20:06:14 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:54883 - "GET / HTTP/1.1" 200 OK
Apr 16 20:06:30 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:25683 - "GET / HTTP/1.1" 200 OK
Apr 16 20:06:47 "ServerName" omr-admin.py[1899]: INFO:     127.0.0.1:56293 - "GET / HTTP/1.1" 200 OK
Apr 16 20:07:03 "ServerName" omr-admin.py[1899]: INFO:     127.0.0.1:59349 - "GET / HTTP/1.1" 200 OK
Apr 16 20:07:20 "ServerName" omr-admin.py[1900]: INFO:     127.0.0.1:49047 - "GET / HTTP/1.1" 200 OK
Apr 16 20:07:36 "ServerName" omr-admin.py[1899]: INFO:     127.0.0.1:48875 - "GET / HTTP/1.1" 200 OK
Apr 16 20:07:52 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:59809 - "GET / HTTP/1.1" 200 OK
Apr 16 20:08:09 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:26499 - "GET / HTTP/1.1" 200 OK
Apr 16 20:08:25 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:53573 - "GET / HTTP/1.1" 200 OK
Apr 16 20:08:42 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:62501 - "GET / HTTP/1.1" 200 OK
Apr 16 20:08:59 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:42425 - "GET / HTTP/1.1" 200 OK
Apr 16 20:09:17 "ServerName" omr-admin.py[1900]: INFO:     127.0.0.1:62083 - "GET / HTTP/1.1" 200 OK
Apr 16 20:09:34 "ServerName" omr-admin.py[1899]: INFO:     127.0.0.1:61311 - "GET / HTTP/1.1" 200 OK
Apr 16 20:09:51 "ServerName" omr-admin.py[1900]: INFO:     127.0.0.1:52787 - "GET / HTTP/1.1" 200 OK
Apr 16 20:10:08 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:29593 - "GET / HTTP/1.1" 200 OK
Apr 16 20:10:25 "ServerName" omr-admin.py[1900]: INFO:     127.0.0.1:36871 - "GET / HTTP/1.1" 200 OK
Apr 16 20:10:42 "ServerName" omr-admin.py[1899]: INFO:     127.0.0.1:23141 - "GET / HTTP/1.1" 200 OK
Apr 16 20:11:00 "ServerName" omr-admin.py[1899]: INFO:     127.0.0.1:17207 - "GET / HTTP/1.1" 200 OK
Apr 16 20:11:17 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:20749 - "GET / HTTP/1.1" 200 OK
Apr 16 20:11:34 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:21985 - "GET / HTTP/1.1" 200 OK
Apr 16 20:11:52 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:30841 - "GET / HTTP/1.1" 200 OK
Apr 16 20:12:09 "ServerName" omr-admin.py[1900]: INFO:     127.0.0.1:42553 - "GET / HTTP/1.1" 200 OK
Apr 16 20:12:26 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:60095 - "GET / HTTP/1.1" 200 OK
Apr 16 20:12:42 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:17535 - "GET / HTTP/1.1" 200 OK
Apr 16 20:12:58 "ServerName" omr-admin.py[1899]: INFO:     127.0.0.1:49205 - "GET / HTTP/1.1" 200 OK
Apr 16 20:13:15 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:19843 - "GET / HTTP/1.1" 200 OK
Apr 16 20:13:32 "ServerName" omr-admin.py[1902]: INFO:     127.0.0.1:61401 - "GET / HTTP/1.1" 200 OK
Apr 16 20:13:48 "ServerName" omr-admin.py[1899]: INFO:     127.0.0.1:48967 - "GET / HTTP/1.1" 200 OK
lines 1-29
```

> you have curl command
`-ash: curl: not found`

I encountered another problem.
If I ping 8.8.8.8, I get a response of 70-80ms (This is normal).
But when I do "opkg update" I get an error:

```
Collected errors:
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-6.6/x86_64/luci/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-6.6/x86_64/packages/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-6.6/x86_64/base/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-6.6/x86_64/routing/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-6.6/x86_64/telephony/Packages.gz, wget returned 255.
 * opkg_download: Failed to download http://openwrt.132lan.ru/packages/21.02/packages/x86_64/modemfeed/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-6.6/x86_64/targets/x86/64/packages/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-6.6/x86_64/packages/x86_64/base/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-6.6/x86_64/packages/x86_64/luci/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-6.6/x86_64/packages/x86_64/openmptcprouter/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-6.6/x86_64/packages/x86_64/packages/Packages.gz, wget returned 255.
```

orakult, 2025-04-16:
No, I took it from here:
https://www.openmptcprouter.com/download

This is very strange, and I noticed only now that "opkg" does not work. Before the image migration, I definitely installed "pptp-client" from opkg.

Ysurac, 2025-04-16:
You really should have curl and wget. Try to install it again.