Ysurac/openmptcprouter #3915: See if DSCP is working

darkman1983, 2025-04-15:
I've wondered if DSCP is actually working.
So i've used tcpdump to see, but that wasn't enough in my eyes, i needed a sneaky minimal GUI to see :-D

Setup:
SQM enabled in OMR for VPN + WAN -> DSCP FLAG's should be set. Is it required for VPN? Dunno :-P

So, DSCP is sent through VPN, which ones and do they arrive on VPS? Do they go out? Let's see...
OMR:
[image]

VPS (Interface with public IP):
[image]

I've also tested with WAN of OMR, but on that Interface (SQM enabled, DSCP should be set) only CS0 (Best Effort) goes out. Reading through internet revelad ISP is overriding or removing DSCP anyways.
But as DSCP is for specific ports most time and OMR sends portforwardings through VPN anyways, you can test.
The ISP is still a thing, because VPN traffic can't be prioritized.

As you can see, DSCP works, but there is a difference of what get's IN/OUT on OMR and VPS.

### The Script (pure bash)
Requirement: tcpdump
Parameters: tun0 default or give interface name on script start
Setup: Install tcpdump if not installed. Save with desired name, eg. tos_counter.sh, chmod with eg. "ug+x", execute with eg. "./tos_counter".

[NoPaste of Script](https://nopaste.boris.sh/?l=shll#XQAAAQCDFAAAAAAAAAARiEJGPfQYaqZnr3qfb2ln/3yapbz1Du582OOTaq+NQCT+LB0WrRk9WgxWZccF0TvF7Vzb6YlvwAYw6J3A0XLsKZKtZRjQ/Y76MhsRbRoZvMyAow1ngV4PEs0H0EpTm6Lb3YlSnWe3cxWwZU8fo9v8SjgdJCFdxKZOewR87FEzSU+3/8SrrnWlrk7PrcMMXY5i5kMCvorTkqeFQvZCQXNgju1dF6USWGiLvryJ6Bne3vKgM609E6rcKNFjvRpqeu2BA+R4zAAmconJxGjWwoMib0QER2u3udR5nNUHonoqj2n+fQeHK4fRvC1n/74RwV+Zf3JDZOgF7cFZj3I1sSe0uq6OfVjN2E12k8chdPolDIyVO/rNCSQiTF8Jx9EUsyKHvrJPOF+CRMVRgufG21Pg6bN2sEy9sekb33RM8hqw1f/mvCST9SVOAFVMcF39tOvypmwZ0EKZ7V6u3zq0xGoevJt5m6Y+bTN9SLXtPeU8CGPe4DL8zcCfRBaZGjdPVu4I+F4eMNlxd7YP31gHzmt9sqPjPRhDchWSiP2Q4QsCoEnhhrMx4W99ltX2f7ipTSsM2IbC1FZc4uCAkh6piCMp9v+35KT0tkfE7o9B8T79PSy1E+wQXVSR4UqTO99IQUDPKqOMuTMm6o7v1HDRE4ua8tzEps0ojjWe0xlwo99kFLsOajQxFmUd9XSTHSwp4hHQr8+H880zQPSBrS3wE2Aw+uS6ZqqUnN1tj1Nsd5gjlhy2LHq7pBMun1P0zreIw6h4XkWz29l3XGTqhdd1Ko/wKx0T15weNzPX/TOJyETw3mlzZyk2G/2JXJc92GCq0GepH3lAQu8bbF6aLun1azxusAIhnvb6JksTmmcmQSMu1qFWSNe6lGkQxqaDqbp3C6Rs0NNmaH7Xx8rWC2pnQQwxm6rRqweg8HdsJ/mrCQP3FfPcDEZlRhGrhTzNziHBOkZkdP5iIGprI5D6es2y26WYSmLO04VtHrHdIhgTcvwKt5EYgeAro346pEO/b08dBMKDkuo3Qve9VTVI9/JOeoyUkTZPhcYwLJLEoSxBlqdAbFZDjlbHg74xOb+2u/nb8d3g1MmcH7WRetkvMvWNWej7nHGXBM7R6e7Y944AtXtD3ApBEfkKLy6lat1w7fEYqdL8s30K+zob+5XtV1Gw2SvWsyG+k3yDNr9XVb2uer2QW2xbrX+J+w3M2mQKCvbKmXds1RvoH+Jye+6UkJLNm5Kv6JuAD02i4/2pnQGr0OhyIJHT8CN/334kK/4Eiyt4BhrmfncnFk+Wkyg7r5QsZexofddCnTql1AobC3KJivTTmZVnymEbF+nLx1oDtftA6rTYP9DXhAHvMaKYV9iOIr+FMLliMRxC0GPo9KJ+RG8LjxX7gPI2cwDtHUrMSAMNLA4aRTjOkGfDhbQHgNZanA0+0sPNTIpf85Tz5yB+DEmOHC/pn9KBwf/SbyX5w4g8zK0l50csuxl/ebQ3k8JevH0B/EAp5jeCPmBnZfMMDjL0YWYdm9okgag9A/f7n0AuYQhEeKyPKpDMpzL3gFzrGOPGpu2Eh875SDpcj1xxPsJpUiZVVZwIVN/Eq92o4nqsifX7VNArnbtnH/PqSsoaZ3SpfxGVJ0SOUW8ND1YJ0s4u8+5/NLV6y2Fsro4LdkSM3B3uka/Dlx5f2bz1WBx3Y+u0Qx+i1VnCVhpXkxrGkuOCmU16syCJlXNlzf3R1qJzMIwrPgTRyNegpZjQ+ILb94Jn3AXH2yWn91PY+H49AMrblIYMmDS5wUWHGYVdhOErbNAiUX4Es7QX+Oeh5bTF4095n2YyCFw8D/fGBJD5dkv7PWJ26hH7qGAspzuqOFZjIgpNgzLzIjSuLAydvPIUXbFgRC8zor41J5u9sNDWGQ0MMQtizi2TcZ540rAoOYHGq8BWEhfvyVwHyP8Q8MaEPDFo8qzZVG0V9BGkn3MnJ/2edbAFyUEaOizcy6oPKQohoAsuwB37ODBoJ5k+xX9N+uffGAi8gAa3AXAOlx1PoxYdRogg+Qon31QnnrttHKqTcUnQeQvsMIoe4nvkJV2VwT/dpmSzQVnO/YngEu5snCYpna/hALTB5n+v3LYdWIl9xGXumjT4GtsG4/2OH7DJqBc6dOA8XEb+9Qp/QU/7htHbA6wihoWnpz/mHqOr0G2V2FyC4wjiI/lArsWdyLR1Df8WlU8SklbDdMWlCOE+93+i3Rl7Sgkat0OZUPbufiiKPa4tFsxCq+F4AAsvE/E+vaj6V61xqz3V5IEgmoQzpmuW7TheRLrPnPPPSmyQUrOZV7oNwSH4wtV+lfjWZ4MNZzbYNVvKHsFJiC7SUg2Waf+62e+DxE5qUaktRLpsHvHdY30v78unY+7Jdte9ocBNxJJ6E3lFTJ6cz2j+KotTudo1hAR8cNEm3j1sDn9608fQkCwRiBv45PBihJphk9p9x/tNxu0CK8Krk2EIVbbcfafyo4TytSaxoCyDu//ym96H)

darkman1983, 2025-04-24:
Script updated

Ysurac, 2025-04-24:
DSCP give strange results on OMR, not exactly sure why.
VPS should always be higher than OMR as you are testing only tun0 on OMR, so only the VPN (it's not exactly the case, maybe some loss...). It's also possible that VPS output remove most of the DSCP tags.

darkman1983, 2025-04-28:
@Ysurac You marked the other one as duplicate, but the other one describes not working DSCP at all. On this one DSCP is in fact working, there're marks on packet. On the other bug all packets have flag 0x0, so no DSCP marking at all!

Ysurac, 2025-04-28:
Both are about DSCP issues and seems related.

darkman1983, 2025-04-28:
Then i will go on in this one!

We have been investigating the correct way to apply DSCP marking for specific services (e.g., Plex streaming) on OpenMPTCPRouter.

After extensive testing, we found that only manually inserted rules without counter work reliably for setting the DSCP field in the mangle chains.

Specifically, the following two commands work perfectly:
```
nft add rule inet fw4 mangle_forward iifname "eth0" tcp dport 40000 ip dscp set cs1
nft add rule inet fw4 mangle_forward oifname "tun0" tcp sport 40000 ip dscp set cs1
```
Important observations:

If the rules are added via the Web GUI (both using the regular firewall rule editor and the special DSCP configuration section), OpenMPTCPRouter always automatically adds a counter to the generated nftables rules.

The presence of a counter breaks DSCP marking in practice. Even though the rules look correct in the nftables ruleset, the packets are no longer properly marked.

Only when the rules are added manually without any counter present, DSCP marking works as expected.

We verified this by comparing:

Manually added rules via command line → working DSCP marking.

GUI-created rules → rules contain counter, DSCP marking ineffective.

This seems to be a general problem for DSCP modifications via GUI on OpenMPTCPRouter, not only for Plex, but for any traffic that relies on precise DSCP setting.

Suggestion:
It would be helpful if there was an option in the Web GUI to disable counter insertion for specific DSCP rules, or to allow a "raw" rule definition without forced counter.

Thanks a lot for your great work – and hopefully this feedback helps improve DSCP handling in future versions!

darkman1983, 2025-05-01:
Will you investigate into this, @Ysurac ? Because this can render DSCP completely unusable...