Ysurac/openmptcprouter #3918: Open menu OMR-Bypass

flimo44, 2025-04-19:
## Current Behavior

Hello,
Since last update impossible to open OMR-Bypass.
I have a error box  blank  ( with no error ) . If i clear the box nothing do.

[image]

## Specifications

  - OpenMPTCProuter version: 6.12.15-x64v3-xanmod1 0.1032
  - OpenMPTCProuter VPS version: v0.62-6.6 r0+28431-92e020b5
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-04-19:
Can you try latest snapshot https://snapshots.openmptcprouter.com/6.6/x86_64/targets/x86/64/ ?

flimo44, 2025-04-19:
Update done same problème. But now box RED.

[image]

flimo44, 2025-04-19:
2 others ^problème find : 
1 : Service vnstat crash :

[image]

[image]

2 : snmpd menu not present. Resolve temporally by install it in software .

Ysurac, 2025-04-19:
It's a VM or a dedicated computer ?
6.12 is only a snapshot, not supported at all, it's only a test.
luci-snmpd is disabled on snapshots for now.

flimo44, 2025-04-19:
It"s a VM

Ysurac, 2025-04-21:
It seems to be errors on VM host and maybe disk. There is CPU errors and disk errors. Check host CPU&Memory usages, if it's ok, check VM usage.

flimo44, 2025-04-21:
hello,
My VM run on UNRAID server i will réinstall it with a new. ( quand j'aurai le temps )
So, i try on a other VM on my synology server. Vnstat run fine but i have always the same OMR-Bypass error :/

Ysurac, 2025-04-23:
Can you check in your browser developer tools what error you can have on OMR-ByPass ?

flimo44, 2025-04-23:
[image]

flimo44, 2025-04-23:
I think i find the problem for acces OMR-bypass
I have 2 LAN  and if i connect  router  from LAN1 it's OK but not from LAN2. I will investigate why was ok before with same configuration.
LAN1 :   192.168.100.x   
LAN2 :   192.168.0.x

ioogithub, 2025-10-23:
[flimo44](https://github.com/flimo44) were you able to get omr-bypass working with any of the latest shapshots or any other versions? If so could you please advise on which version you got working? I have OMR-bypass working as expected in version 0.61rc2-6.6 however no version past this are working. I have tested all of the fresh images between 0.61 up to the latest 0.64-snapshot-6.12 r0+31388-17badf6099 without success.

OMr-bypass will temporarily work on all of these versions for a short time only after the Save & Apply button is clicked on the wizard page but after a reboot or after some time a script runs and breaks it again. Are you aware of which scripts are likely involved in setting these routes? I am willing to look into the issue more deeply but I don't really know where to start.