Ysurac/openmptcprouter #3922: Cant access to puplic ip shadowsocks rust

nochschnellerantworten, 2025-04-22:
[image]

I cant get puplic Ip Address.
 Iam using latest vps script and openmptcprouter-v0.62rc1-6.6-r0+28431-92e020b50f-mediatek-filogic-z8109ax-128m-initramfs-kernel.

thank you very much Ysurac

nochschnellerantworten, 2025-04-22:
cant download the snapshot its coming 404 fail when i click on the .bin file.
Iam using right now v0.62rc1-6.6.
maybe its because Iam using [HC-G80](https://www.huastlink.com/hc-g80/) as router

Ysurac, 2025-04-23:
I don't know where you click: it's not a supported router.

nochschnellerantworten, 2025-04-24:
then ill buy a Ubiquiti Edgerouter X or do you have recommandation? preferably a router with 8 possible internet accesses

Ysurac, 2025-04-24:
Not a Edgerouter X, really too slow for anything.
Then it depends the aggregated speed you want.

nochschnellerantworten, 2025-05-08:
around 1 Gigabit.

I have very bad connection, ill need at least 5 wan ports for 1 gigabit. 
Can i build by my own?

Ysurac, 2025-05-09:
For 1Gb/s, I would use something based on x86_64 or a RPI5 (I can reach 950Mb/S due to the 1Gb/s port).
What connections will be used as WAN ?

nochschnellerantworten, 2025-05-09:
Mostly mobil and 1 starlink

nochschnellerantworten, 2025-05-09:
I have one 520ngl with waveshare will it work? or what do you recommend for multi mobil modems?

Ysurac, 2025-05-09:
For the mobile part, make sure you have different provider, else it will only share available bandwidth.
I'm using some Quectel RM520N-GL modems but for the waveshare router it's not supported to install OpenMPTCProuter on it. I don't know any platform that support up to 1Gb/s and have multi modems support.

nochschnellerantworten, 2025-05-09:
There is only Telekom available. 
what if i have 3 sim card routers and connect them via wan?

Ysurac, 2025-05-09:
You can but if SIMs card are on same provider you are sharing the "antenna" bandwidth between your cards and other customers. So in many case you will not see anything better when you aggregate them.

nochschnellerantworten, 2025-05-10:
I see. so what u recomment? 
I think i will try it, can i add on RPI5 more eth Ports?

Ysurac, 2025-05-10:
Via USB and with a powered USB hub or maybe some extension card..

nochschnellerantworten, 2025-05-10:
so what router u recommend? If i want to have 4 eth ports?

Ysurac, 2025-05-10:
With 1Gb/s I would try some x86_64 box with AES-NI support on CPU that give better result than RPI5 on benchmark tests (it's not always the case)

nochschnellerantworten, 2025-07-25:
[image]  i have now RPI5 same problem

nochschnellerantworten, 2025-07-25:
alpha.config=alpha
alpha.config.nav_01='/cgi-bin/luci/admin/status/overview'
alpha.config.nav_02='/cgi-bin/luci/admin/services/ttyd'
alpha.config.nav_03='/cgi-bin/luci/admin/network/network'
alpha.config.blur='20'
alpha.config.color='#2222359a'
ddns.global=ddns
ddns.global.ddns_dateformat='%F %R'
ddns.global.ddns_loglines='250'
ddns.global.upd_privateip='0'
ddns.myddns_ipv4=service
ddns.myddns_ipv4.service_name='dyndns.org'
ddns.myddns_ipv4.lookup_host='yourhost.example.com'
ddns.myddns_ipv4.domain='yourhost.example.com'
ddns.myddns_ipv4.username='your_username'
ddns.myddns_ipv4.password='your_pasxxxxxx'
ddns.myddns_ipv4.interface='wan'
ddns.myddns_ipv4.ip_source='network'
ddns.myddns_ipv4.ip_network='wan'
ddns.myddns_ipv6=service
ddns.myddns_ipv6.update_url='http://[USERNAME]:[PASSWORD]@your.provider.net/nic/update?hostname=[DOMAIN]&myip=[IP]'
ddns.myddns_ipv6.lookup_host='yourhost.example.com'
ddns.myddns_ipv6.domain='yourhost.example.com'
ddns.myddns_ipv6.username='your_username'
ddns.myddns_ipv6.password='your_pasxxxxxx'
ddns.myddns_ipv6.use_ipv6='1'
ddns.myddns_ipv6.interface='wan6'
ddns.myddns_ipv6.ip_source='network'
ddns.myddns_ipv6.ip_network='wan6'
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='0'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='0'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].cachesize='1000'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].filter_aaaa='0'
dhcp.@dnsmasq[0].filter_a='0'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].dnsforwardmax='1500'
dhcp.@dnsmasq[0].quietdhcp='1'
dhcp.@dnsmasq[0].interface='loopback' 'lan' 'omrvpn'
dhcp.@dnsmasq[0].notinterface='wan1' 'wan2'
dhcp.@dnsmasq[0].server='127.0.0.1#5355'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv4='server'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
[... 2741 lines cut ...]
xray.omrtest.port='1111'
xray.omrtest.protocol='socks'
xray.omrtest.listen='127.0.0.1'
xray.omrtest.s_socks_auth='noauth'
xray.omrtest.s_socks_udp='1'
xray.omrtest.s_socks_ip='127.0.0.1'
xray.omrtest.s_socks_userlevel='0'
xray.main_reverse=reverse
xray.main_reverse.enabled='1'
xray.main_reverse.bridges='omrbridge|omr.lan'
xray.omrexit=outbound
xray.omrexit.protocol='freedom'
xray.omrexit.tag='out'
xray.omrrouting=routing_rule
xray.omrrouting.type='field'
xray.omrrouting.inbound_tag='omrbridge'
xray.omrrouting.outbound_tag='omrout_tunnel'
xray.omrrouting.domain='full:omr.lan'
xray.omrroutingo=routing_rule
xray.omrroutingo.type='field'
xray.omrroutingo.inbound_tag='omrbridge'
xray.omrroutingo.outbound_tag='out'
xray.main_routing=routing
xray.main_routing.enabled='1'
xray.main_routing.rules='omrrouting' 'omrroutingo'

nochschnellerantworten, 2025-07-25:
i also restarted the server

Ysurac, 2025-07-30:
So it's on a RPI5 ? You have one interface, it's not possible to use DHCP for WAN here (at least without some config on DHCP server). It also doesn't seems to use MacVLAN, you changed something on default configuration ?

nochschnellerantworten, 2025-08-01:
I am using 2 usb to lan. I have now reset all settings. I just entered server ip and key. And under wan1 macvlan I entered there the ip address and gateway but still no connection

Ysurac, 2025-08-01:
USB device are often not a good idea. You may need a USB powered hub.
Why are you using USB devices ?

closed 2025-11-04 by github-actions[bot]