Ysurac/openmptcprouter #3923: ZBT Z8102AX-M2-A-T 5G Modem Configuration

chubx, 2025-04-25:
## Expected Behavior
In Network > Interfaces > General Settings for the 5G Modem > Authentication Type > Set to NONE, and save and apply > Authentication Type Should be set to NONE, but It reverts back to `PAP/CHAP`

In Network > Interfaces > General Settings for the 5G Modem > Authentication Type > Set to NONE, and save and apply > **Authentication Type Should be set to NONE**

## Current Behavior

In Network > Interfaces > General Settings for the 5G Modem > Authentication Type > Set to NONE, and save and apply > Authentication Type Should be set to NONE, but It reverts back to `PAP/CHAP`

**Logs**:

```
Fri Apr 25 01:54:38 2025 daemon.notice netifd: 4glte (29733): couldn't load bearer path: disconnecting anyway
Fri Apr 25 01:54:38 2025 daemon.info [21391]: <inf> [modem0] processing user request to disconnect modem: all bearers
Fri Apr 25 01:54:38 2025 daemon.info [21391]: <inf> [modem0] all requested bearers disconnected
Fri Apr 25 01:54:38 2025 daemon.notice netifd: Interface '4glte' is now down
Fri Apr 25 01:54:40 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Apr 25 01:54:40 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Apr 25 01:54:40 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Apr 25 01:54:47 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Fri Apr 25 01:54:48 2025 user.notice omr-bypass: Starting OMR-ByPass...
Fri Apr 25 01:54:54 2025 daemon.warn netifd: Cannot set device name: '/sys/devices/platform/soc/11200000' is longer than max size 15
Fri Apr 25 01:54:54 2025 daemon.warn netifd: Failed to initalize device '/sys/devices/platform/soc/11200000'
Fri Apr 25 01:54:54 2025 daemon.warn netifd: Cannot set device name: '/sys/devices/platform/soc/11200000' is longer than max size 15
Fri Apr 25 01:54:54 2025 daemon.warn netifd: Failed to initalize device '/sys/devices/platform/soc/11200000'
Fri Apr 25 01:54:57 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Fri Apr 25 01:54:58 2025 user.notice omr-bypass: OMR-ByPass is running
Fri Apr 25 01:55:04 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Fri Apr 25 01:55:55 2025 user.notice post-tracking-002-error: No answer from 4glte (), restart interface
Fri Apr 25 01:55:55 2025 user.notice post-tracking-002-error: Set 4glte down
Fri Apr 25 01:56:00 2025 user.notice post-tracking-002-error: Set 4glte up
Fri Apr 25 01:56:00 2025 daemon.warn netifd: Cannot set device name: '/sys/devices/platform/soc/11200000' is longer than max size 15
Fri Apr 25 01:56:00 2025 daemon.warn netifd: Failed to initalize device '/sys/devices/platform/soc/11200000'
Fri Apr 25 01:56:00 2025 daemon.notice netifd: Interface '4glte' is setting up now
Fri Apr 25 01:56:00 2025 daemon.notice netifd: 4glte (10826): modem available at /org/freedesktop/ModemManager1/Modem/0
Fri Apr 25 01:56:01 2025 daemon.info [21391]: <inf> [modem0] processing user request to enable modem...
Fri Apr 25 01:56:01 2025 daemon.info [21391]: <inf> [modem0] enabled modem
Fri Apr 25 01:56:01 2025 daemon.notice netifd: 4glte (10826): successfully enabled the modem
Fri Apr 25 01:56:01 2025 daemon.notice netifd: 4glte (10826): setting allowed mode to 'any'
Fri Apr 25 01:56:01 2025 daemon.info [21391]: <inf> [modem0] processing user request to set current modes to 'any' (preferred 'none')...
Fri Apr 25 01:56:01 2025 daemon.warn [21391]: <wrn> [modem0] failed setting current modes to 'any' (preferred 'none'): Setting allowed modes not supported
Fri Apr 25 01:56:01 2025 daemon.notice netifd: 4glte (10826): error: couldn't set current modes: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Unsupported: Unsupported: Setting allowed modes not supported'
Fri Apr 25 01:56:01 2025 daemon.notice netifd: 4glte (10826): starting connection with apn 'internet.globe.com.ph'...
Fri Apr 25 01:56:01 2025 daemon.info [21391]: <inf> [modem0] processing user request to connect modem...
Fri Apr 25 01:56:01 2025 daemon.info [21391]: <inf> [modem0]   apn: internet.globe.com.ph
Fri Apr 25 01:56:01 2025 daemon.info [21391]: <inf> [modem0]   ip type: ipv4v6
Fri Apr 25 01:56:01 2025 daemon.info [21391]: <inf> [modem0]   allowed auth: unknown
Fri Apr 25 01:56:01 2025 daemon.info [21391]: <inf> [modem0]   allow roaming: yes
Fri Apr 25 01:56:01 2025 daemon.notice [21391]: <msg> [modem0] simple connect started...
Fri Apr 25 01:56:01 2025 daemon.notice [21391]: <msg> [modem0] simple connect state (6/10): register
Fri Apr 25 01:56:01 2025 daemon.info [21391]: <inf> [modem0] launching automatic network registration...
Fri Apr 25 01:56:01 2025 user.notice omr-schedule-010-services: Set firewall on server vps
Fri Apr 25 01:57:01 2025 daemon.info [21391]: <inf> [modem0] 3GPP registration check timed out
Fri Apr 25 01:57:01 2025 daemon.warn [21391]: <wrn> [modem0] registration in network failed: Network timeout
Fri Apr 25 01:57:01 2025 daemon.notice netifd: 4glte (10826): error: couldn't connect the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.MobileEquipment.NetworkTimeout: Network timeout'
Fri Apr 25 01:57:01 2025 daemon.notice netifd: 4glte (18492): stopping network
Fri Apr 25 01:57:01 2025 daemon.notice netifd: 4glte (18492): couldn't load bearer path: disconnecting anyway
Fri Apr 25 01:57:01 2025 daemon.info [21391]: <inf> [modem0] processing user request to disconnect modem: all bearers
Fri Apr 25 01:57:01 2025 daemon.info [21391]: <inf> [modem0] all requested bearers disconnected
Fri Apr 25 01:57:01 2025 daemon.notice netifd: Interface '4glte' is now down
Fri Apr 25 01:57:03 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Apr 25 01:57:03 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Apr 25 01:57:03 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Apr 25 01:57:11 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Fri Apr 25 01:57:12 2025 user.notice omr-bypass: Starting OMR-ByPass...
Fri Apr 25 01:57:18 2025 daemon.warn netifd: Cannot set device name: '/sys/devices/platform/soc/11200000' is longer than max size 15
Fri Apr 25 01:57:18 2025 daemon.warn netifd: Failed to initalize device '/sys/devices/platform/soc/11200000'
Fri Apr 25 01:57:18 2025 daemon.warn netifd: Cannot set device name: '/sys/devices/platform/soc/11200000' is longer than max size 15
Fri Apr 25 01:57:18 2025 daemon.warn netifd: Failed to initalize device '/sys/devices/platform/soc/11200000'
```
## Possible Solution

## Steps to Reproduce the Problem

1. In Network > Interfaces > General Settings for the 5G Modem > Authentication Type > Set to NONE
2. and save and apply > Authentication Type Should be set to NONE
3. Go back to the setting page, and you will see It reverts back to `PAP/CHAP`

## Context (Environment)

I'm using GOMO.PH SIM Card.

Already tested on Mobile Phone, no problem.

Problem is when using it with the router.

I'm not sure whether it's the router or SIM.

## Specifications

  - OpenMPTCProuter version: v0.62-6.6 r0+28431-92e020b50f
  - OpenMPTCProuter VPS version: 6.6.36-x64v3-xanmod1 0.1032
  - OpenMPTCProuter VPS provider: DigitalOcean
  - OpenMPTCProuter platform: ZBT Z8102AX-M2-A-T
  - Country: Philippines

## Screenshots
[image]
[image]

chubx, 2025-04-26:
> There is a bug, default is "none". You can check via "uci get network.4glte.auth" via SSH on the router if it's correct or not.

@Ysurac , thanks! the output is below:
```
root@OpenMPTCProuter:~# uci get network.4glte.auth
uci: Entry not found
```

> Do you know where "/sys/devices/platform/soc/11200000" is coming from ? If not it should also be in "uci show network" via SSH on the router.

**uci show network** output:

`network.4glte.device='/sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1'`

chubx, 2025-04-26:
Also, in case you need this below.

```
root@OpenMPTCProuter:~# mmcli -L
    /org/freedesktop/ModemManager1/Modem/0 [Quectel] RM500U-EA
```

```
root@OpenMPTCProuter:~# mmcli -m 0
  --------------------------------
  General  |                 path: /org/freedesktop/ModemManager1/Modem/0
           |            device id: <redacted>
  --------------------------------
  Hardware |         manufacturer: Quectel
           |                model: RM500U-EA
           |    firmware revision: RM500UEAAAR03A06M2G_01.001.01.001
           |            supported: gsm-umts
           |              current: gsm-umts
           |         equipment id: <redacted>
  --------------------------------
  System   |               device: /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1
           |              physdev: /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1
           |              drivers: cdc_ncm, option1
           |               plugin: quectel
           |         primary port: ttyUSB2
           |                ports: eth2 (net), ttyUSB2 (at), ttyUSB3 (at)
  --------------------------------
  Numbers  |                  own: <redacted>
  --------------------------------
  Status   |       unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                state: enabled
           |          power state: on
           |       signal quality: 0% (recent)
  --------------------------------
  Modes    |            supported: allowed: any; preferred: none
           |              current: allowed: any; preferred: none
  --------------------------------
  3GPP     |                 imei: <redacted>
           |         registration: idle
           | packet service state: detached
  --------------------------------
  3GPP EPS | ue mode of operation: csps-2
  --------------------------------
  SIM      |     primary sim path: /org/freedesktop/ModemManager1/SIM/0

```

Ysurac, 2025-04-26:
I was just writing to ask that.
It's configured in ncm, I only tested Quectel modems in mbim mode. Can you switch it to mbim (easier to use with ModemManager) ? ```mmcli -m 0 --command='AT+QCFG="usbnet",2'```

Can you give me the result of ```uci show network.4glte``` ?

chubx, 2025-04-26:
> I was just writing to ask that. It's configured in ncm, I only tested Quectel modems in mbim mode. Can you switch it to mbim (easier to use with ModemManager) ? `mmcli -m 0 --command='AT+QCFG="usbnet",2'`

```
root@OpenMPTCProuter:/etc/config# mmcli -m 0 --command='AT+QCFG="usbnet",2'
response: ''
```

> Can you give me the result of `uci show network.4glte` ?

```
root@OpenMPTCProuter:/etc/config# uci show network.4glte
network.4glte=interface
network.4glte.proto='mbim'
network.4glte.label='4glte'
network.4glte.peerdns='0'
network.4glte.pdptype='ipv4v6'
network.4glte.device='/sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1'
network.4glte.apn='gomo.ph'
network.4glte.auth='none'
network.4glte.multipath='off'
network.4glte.addlatency='0'
network.4glte.metric='24'
network.4glte.ip4table='24'
network.4glte.ip6table='624'

```

> I was just writing to ask that. It's configured in ncm, I only tested Quectel modems in mbim mode. Can you switch it to mbim (easier to use with ModemManager)

Can I please ask what do you mean by "**easier to use with ModemManager**"?

I may be doing it wrong, but the setup is like this:
1. In Network > Interfaces > **4glte** > Edit :
**Protocol**: MBIM Cellular
**Modem device**:` /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1`

2. In System > OpenMPTCP Router > Settings Wizard > Interface Settings > 4glte 
**Type**: Normal
**Protocol**: Other
**Multipath TCP**: Enabled

chubx, 2025-04-26:
**With MBIM Cellular** Mode, the output for your requested commands are:

```
root@OpenMPTCProuter:~# mmcli -m 0 --command='AT+QCFG="usbnet",2'
response: ''
```
```

root@OpenMPTCProuter:~# uci show network.4glte

network.4glte=interface
network.4glte.proto='mbim'
network.4glte.label='4glte'
network.4glte.peerdns='0'
network.4glte.pdptype='ipv4v6'
network.4glte.multipath='on'
network.4glte.addlatency='0'
network.4glte.metric='24'
network.4glte.ip4table='24'
network.4glte.ip6table='624'
network.4glte.defaultroute='0'
network.4glte.delegate='0'
network.4glte.apn='gomo.ph'
network.4glte.auth='none'
network.4glte.ipv6='0'
```

chubx, 2025-04-26:
**Bug*? I think another bug in MBIM configuration page.

When I edit in Network > Interface > 4glte > Modem Device: `/sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1`

After saving, the Error comes up: `Error: Network device is not present`

And when I use command **uci show network.4glte**, output is:
```

network.4glte=interface
network.4glte.proto='mbim'
network.4glte.label='4glte'
network.4glte.pdptype='ipv4'
network.4glte.apn='gomo.ph'
network.4glte.auth='none'
network.4glte.multipath='off'
network.4glte.addlatency='0'
network.4glte.metric='24'
network.4glte.ip4table='24'
network.4glte.ip6table='624'
network.4glte.dns='1.1.1.1' '8.8.8.8'
network.4glte.device='/sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1'

```

Ysurac, 2025-04-26:
Remove DNS set on the interface, set it as ModemManager proto not as mbim and reboot

chubx, 2025-04-26:
> Remove DNS set on the interface, set it as ModemManager proto not as mbim and reboot

Is this where i set the protocol to **ModemManager**?

[image]

If so, I already did. And still the authentication goes back to PAP/CHAP, after setting to NONE.

Plus, the error:
**Error: Unable to set preferred mode.**

Ysurac, 2025-04-26:
Can I have again the result of mmcli ?

chubx, 2025-04-27:
> Can I have again the result of mmcli ?

Here's my mmcli output, while interface is set to ModemManager:

```

root@OpenMPTCProuter:~# mmcli -m 0
  --------------------------------
  General  |                 path: /org/freedesktop/ModemManager1/Modem/0
           |            device id: <redacted>
  --------------------------------
  Hardware |         manufacturer: Quectel
           |                model: RM500U-EA
           |    firmware revision: sprd
           |         h/w revision: UDX710_modem
           |            supported: gsm-umts, lte, 5gnr
           |              current: gsm-umts, lte, 5gnr
           |         equipment id: <redacted>
  --------------------------------
  System   |               device: /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1
           |              physdev: /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1
           |              drivers: cdc_mbim, option1
           |               plugin: quectel
           |         primary port: cdc-wdm0
           |                ports: cdc-wdm0 (mbim), ttyUSB2 (at), ttyUSB3 (at), wwan0 (net)
  --------------------------------
  Numbers  |                  own: <redacted>
  --------------------------------
  Status   |       unlock retries: sim-pin (3)
           |                state: connected
           |          power state: on
           |          access tech: lte
           |       signal quality: 54% (cached)
  --------------------------------
  Modes    |            supported: allowed: 2g, 3g, 4g, 5g; preferred: none
           |              current: allowed: 2g, 3g, 4g, 5g; preferred: none
  --------------------------------
  IP       |            supported: ipv4, ipv6, ipv4v6
  --------------------------------
  3GPP     |                 imei: <redacted>
           |          operator id: <redacted>
           |        operator name: Globe Telecom-PH
           |         registration: home
           | packet service state: attached
  --------------------------------
  3GPP EPS | ue mode of operation: ps-2
  --------------------------------
  SIM      |     primary sim path: /org/freedesktop/ModemManager1/SIM/0
  --------------------------------
  Bearer   |                paths: /org/freedesktop/ModemManager1/Bearer/0
```

Ysurac, 2025-04-27:
Seems far better.  For the preferred mode, only one mode is supported that is the current one, quite strange.
I have this on a RM520N-GL:
```
  Modes     |               supported: allowed: 3g; preferred: none
            |                          allowed: 4g; preferred: none
            |                          allowed: 3g, 4g; preferred: 4g
            |                          allowed: 3g, 4g; preferred: 3g
            |                          allowed: 5g; preferred: none
            |                          allowed: 4g, 5g; preferred: 5g
            |                          allowed: 4g, 5g; preferred: 4g
            |                          allowed: 3g, 5g; preferred: 5g
            |                          allowed: 3g, 5g; preferred: 3g
            |                          allowed: 3g, 4g, 5g; preferred: 5g
            |                          allowed: 3g, 4g, 5g; preferred: 4g
            |                          allowed: 3g, 4g, 5g; preferred: 3g
            |                 current: allowed: 3g, 4g, 5g; preferred: 5g
```

chubx, 2025-04-29:
> Seems far better. For the preferred mode, only one mode is supported that is the current one, quite strange. I have this on a RM520N-GL:
>
> ```
>   Modes     |               supported: allowed: 3g; [...]

@Ysurac I've tried single mode 4G only, but that did not work, and other combinations.

The only thing that made it worked is setting it to `any`

Ysurac, 2025-04-29:
"any" is the only available mode on your modem.
OpenMPTCProuter status page ? It's when some tests timeout. All is green or not ?

chubx, 2025-04-29:
> "any" is the only available mode on your modem. OpenMPTCProuter status page ? It's when some tests timeout. All is green or not ?

Page load is very slow when I go to (this has been the case since I flashed it):
System > OpenMPTCProuter > Settings Wizard

On the router all green.

eyramm, 2025-05-15:
I have the exact issue with the same router @chubx 
I havent been able to get any SIM card to work in the two SIM slots and the Settings Wizard page is painfully slow
Have you been able to figure it out?

Ysurac, 2025-05-15:
@eyramm also with ModemManager ? and what are the modems models ?

davidohne, 2025-05-26:
Here it's the same. Setup Wizard takes about 60-90 seconds to load.

Ysurac, 2025-05-27:
So same question

davidohne, 2025-05-27:
Ah sorry, didn't see it.
2x T99w175 using modem manager for both. Using bbr as algorithm and the mode where every package gets send to every connection.

Ysurac, 2025-06-10:
I would need Status->System log and ```htop``` command result via SSH.
The box was mostly tested with Quectel modems in MBIM mode.