Ysurac/openmptcprouter #3924: What is the problem?

lhc1993, 2025-04-26:
## Expected Behavior

The network suddenly got stuck. I don't know what happened. I hope to clearly identify the problem.
## Current Behavior

Sat Apr 26 21:33:05 2025 daemon.info hostapd: phy1-ap0: STA 40:1c:83:5b:02:8e RADIUS: starting accounting session 8D48A5AF933165DF
Sat Apr 26 21:33:05 2025 daemon.info hostapd: phy1-ap0: STA 40:1c:83:5b:02:8e WPA: pairwise key handshake completed (RSN)
Sat Apr 26 21:33:05 2025 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 40:1c:83:5b:02:8e
Sat Apr 26 21:50:25 2025 user.notice post-tracking-020-status: Check API configuration...
Sat Apr 26 21:50:25 2025 user.notice post-tracking-020-status: Check API configuration... Done
Sat Apr 26 22:00:21 2025 daemon.info hostapd: phy1-ap0: STA 08:f9:e0:66:dd:1d IEEE 802.11: disconnected due to excessive missing ACKs
Sat Apr 26 22:00:21 2025 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 08:f9:e0:66:dd:1d
Sat Apr 26 22:00:26 2025 kern.warn kernel: [18050.121084] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 18
Sat Apr 26 22:00:27 2025 daemon.notice hostapd: phy1-ap0: STA 08:f9:e0:66:dd:1d IEEE 802.11: did not acknowledge authentication response
Sat Apr 26 22:00:28 2025 daemon.info hostapd: phy1-ap0: STA 08:f9:e0:66:dd:1d IEEE 802.11: associated (aid 3)
Sat Apr 26 22:00:29 2025 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 08:f9:e0:66:dd:1d auth_alg=open
Sat Apr 26 22:00:29 2025 daemon.info hostapd: phy1-ap0: STA 08:f9:e0:66:dd:1d RADIUS: starting accounting session 9B75289820299A2E
Sat Apr 26 22:00:29 2025 daemon.info hostapd: phy1-ap0: STA 08:f9:e0:66:dd:1d WPA: pairwise key handshake completed (RSN)
Sat Apr 26 22:00:29 2025 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 08:f9:e0:66:dd:1d
Sat Apr 26 22:01:18 2025 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 08:f9:e0:66:dd:03
Sat Apr 26 22:01:18 2025 daemon.notice hostapd: phy1-ap0: STA 08:f9:e0:66:dd:03 IEEE 802.11: did not acknowledge authentication response
Sat Apr 26 22:01:18 2025 daemon.info hostapd: phy1-ap0: STA 08:f9:e0:66:dd:03 IEEE 802.11: associated (aid 2)
Sat Apr 26 22:01:18 2025 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 08:f9:e0:66:dd:03 auth_alg=open
Sat Apr 26 22:01:18 2025 daemon.info hostapd: phy1-ap0: STA 08:f9:e0:66:dd:03 RADIUS: starting accounting session D8F026B58ACFA2D0
Sat Apr 26 22:01:18 2025 daemon.info hostapd: phy1-ap0: STA 08:f9:e0:66:dd:03 WPA: pairwise key handshake completed (RSN)
Sat Apr 26 22:01:18 2025 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 08:f9:e0:66:dd:03
Sat Apr 26 22:01:24 2025 daemon.notice netifd: pcie_mhi0 (25795): udhcpc: sending renew to server 10.170.140.62
Sat Apr 26 22:01:24 2025 daemon.notice netifd: pcie_mhi0 (25795): udhcpc: lease of 10.170.140.61 obtained from 10.170.140.62, lease time 7200
Sat Apr 26 22:05:47 2025 daemon.notice netifd: pcie_mhi1 (29378): udhcpc: sending renew to server 10.129.207.128
Sat Apr 26 22:05:47 2025 daemon.notice netifd: pcie_mhi1 (29378): udhcpc: lease of 10.129.207.127 obtained from 10.129.207.128, lease time 7200
Sat Apr 26 22:06:29 2025 user.notice post-tracking-002-error: omrvpn down because IPv4 gateway down
Sat Apr 26 22:06:30 2025 user.notice post-tracking-002-error: OpenVPN down, restart it
Sat Apr 26 22:06:30 2025 daemon.err openvpn(omr)[27820]: event_wait : Interrupted system call (fd=-1,code=4)
Sat Apr 26 22:06:30 2025 daemon.notice openvpn(omr)[27820]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Sat Apr 26 22:06:30 2025 daemon.notice openvpn(omr)[27820]: net_addr_v4_del: 10.255.252.2 dev tun0
Sat Apr 26 22:06:30 2025 daemon.notice ttyd[22634]: [2025/04/26 22:06:30:9838] N: rops_handle_POLLIN_netlink: DELADDR
Sat Apr 26 22:06:30 2025 daemon.notice ttyd[22634]: [2025/04/26 22:06:30:9864] N: rops_handle_POLLIN_netlink: DELADDR
Sat Apr 26 22:06:31 2025 user.notice NET: hotplug (iface): action='remove' interface='tun0'
Sat Apr 26 22:06:31 2025 daemon.info ModemManager[32033]: hotplug: remove network interface tun0: event processed
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[27820]: /usr/libexec/openvpn-hotplug down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[27820]: SIGTERM[hard,] received, process exiting
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: OpenVPN 2.6.13 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: library versions: OpenSSL 3.0.15 3 Sep 2024, LZO 2.10
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: DCO version: N/A
Sat Apr 26 22:06:31 2025 daemon.warn openvpn(omr)[32156]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Sat Apr 26 22:06:31 2025 daemon.warn openvpn(omr)[32156]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: TCP/UDP: Preserving recently used remote address: [AF_INET]8.133.203.163:65301
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: Attempting to establish TCP connection with [AF_INET]8.133.203.163:65301
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: TCP connection established with [AF_INET]8.133.203.163:65301
Sat Apr 26 22:06:31 2025 daemon.warn openvpn(omr)[32156]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Not supported (errno=95)
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: TCPv4_CLIENT link local: (not bound)
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: TCPv4_CLIENT link remote: [AF_INET]8.133.203.163:65301
Sat Apr 26 22:06:31 2025 daemon.notice netifd: Interface 'omrvpn' is disabled
Sat Apr 26 22:06:31 2025 daemon.notice netifd: Network device 'tun0' link is down
Sat Apr 26 22:06:31 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sat Apr 26 22:06:31 2025 daemon.notice netifd: omrvpn (28020): udhcpc: read error: Network is down, reopening socket
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: [server] Peer Connection Initiated with [AF_INET]8.133.203.163:65301
Sat Apr 26 22:06:31 2025 daemon.notice netifd: omrvpn (28020): udhcpc: bind: No such device
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: TUN/TAP device tun0 opened
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: net_iface_mtu_set: mtu 1420 for tun0
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: net_iface_up: set tun0 up
Sat Apr 26 22:06:31 2025 daemon.notice netifd: Network device 'tun0' link is up
Sat Apr 26 22:06:31 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: net_addr_v4_add: 10.255.252.3/24 dev tun0
Sat Apr 26 22:06:31 2025 daemon.notice openvpn(omr)[32156]: /usr/libexec/openvpn-hotplug up omr tun0 1420 0 10.255.252.3 255.255.255.0 init
Sat Apr 26 22:06:31 2025 daemon.notice netifd: Interface 'omrvpn' is now down
Sat Apr 26 22:06:31 2025 daemon.notice netifd: Interface 'omrvpn' is enabled
Sat Apr 26 22:06:31 2025 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sat Apr 26 22:06:31 2025 daemon.notice netifd: omrvpn (32345): udhcpc: started, v1.36.1
Sat Apr 26 22:06:31 2025 user.notice NET: hotplug (iface): action='add' interface='tun0'
Sat Apr 26 22:06:31 2025 daemon.notice netifd: omrvpn (32345): udhcpc: broadcasting discover
Sat Apr 26 22:06:32 2025 daemon.info ModemManager[32278]: hotplug: add network interface tun0: event processed
Sat Apr 26 22:06:34 2025 daemon.notice netifd: omrvpn (32345): udhcpc: broadcasting discover
Sat Apr 26 22:06:35 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was 10.170.140.62)
Sat Apr 26 22:06:36 2025 daemon.notice openvpn(omr)[32156]: Initialization Sequence Completed
Sat Apr 26 22:06:37 2025 daemon.notice netifd: omrvpn (32345): udhcpc: broadcasting discover
Sat Apr 26 22:06:43 2025 daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules
Sat Apr 26 22:06:44 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sat Apr 26 22:06:44 2025 user.notice omr-bypass: Starting OMR-ByPass...
Sat Apr 26 22:06:55 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
[... cut ...]
Sat Apr 26 22:06:57 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sat Apr 26 22:07:00 2025 authpriv.info dropbear[14528]: Child connection from 106.37.96.11:26049
Sat Apr 26 22:07:01 2025 user.notice omr-schedule-010-services: Set firewall on server vps
Sat Apr 26 22:07:01 2025 authpriv.notice dropbear[14528]: Password auth succeeded for 'root' from 106.37.96.11:26049

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

vecjh, 2025-04-26:
```
Sat Apr 26 22:06:29 2025 user.notice post-tracking-002-error: omrvpn down because IPv4 gateway down
...
Sat Apr 26 22:06:31 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
...
Sat Apr 26 22:06:31 2025 daemon.notice netifd: Network device 'tun0' link is up
Sat Apr 26 22:06:31 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity
```

I think you had network hiccup and it should now be recovered, shouldn't it?

lhc1993, 2025-04-26:
Generally, errors like this will occur, which means it is a network problem.
post-tracking-002-error: pcie_mhi0 (pcie_mhi0) switched off because check error and ping from 10.170.140.61 error (114.114.114.114,1.1.1.1,208.67.222.222)

lhc1993, 2025-04-26:
in vps logs:
Apr 26 23:32:05 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1170]: INFO:     127.0.0.1:25529 - "GET / HTTP/1.1" 200 OK
Apr 26 23:32:16 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1169]: INFO:     127.0.0.1:13781 - "GET / HTTP/1.1" 200 OK
Apr 26 23:32:19 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1170]: INFO:     36.60.202.140:16994 - "GET /status HTTP/1.1" 200 OK
Apr 26 23:32:28 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1169]: INFO:     127.0.0.1:11437 - "GET / HTTP/1.1" 200 OK
Apr 26 23:32:39 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1169]: INFO:     36.60.202.140:16859 - "GET /status HTTP/1.1" 200 OK
Apr 26 23:32:40 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1169]: INFO:     127.0.0.1:55625 - "GET / HTTP/1.1" 200 OK
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: WARNING: Failed to stat CRL file, not reloading CRL.
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: WARNING: Failed to stat CRL file, not reloading CRL.
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: TCP connection established with [AF_INET6]::ffff:36.60.202.140:16939
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: Socket flags: TCP_NODELAY=1 succeeded
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Operation not supported (errno=95)
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: Note: enable extended error passing on TCP/UDP socket failed (IPV6_RECVERR): Operation not supported (errno=95)
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 TLS: Initial packet from [AF_INET6]::ffff:36.60.202.140:16939, sid=1a5f9de7 bc6d05d2
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 VERIFY OK: depth=1, CN=Easy-RSA CA
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 VERIFY OK: depth=0, CN=openmptcprouter
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 peer info: IV_VER=2.6.13
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 peer info: IV_PLAT=linux
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 peer info: IV_TCPNL=1
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 peer info: IV_MTU=1600
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 peer info: IV_NCP=2
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 peer info: IV_CIPHERS=AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 peer info: IV_PROTO=990
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 peer info: IV_LZO_STUB=1
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 peer info: IV_COMP_STUB=1
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 peer info: IV_COMP_STUBv2=1
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 TLS: move_session: dest=TM_ACTIVE src=TM_INITIAL reinit_src=1
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 WARNING: Failed to stat CRL file, not reloading CRL.
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 TLS: tls_multi_process: initial untrusted session promoted to trusted
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256_GCM_SHA384, peer certificate: 2048 bit>
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: 36.60.202.140:16939 [openmptcprouter] Peer Connection Initiated with [AF_INET6]::ffff:36.60.202.140:16939
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: openmptcprouter/36.60.202.140:16939 MULTI_sva: pool returned IPv4=10.255.252.2, IPv6=(Not enabled)
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: openmptcprouter/36.60.202.140:16939 Could not access file 'ccd/openmptcprouter': Permission denied (errno=13)
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: openmptcprouter/36.60.202.140:16939 Could not access file 'ccd/DEFAULT': Permission denied (errno=13)
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: openmptcprouter/36.60.202.140:16939 MULTI: Learn: 10.255.252.2 -> openmptcprouter/36.60.202.140:16939
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: openmptcprouter/36.60.202.140:16939 MULTI: primary virtual IP for openmptcprouter/36.60.202.140:16939: 10.255.2>
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: openmptcprouter/36.60.202.140:16939 SENT CONTROL [openmptcprouter]: 'PUSH_REPLY,sndbuf 262144,rcvbuf 262144,rou>
Apr 26 23:32:52 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1176]: INFO:     127.0.0.1:60925 - "GET / HTTP/1.1" 200 OK
Apr 26 23:32:53 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: openmptcprouter/36.60.202.140:16939 Data Channel: cipher 'AES-256-GCM', peer-id: 0
Apr 26 23:32:53 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: openmptcprouter/36.60.202.140:16939 Timers: ping 10, ping-restart 480
Apr 26 23:32:53 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: openmptcprouter/36.60.202.140:16939 Protocol options: protocol-flags cc-exit tls-ekm dyn-tls-crypt
Apr 26 23:32:56 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1169]: INFO:     36.60.202.140:16798 - "POST /token HTTP/1.1" 200 OK
Apr 26 23:32:56 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:55391
Apr 26 23:32:56 iZuf648xlh6tywbwu7y2uvZ ovpn-tun0[636]: MANAGEMENT: CMD 'status'
[... cut ...]
Apr 26 23:32:58 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1169]: INFO:     36.60.202.140:16805 - "POST /shorewalllist HTTP/1.1" 200 OK
Apr 26 23:32:58 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1170]: INFO:     36.60.202.140:16806 - "POST /shorewalllist HTTP/1.1" 200 OK
Apr 26 23:32:59 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1169]: INFO:     36.60.202.140:16975 - "POST /shorewalllist HTTP/1.1" 200 OK
Apr 26 23:32:59 iZuf648xlh6tywbwu7y2uvZ omr-admin.py[1169]: INFO:     36.60.202.140:16976 - "GET /status HTTP/1.1" 200 OK

Ysurac, 2025-04-26:
Please complete the template.
Can you also explain more the issue: network problem on LAN, wifi ? it's still working from the router with status page ok or not ?

closed 2025-07-31 by github-actions[bot]