Ysurac/openmptcprouter #3926: kernel 6.1 not successful building

GermanAizek, 2025-04-27:
@Ysurac,
```
make menuconfig
make -j1 V=sc
```

I tried `make clean`. Others kernel successful building from sources.

```
touch /tmp/openmptcprouter/x86_64/6.1/source/build_dir/host/zstd-1.5.5/.prepared0c3c55a156970e7ac9449d24053ec2a9_18f1e190c5d53547fed41a3eaa76e9e9
/tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/bin/sed -e "s|@CC@|'/tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/bin/gcc',|" -e "s|@CXX@|'/tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/bin/g++',|" -e "s|@PKGCONFIG@|/tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-O2', '-I/tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/include', '-I/tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/include',|" -e "s|@CXXFLAGS@|'-I/tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/include',|" -e "s|@LDFLAGS@|'-L/tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/lib',|" -e "s|@PREFIX@|/tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host|" < /tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/lib/meson/openwrt-native.txt.in > /tmp/openmptcprouter/x86_64/6.1/source/build_dir/host/zstd-1.5.5/openwrt-native.txt
bash: line 1: /tmp/openmptcprouter/x86_64/6.1/source/staging_dir/host/lib/meson/openwrt-native.txt.in: No such file or directory
make[3]: *** [Makefile:36: /tmp/openmptcprouter/x86_64/6.1/source/build_dir/host/zstd-1.5.5/.configured] Error 1
make[3]: Leaving directory '/tmp/openmptcprouter/x86_64/6.1/source/tools/zstd'
time: tools/zstd/compile#0.06#0.02#0.08
    ERROR: tools/zstd failed to build.
make[2]: *** [tools/Makefile:233: tools/zstd/compile] Error 1
```

GermanAizek, 2025-04-27:
I found similar issues:
https://lists.openwrt.org/pipermail/openwrt-devel/2020-September/031528.html
https://git.openwrt.org/?p=openwrt/openwrt.git;a=blobdiff;f=tools/meson/src/openwrt-native.txt.in;fp=tools/meson/src/openwrt-native.txt.in;h=50308ec8e433b1d9ba4c25ef492057edde00f0d4;hp=0000000000000000000000000000000000000000;hb=c4dfdde2eaa399025de1032bed8ee8a5f8babb24;hpb=4b898afdd2fd9dbbc99dbedca0d50785e3ec063d

GermanAizek, 2025-04-27:
Solution from https://lists.openwrt.org/pipermail/openwrt-devel/2020-September/031531.html
Use `make dirclean` not working, same error.

```
ERROR: tools/zstd failed to build.
```

Ysurac, 2025-04-27:
Kernel 6.1 is not supported any more in OpenMPTCProuter, only kernel 6.6 is currently supported.

GermanAizek, 2025-04-27:
> Kernel 6.1 is not supported any more in OpenMPTCProuter, only kernel 6.6 is currently supported.

Many thanks

closed 2025-04-27 by GermanAizek