Ysurac/openmptcprouter #3930: Home fiber optic network only 4-5M upload speed.

fareign, 2025-04-29:
## Expected Behavior

when I use only one home fiber, upload speed  is very slow.
I try select diffent proxy or VPN.  is the same .
but when I use mobile sim card . it can run up.

[image]

[image]

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.63snapshot 6.12 kernel
  - OpenMPTCProuter VPS version: 0.1032
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country:

Ysurac, 2025-04-29:
You tried another VPS provider ? What is the latency when you use mobile ?

fareign, 2025-04-29:
mobile is ok. about 40ms, fiber is 30ms download is about 600mbps,but upload only 5mbps.
I test 0.62 have the same problem.
but 0.61 and 0.60 is fine.

fareign, 2025-04-29:
I go back to test 0.60,have the same problem.maybe the problem is the VPS provider.

fareign, 2025-05-01:
I change another VPS provider.have the same problem.
also I find ,ss Proxy is about 5M
and v2ray only 3M.
so this is very strange

fareign, 2025-05-01:
can't get public IP from xray vless reality,I tryed 0.1030 1031 1032VPS.

fareign, 2025-05-01:
YES，china. the fiber is china mobile .I will call my friends test other fiber as chinanet. I remember not long ago it is no problem.

closed 2025-08-04 by github-actions[bot]