Ysurac/openmptcprouter #3936: home fiber network upload speed test

fareign, 2025-05-06:
[image]
I Disable the VPS

[image]
use omr speed test .only 3mbits

[image]
when I transform a file to the VPS,it can reach 6.4M/S(50Mbits)

So where is the wrong?

## Specifications

  - OpenMPTCProuter version: 0.63 snapshot 6.6
  - OpenMPTCProuter VPS version: 0.1032
  - OpenMPTCProuter VPS provider:aliyun
  - OpenMPTCProuter platform: 
  - Country: CN

Ysurac, 2025-05-06:
Duplicate of #3930
This mean upload from WAN2 to VPS is slow. Here you can see on IPerf that many packets need to be retried, try to set a lower MTU.

fareign, 2025-05-06:
I set mtu to1450, have no effect

fareign, 2025-05-06:
set to 1380

[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   768 KBytes  6.29 Mbits/sec  531   40.8 KBytes       
[  5]   1.00-2.00   sec   256 KBytes  2.10 Mbits/sec  173   51.7 KBytes       
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    5   1.36 KBytes       
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    1   1.36 KBytes       
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    1   1.36 KBytes       
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    0   1.36 KBytes       
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    0   1.36 KBytes       
[  5]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    1   1.36 KBytes       
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0   1.36 KBytes       
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    0   1.36 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.00 MBytes   839 Kbits/sec  712            sender
[  5]   0.00-10.03  sec   896 KBytes   732 Kbits/sec                  receiver

iperf Done.
root@OpenMPTCProuter:~# omr-iperf vps
Connecting to host 47.239.126.71, port 65400
[  5] local 192.168.50.101 port 28221 connected to 47.239.126.71 port 65400
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  1.00 MBytes  8.38 Mbits/sec  576   15.5 KBytes       
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec  252   12.7 KBytes       
[  5]   2.00-3.00   sec   384 KBytes  3.15 Mbits/sec   18   12.7 KBytes       
[  5]   3.00-4.00   sec   384 KBytes  3.15 Mbits/sec   17   15.5 KBytes       
[  5]   4.00-5.00   sec   384 KBytes  3.15 Mbits/sec   11   12.7 KBytes       
[  5]   5.00-6.00   sec   256 KBytes  2.10 Mbits/sec    3   11.2 KBytes       
[  5]   6.00-7.00   sec   384 KBytes  3.15 Mbits/sec   57   12.7 KBytes       
[  5]   7.00-8.00   sec   384 KBytes  3.14 Mbits/sec   11   11.2 KBytes       
[  5]   8.00-9.00   sec   384 KBytes  3.15 Mbits/sec   27   15.5 KBytes       
[  5]   9.00-10.00  sec   384 KBytes  3.15 Mbits/sec   16   14.1 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  3.88 MBytes  3.25 Mbits/sec  988            sender
[  5]   0.00-10.04  sec  3.75 MBytes  3.13 Mbits/sec                  receiver

fareign, 2025-05-06:
[image]

I user fiber not by OMR router,has the same problem,
but when I use -R.
then upload is OK.

fareign, 2025-05-07:
I find some usefull message.
I change this VPS system to normal debian. upload gones oK.
so ,there maybe some app couse this.

Ysurac, 2025-05-07:
Can you check result in ```dmesg``` command on VPS ?

fareign, 2025-05-07:
[    4.702895] usbhid: USB HID core driver
[    4.705123] input: QEMU QEMU USB Tablet as /devices/pci0000:00/0000:00:01.2/usb1/1-1/1-1:1.0/0003:0627:0001.0001/input/input6
[    4.705782] hid-generic 0003:0627:0001.0001: input,hidraw0: USB HID v0.01 Pointer [QEMU QEMU USB Tablet] on usb-0000:00:01.2-1/input0
[    4.819793] EXT4-fs (vda3): mounted filesystem 1cb54a71-1036-42f8-bfc9-4242d97ab77b ro with ordered data mode. Quota mode: none.
[    5.046330] systemd[1]: Inserted module 'autofs4'
[    5.089204] systemd[1]: systemd 252.36-1~deb12u1 running in system mode (+PAM +AUDIT +SELINUX +APPARMOR +IMA +SMACK +SECCOMP +GCRYPT -GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN +IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 -PWQUALITY +P11KIT +QRENCODE +TPM2 +BZIP2 +LZ4 +XZ +ZLIB +ZSTD -BPF_FRAMEWORK -XKBCOMMON +UTMP +SYSVINIT default-hierarchy=unified)
[    5.091169] systemd[1]: Detected virtualization kvm.
[    5.091468] systemd[1]: Detected architecture x86-64.
[    5.097341] systemd[1]: Hostname set to <iZwz9ecl90hv4wg2bwwqqmZ>.
[    5.555792] systemd[1]: Queued start job for default target graphical.target.
[    5.567110] systemd[1]: Created slice system-dsvpn\x2dserver.slice - Slice /system/dsvpn-server.
[    5.569096] systemd[1]: Created slice system-getty.slice - Slice /system/getty.
[    5.571466] systemd[1]: Created slice system-glorytun\x2dtcp.slice - Slice /system/glorytun-tcp.
[    5.573736] systemd[1]: Created slice system-glorytun\x2dudp.slice - Slice /system/glorytun-udp.
[    5.575968] systemd[1]: Created slice system-mlvpn.slice - Slice /system/mlvpn.
[    5.578350] systemd[1]: Created slice system-modprobe.slice - Slice /system/modprobe.
[    5.580667] systemd[1]: Created slice system-omr6in4.slice - Slice /system/omr6in4.
[    5.582958] systemd[1]: Created slice system-openvpn.slice - Slice /system/openvpn.
[    5.585191] systemd[1]: Created slice system-serial\x2dgetty.slice - Slice /system/serial-getty.
[    5.587342] systemd[1]: Created slice system-shadowsocks\x2dlibev\x2dmanager.slice - Slice /system/shadowsocks-libev-manager.
[    5.589674] systemd[1]: Created slice system-wg\x2dquick.slice - Slice /system/wg-quick.
[    5.591738] systemd[1]: Created slice user.slice - User and Session Slice.
[    5.593098] systemd[1]: Started systemd-ask-password-console.path - Dispatch Password Requests to Console Directory Watch.
[    5.595201] systemd[1]: Started systemd-ask-password-wall.path - Forward Password Requests to Wall Directory Watch.
[    5.597338] systemd[1]: Set up automount proc-sys-fs-binfmt_misc.automount - Arbitrary Executable File Formats File System Automount Point.
[    5.599779] systemd[1]: Expecting device dev-disk-by\x2duuid-0420\x2dD9F6.device - /dev/disk/by-uuid/0420-D9F6...
[    5.601555] systemd[1]: Expecting device dev-ttyS0.device - /dev/ttyS0...
[    5.603234] systemd[1]: Reached target cryptsetup.target - Local Encrypted Volumes.
[    5.604475] systemd[1]: Reached target integritysetup.target - Local Integrity Protected Volumes.
[    5.605756] systemd[1]: Reached target paths.target - Path Units.
[    5.607325] systemd[1]: Reached target remote-fs.target - Remote File Systems.
[    5.609030] systemd[1]: Reached target slices.target - Slice Units.
[    5.610663] systemd[1]: Reached target swap.target - Swaps.
[    5.612155] systemd[1]: Reached target time-set.target - System Time Set.
[    5.613968] systemd[1]: Reached target veritysetup.target - Local Verity Protected Volumes.
[    5.615669] systemd[1]: Listening on systemd-fsckd.socket - fsck to fsckd communication Socket.
[    5.617452] systemd[1]: Listening on systemd-initctl.socket - initctl Compatibility Named Pipe.
[    5.620211] systemd[1]: Listening on systemd-journald-audit.socket - Journal Audit Socket.
[    5.622777] systemd[1]: Listening on systemd-journald-dev-log.socket - Journal Socket (/dev/log).
[    5.625322] systemd[1]: Listening on systemd-journald.socket - Journal Socket.
[    5.627903] systemd[1]: Listening on systemd-networkd.socket - Network Service Netlink Socket.
[    5.630539] systemd[1]: Listening on systemd-udevd-control.socket - udev Control Socket.
[    5.632382] systemd[1]: Listening on systemd-udevd-kernel.socket - udev Kernel Socket.
[    5.642125] systemd[1]: Mounting dev-hugepages.mount - Huge Pages File System...
[    5.646853] systemd[1]: Mounting dev-mqueue.mount - POSIX Message Queue File System...
[    5.653682] systemd[1]: Mounting sys-kernel-debug.mount - Kernel Debug File System...
[    5.656523] systemd[1]: sys-kernel-tracing.mount - Kernel Trace File System was skipped because of an unmet condition check (ConditionPathExists=/sys/kernel/tracing).
[    5.669214] systemd[1]: Starting keyboard-setup.service - Set the console keyboard layout...
[    5.673427] systemd[1]: Starting kmod-static-nodes.service - Create List of Static Device Nodes...
[    5.690165] systemd[1]: Starting modprobe@configfs.service - Load Kernel Module configfs...
[    5.706171] systemd[1]: Starting modprobe@dm_mod.service - Load Kernel Module dm_mod...
[    5.709241] systemd[1]: Starting modprobe@drm.service - Load Kernel Module drm...
[    5.731184] systemd[1]: Starting modprobe@efi_pstore.service - Load Kernel Module efi_pstore...
[    5.744665] systemd[1]: Starting modprobe@fuse.service - Load Kernel Module fuse...
[    5.747913] pstore: Using crash dump compression: deflate
[    5.749588] pstore: Registered efi_pstore as persistent store backend
[... cut ...]
[   14.255396] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[   14.261639] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[   17.403305] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   18.215630] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   18.231232] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[   26.882301] systemd[794]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set

fareign, 2025-05-07:
I stoped some OMR service . but it is no usefull.

Ysurac, 2025-05-07:
Can you try, only for the test, to disable ```shorewall``` service ?

fareign, 2025-05-07:
I first disabled it.but nothing change. I even reboot it.

Ysurac, 2025-05-09:
What is the result directly from the box with ```omr-iperf-vps``` ?

fareign, 2025-05-09:
> What is the result directly from the box with `omr-iperf-vps` ?

then same . maybe is the fiber problem.

closed 2025-08-12 by github-actions[bot]