Ysurac/openmptcprouter #3938: Is OMR 0.62 supposed to work on N100 PC ?

patimages, 2025-05-07:
Hi, 

This message is for the community but also Ycarus as I remember that for the last update, we had lots of issues with N100 and Ycarus bought one for testing/fixing purposes. Was 0.62 also tested on N100 ? I tried yesterday and ended up in a booting-rebooting loop. I should have taken a picture but did not have time and had to quickly re-install 0.61. 

Many thanks for any help !

Ysurac, 2025-05-07:
Duplicate of #3804

patimages, 2025-05-07:
Sorry, I missed it ! Would the actual snapshot found here be the proper one or should I go back in the snapshot version history ? https://snapshots.openmptcprouter.com/6.6/x86_64/packages/x86_64/openmptcprouter/
Thanks again !

Ysurac, 2025-05-07:
There is no snapshot version history.
The link is: https://snapshots.openmptcprouter.com/6.6/x86_64/targets/x86/64/

patimages, 2025-05-07:
Got it, thanks !

patimages, 2025-05-07:
Top support, as always !

closed 2025-05-07 by patimages