Ysurac/openmptcprouter #3939: Charity-Projekt Marienkäfer

derMarienkaefer, 2025-05-08:
## Expected Behavior

 With the update from version 0.5.9 to version 0.6.2, we encountered the following issues: We are streaming live from a moving bus using 5 mobile networks and 3 Starlinks, while the control room connects remotely to the vehicle. With version 0.5.9, this worked seamlessly. However, with version 0.6.2, we are experiencing connection drops as the vehicle moves.

## Current Behavior

[image]

[message.txt](https://github.com/user-attachments/files/20103449/message.txt)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2025-05-08:
Can you try to update to latest snapshot ?

Bastiiiiixx, 2025-05-08:
we have v0.62-6.6 on it which version do you mean ? @Ysurac

Ysurac, 2025-05-09:
This one if on x86_64: https://snapshots.openmptcprouter.com/6.6/x86_64/targets/x86/64/

lhc1993, 2025-05-09:
> Can you try to update to latest snapshot ?

Could you also help clarify the root cause and specifically what was fixed in the new version to prevent this issue from occurring? I'd like to learn from it. Thank you.

Ysurac, 2025-05-09:
I don't say it's fixed, only this need to be tested.
Changelog is available here: https://github.com/Ysurac/openmptcprouter-feeds/commits/develop/

closed 2025-08-12 by github-actions[bot]