Ysurac/openmptcprouter #3941: tun0 is dropped every 10 ~ 15 minutes

moussa11, 2025-05-12:
## Expected Behavior
Need support to troubleshoot what is going on, seems tun0 is dropped every 10~15 minutes. The below messages repeat. 

Nano R5S
eth0 = lan
eth1 = wan1 -> ZyXEL 4G LTE7490-m904
eth2 = wan2 -> DSL modem (bridge mode) PPPOE

[image]

## Current Behavior

Mon May 12 09:07:04 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 12 09:07:09 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 12 09:07:10 2025 daemon.err openvpn(omr)[13408]: event_wait : Interrupted system call (fd=-1,code=4)
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[13408]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[13408]: net_addr_v4_del: 10.255.252.2 dev tun0
Mon May 12 09:07:10 2025 daemon.notice ttyd[30902]: [2025/05/12 09:07:10:1125] N: rops_handle_POLLIN_netlink: DELADDR
Mon May 12 09:07:10 2025 daemon.notice ttyd[30902]: [2025/05/12 09:07:10:1142] N: rops_handle_POLLIN_netlink: DELADDR
Mon May 12 09:07:10 2025 user.notice NET: hotplug (iface): action='remove' interface='tun0'
Mon May 12 09:07:10 2025 daemon.info ModemManager[26575]: hotplug: remove network interface tun0: event processed
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[13408]: /usr/libexec/openvpn-hotplug down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[13408]: SIGTERM[hard,] received, process exiting
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: OpenVPN 2.6.13 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: library versions: OpenSSL 3.0.15 3 Sep 2024, LZO 2.10
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: DCO version: N/A
Mon May 12 09:07:10 2025 daemon.warn openvpn(omr)[26680]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mon May 12 09:07:10 2025 daemon.warn openvpn(omr)[26680]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: TCP/UDP: Preserving recently used remote address: [AF_INET]3.xxx.xxx.4:65301
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: Attempting to establish TCP connection with [AF_INET]3.xxx.xxx.4:65301
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: TCP connection established with [AF_INET]3.xxx.xxx.4:65301
Mon May 12 09:07:10 2025 daemon.warn openvpn(omr)[26680]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Not supported (errno=95)
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: TCPv4_CLIENT link local: (not bound)
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: TCPv4_CLIENT link remote: [AF_INET]3.xxx.xxx.4:65301
Mon May 12 09:07:10 2025 daemon.notice netifd: Interface 'omrvpn' is disabled
Mon May 12 09:07:10 2025 daemon.notice netifd: Network device 'tun0' link is down
Mon May 12 09:07:10 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Mon May 12 09:07:10 2025 daemon.notice netifd: omrvpn (13679): udhcpc: read error: Network is down, reopening socket
Mon May 12 09:07:10 2025 daemon.notice netifd: omrvpn (13679): udhcpc: bind: No such device
Mon May 12 09:07:10 2025 daemon.notice netifd: Interface 'omrvpn' is now down
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: [server] Peer Connection Initiated with [AF_INET]3.xxx.xxx.4:65301
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: TUN/TAP device tun0 opened
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: net_iface_mtu_set: mtu 1420 for tun0
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: net_iface_up: set tun0 up
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: net_addr_v4_add: 10.255.252.2/24 dev tun0
Mon May 12 09:07:10 2025 daemon.notice netifd: Interface 'omrvpn' is enabled
Mon May 12 09:07:10 2025 daemon.notice netifd: Network device 'tun0' link is up
Mon May 12 09:07:10 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Mon May 12 09:07:10 2025 daemon.notice netifd: Interface 'omrvpn' is setting up now
Mon May 12 09:07:10 2025 daemon.notice openvpn(omr)[26680]: /usr/libexec/openvpn-hotplug up omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Mon May 12 09:07:10 2025 daemon.notice netifd: omrvpn (26891): udhcpc: started, v1.36.1
Mon May 12 09:07:10 2025 user.notice NET: hotplug (iface): action='add' interface='tun0'
Mon May 12 09:07:11 2025 daemon.notice netifd: omrvpn (26891): udhcpc: broadcasting discover
Mon May 12 09:07:11 2025 daemon.info ModemManager[26849]: hotplug: add network interface tun0: event processed
Mon May 12 09:07:11 2025 daemon.info unbound: [14103:0] info: service stopped (unbound 1.21.0).
Mon May 12 09:07:11 2025 daemon.info unbound: [14103:0] info: server stats for thread 0: 0 queries, 0 answers from cache, 0 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Mon May 12 09:07:11 2025 daemon.info unbound: [14103:0] info: server stats for thread 0: requestlist max 0 avg 0 exceeded 0 jostled 0
Mon May 12 09:07:11 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Mon May 12 09:07:12 2025 daemon.notice unbound: [27251:0] notice: init module 0: respip
Mon May 12 09:07:12 2025 daemon.notice unbound: [27251:0] notice: init module 1: validator
Mon May 12 09:07:12 2025 daemon.notice unbound: [27251:0] notice: init module 2: iterator
Mon May 12 09:07:12 2025 daemon.info unbound: [27251:0] info: start of service (unbound 1.21.0).
Mon May 12 09:07:12 2025 daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Mon May 12 09:07:12 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Mon May 12 09:07:12 2025 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Mon May 12 09:07:12 2025 daemon.warn dnsmasq[1]: warning: interface omrvpn does not currently exist
Mon May 12 09:07:12 2025 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Mon May 12 09:07:12 2025 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5355
Mon May 12 09:07:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Mon May 12 09:07:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
[... 18 lines cut ...]
Mon May 12 09:13:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: sending renew to server 10.xxx.xxx.173
Mon May 12 09:13:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: lease of 10.xxx.xxx.174 obtained from 10.xxx.xxx.173, lease time 1200
Mon May 12 09:23:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: sending renew to server 10.xxx.xxx.173
Mon May 12 09:23:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: lease of 10.xxx.xxx.174 obtained from 10.xxx.xxx.173, lease time 1200
Mon May 12 09:32:23 2025 user.notice post-tracking-020-status: Check API configuration...
Mon May 12 09:32:23 2025 user.notice post-tracking-020-status: Check API configuration... Done
Mon May 12 09:33:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: sending renew to server 10.xxx.xxx.173
Mon May 12 09:33:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: lease of 10.xxx.xxx.174 obtained from 10.xxx.xxx.173, lease time 1200
Mon May 12 09:43:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: sending renew to server 10.xxx.xxx.173
Mon May 12 09:43:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: lease of 10.xxx.xxx.174 obtained from 10.xxx.xxx.173, lease time 1200
Mon May 12 09:53:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: sending renew to server 10.xxx.xxx.173
Mon May 12 09:53:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: lease of 10.xxx.xxx.174 obtained from 10.xxx.xxx.173, lease time 1200
Mon May 12 10:03:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: sending renew to server 10.xxx.xxx.173
Mon May 12 10:03:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: lease of 10.xxx.xxx.174 obtained from 10.xxx.xxx.173, lease time 1200
Mon May 12 10:13:03 2025 daemon.notice netifd: wan1 (6006): udhcpc: sending renew to server 10.xxx.xxx.173
[... cut ...] ase time 1200
Mon May 12 10:23:04 2025 daemon.notice netifd: wan1 (6006): udhcpc: sending renew to server 10.xxx.xxx.173
Mon May 12 10:23:04 2025 daemon.notice netifd: wan1 (6006): udhcpc: lease of 10.xxx.xxx.174 obtained from 10.xxx.xxx.173, lease time 1200

## Specifications

  - OpenMPTCProuter version: v0.62-6.6 r0+28431-92e020b50f
  - OpenMPTCProuter VPS version: 6.12.15-x64v3-xanmod1 0.1032
  - OpenMPTCProuter VPS provider: AWS
  - OpenMPTCProuter platform: FriendlyARM NanoPi R5S (64-bits)

Ysurac, 2025-05-12:
Can I get more log ? A bit above the "event_wait : Interrupted system call" error

moussa11, 2025-05-13:
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: DCO version: N/A
Mon May 12 18:12:10 2025 daemon.warn openvpn(omr)[25131]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mon May 12 18:12:10 2025 daemon.warn openvpn(omr)[25131]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: TCP/UDP: Preserving recently used remote address: [AF_INET]3.xxx.xxx.4:65301
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: Attempting to establish TCP connection with [AF_INET]3.xxx.xxx.4:65301
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: TCP connection established with [AF_INET]3.xxx.xxx.4:65301
Mon May 12 18:12:10 2025 daemon.warn openvpn(omr)[25131]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Not supported (errno=95)
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: TCPv4_CLIENT link local: (not bound)
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: TCPv4_CLIENT link remote: [AF_INET]3.xxx.xxx.4:65301
Mon May 12 18:12:10 2025 daemon.notice netifd: Interface 'omrvpn' is disabled
Mon May 12 18:12:10 2025 daemon.notice netifd: Network device 'tun0' link is down
Mon May 12 18:12:10 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Mon May 12 18:12:10 2025 daemon.notice netifd: omrvpn (7092): udhcpc: read error: Network is down, reopening socket
Mon May 12 18:12:10 2025 daemon.notice netifd: omrvpn (7092): udhcpc: bind: No such device
Mon May 12 18:12:10 2025 daemon.notice netifd: Interface 'omrvpn' is now down
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: [server] Peer Connection Initiated with [AF_INET]3.xxx.xxx.4:65301
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: TUN/TAP device tun0 opened
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: net_iface_mtu_set: mtu 1420 for tun0
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: net_iface_up: set tun0 up
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: net_addr_v4_add: 10.255.252.2/24 dev tun0
Mon May 12 18:12:10 2025 daemon.notice openvpn(omr)[25131]: /usr/libexec/openvpn-hotplug up omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Mon May 12 18:12:10 2025 daemon.notice netifd: Interface 'omrvpn' is enabled
Mon May 12 18:12:10 2025 daemon.notice netifd: Network device 'tun0' link is up
Mon May 12 18:12:10 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Mon May 12 18:12:10 2025 daemon.notice netifd: Interface 'omrvpn' is setting up now
Mon May 12 18:12:10 2025 daemon.notice netifd: omrvpn (25360): udhcpc: started, v1.36.1
Mon May 12 18:12:10 2025 user.notice NET: hotplug (iface): action='add' interface='tun0'
Mon May 12 18:12:10 2025 daemon.notice netifd: omrvpn (25360): udhcpc: broadcasting discover
Mon May 12 18:12:10 2025 daemon.info ModemManager[25315]: hotplug: add network interface tun0: event processed
Mon May 12 18:12:11 2025 daemon.info unbound: [7545:0] info: service stopped (unbound 1.21.0).
Mon May 12 18:12:11 2025 daemon.info unbound: [7545:0] info: server stats for thread 0: 0 queries, 0 answers from cache, 0 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Mon May 12 18:12:11 2025 daemon.info unbound: [7545:0] info: server stats for thread 0: requestlist max 0 avg 0 exceeded 0 jostled 0
Mon May 12 18:12:11 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Mon May 12 18:12:12 2025 daemon.notice unbound: [25787:0] notice: init module 0: respip
Mon May 12 18:12:12 2025 daemon.notice unbound: [25787:0] notice: init module 1: validator
Mon May 12 18:12:12 2025 daemon.notice unbound: [25787:0] notice: init module 2: iterator
Mon May 12 18:12:12 2025 daemon.info unbound: [25787:0] info: start of service (unbound 1.21.0).
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Mon May 12 18:12:12 2025 daemon.warn dnsmasq[1]: warning: interface omrvpn does not currently exist
Mon May 12 18:12:12 2025 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5355
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon May 12 18:12:12 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
[... cut ...]
Tue May 13 03:16:31 2025 daemon.info unbound: [28303:0] info: generate keytag query _ta-4f66-9728. NULL IN
Tue May 13 03:23:16 2025 daemon.notice netifd: wan1 (6006): udhcpc: sending renew to server 10.xxx.xxx.173
Tue May 13 03:23:16 2025 daemon.notice netifd: wan1 (6006): udhcpc: lease of 10.xxx.xxx.174 obtained from 10.xxx.xxx.173, lease time 1200
Tue May 13 03:27:06 2025 daemon.err uhttpd[5145]: [info] luci: accepted login on /admin/system/openmptcprouter/status for root from 192.168.100.134

moussa11, 2025-05-13:
[image]

AWS VPS port configuration

Ysurac, 2025-05-13:
What is the result, on the VPS of ```journalctl -y openvpn@tun0``` ?

moussa11, 2025-05-13:
The below message repeated 10k's times:

May 12 07:09:00 ip-172-26-11-214 systemd-journald[221]: [🡕] Suppressed 101863 messages from openvpn@tun0.service
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)
May 12 07:09:00 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/92.62.167.9:43301 read TCPv6_SERVER []: Transport endpoint is not connected (fd=11,code=107)

moussa11, 2025-05-13:
disabled IPv6 on AWS VPS, and used command `journalctl -u openvpn@tun0 --since "5 min ago"`

May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:49719 Connection reset, restarting [0]
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:49719 SIGUSR1[soft,connection-reset] received, client-instance restarting
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: WARNING: Failed to stat CRL file, not reloading CRL.
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: WARNING: Failed to stat CRL file, not reloading CRL.
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: TCP connection established with [AF_INET6]::ffff:91.xxx.xxx.184:53446
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: Socket flags: TCP_NODELAY=1 succeeded
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Operation not supported (errno=95)
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: Note: enable extended error passing on TCP/UDP socket failed (IPV6_RECVERR): Operation not supported (errno=95)
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 TLS: Initial packet from [AF_INET6]::ffff:91.xxx.xxx.184:53446, sid=ed1e3fbf 2fb61b83
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 VERIFY OK: depth=1, CN=Easy-RSA CA
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 VERIFY OK: depth=0, CN=openmptcprouter
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 peer info: IV_VER=2.6.13
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 peer info: IV_PLAT=linux
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 peer info: IV_TCPNL=1
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 peer info: IV_MTU=1600
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 peer info: IV_NCP=2
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 peer info: IV_CIPHERS=AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 peer info: IV_PROTO=990
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 peer info: IV_LZO_STUB=1
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 peer info: IV_COMP_STUB=1
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 peer info: IV_COMP_STUBv2=1
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 TLS: move_session: dest=TM_ACTIVE src=TM_INITIAL reinit_src=1
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 WARNING: Failed to stat CRL file, not reloading CRL.
May 13 13:50:09 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 TLS: tls_multi_process: initial untrusted session promoted to trusted
May 13 13:50:10 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256_GCM_SHA384, peer certificate: 2048 bit RSA, signature: RSA-SHA256
May 13 13:50:10 ip-172-26-11-214 ovpn-tun0[379]: 91.xxx.xxx.184:53446 [openmptcprouter] Peer Connection Initiated with [AF_INET6]::ffff:91.xxx.xxx.184:53446
May 13 13:50:10 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:53446 MULTI_sva: pool returned IPv4=10.255.252.2, IPv6=(Not enabled)
May 13 13:50:10 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:53446 Could not access file 'ccd/openmptcprouter': Permission denied (errno=13)
May 13 13:50:10 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:53446 Could not access file 'ccd/DEFAULT': Permission denied (errno=13)
May 13 13:50:10 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:53446 MULTI: Learn: 10.255.252.2 -> openmptcprouter/91.xxx.xxx.184:53446
May 13 13:50:10 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:53446 MULTI: primary virtual IP for openmptcprouter/91.xxx.xxx.184:53446: 10.255.252.2
May 13 13:50:10 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:53446 SENT CONTROL [openmptcprouter]: 'PUSH_REPLY,sndbuf 262144,rcvbuf 262144,route-gateway 10.255.252.1,topology subnet,ping 10,pi>
May 13 13:50:11 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:53446 Data Channel: cipher 'AES-256-GCM', peer-id: 0
May 13 13:50:11 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:53446 Timers: ping 10, ping-restart 480
May 13 13:50:11 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:53446 Protocol options: protocol-flags cc-exit tls-ekm dyn-tls-crypt
May 13 13:50:11 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:53446 Connection reset, restarting [0]
May 13 13:50:11 ip-172-26-11-214 ovpn-tun0[379]: openmptcprouter/91.xxx.xxx.184:53446 SIGUSR1[soft,connection-reset] received, client-instance restarting
May 13 13:50:12 ip-172-26-11-214 ovpn-tun0[379]: WARNING: Failed to stat CRL file, not reloading CRL.
May 13 13:50:12 ip-172-26-11-214 ovpn-tun0[379]: WARNING: Failed to stat CRL file, not reloading CRL.
May 13 13:50:12 ip-172-26-11-214 ovpn-tun0[379]: TCP connection established with [AF_INET6]::ffff:91.xxx.xxx.184:34245
May 13 13:50:12 ip-172-26-11-214 ovpn-tun0[379]: Socket flags: TCP_NODELAY=1 succeeded
May 13 13:50:12 ip-172-26-11-214 ovpn-tun0[379]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Operation not supported (errno=95)
[... cut ...]
May 13 13:54:32 ip-172-26-11-214 ovpn-tun0[379]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:59625
May 13 13:54:32 ip-172-26-11-214 ovpn-tun0[379]: MANAGEMENT: CMD 'status'
May 13 13:54:32 ip-172-26-11-214 ovpn-tun0[379]: MANAGEMENT: Client disconnected
May 13 13:54:33 ip-172-26-11-214 ovpn-tun0[379]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:59635
May 13 13:54:33 ip-172-26-11-214 ovpn-tun0[379]: MANAGEMENT: CMD 'status'
May 13 13:54:33 ip-172-26-11-214 ovpn-tun0[379]: MANAGEMENT: Client disconnected

Ysurac, 2025-05-13:
Can you check with latest snapshot ? https://snapshots.openmptcprouter.com/6.6/
Seems to be after a firewall reload and I changed some part on this, maybe this is already solved...

Ysurac, 2025-05-14:
Can I get log from router too ?
(Would be better to fix, but if you just want router to work, you can use another VPN in System->OpenMPTCProuter, wizard tab, adcanced settings checkbox)

moussa11, 2025-05-14:
Happy to fix. Its a remote setup, won't have easy physical access if it bricks. 

Wed May 14 02:59:45 2025 daemon.notice netifd: Interface 'omrvpn' is now down
Wed May 14 02:59:46 2025 daemon.notice openvpn(omr)[17093]: [server] Peer Connection Initiated with [AF_INET]3.xxx.xxx.4:65301
Wed May 14 02:59:46 2025 daemon.notice openvpn(omr)[17093]: TUN/TAP device tun0 opened
Wed May 14 02:59:46 2025 daemon.notice openvpn(omr)[17093]: net_iface_mtu_set: mtu 1420 for tun0
Wed May 14 02:59:46 2025 daemon.notice openvpn(omr)[17093]: net_iface_up: set tun0 up
Wed May 14 02:59:46 2025 daemon.notice openvpn(omr)[17093]: net_addr_v4_add: 10.255.252.2/24 dev tun0
Wed May 14 02:59:46 2025 daemon.notice openvpn(omr)[17093]: /usr/libexec/openvpn-hotplug up omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Wed May 14 02:59:46 2025 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed May 14 02:59:46 2025 daemon.notice netifd: Network device 'tun0' link is up
Wed May 14 02:59:46 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed May 14 02:59:46 2025 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed May 14 02:59:46 2025 daemon.notice netifd: omrvpn (17740): udhcpc: started, v1.36.1
Wed May 14 02:59:46 2025 user.notice NET: hotplug (iface): action='add' interface='tun0'
Wed May 14 02:59:46 2025 daemon.notice netifd: omrvpn (17740): udhcpc: broadcasting discover
Wed May 14 02:59:46 2025 daemon.info ModemManager[717]: hotplug: checking if ModemManager is available...
Wed May 14 02:59:46 2025 daemon.info ModemManager[718]: hotplug: Checking if ModemManager is available...
Wed May 14 02:59:46 2025 daemon.info ModemManager[717]: hotplug: ModemManager not yet available
Wed May 14 02:59:46 2025 daemon.info ModemManager[718]: hotplug: ModemManager not yet available (count=19)
Wed May 14 02:59:46 2025 daemon.info ModemManager[17667]: hotplug: add network interface tun0: event processed
Wed May 14 02:59:47 2025 daemon.info ModemManager[717]: hotplug: checking if ModemManager is available...
Wed May 14 02:59:47 2025 daemon.info ModemManager[718]: hotplug: Checking if ModemManager is available...
Wed May 14 02:59:47 2025 daemon.info ModemManager[718]: hotplug: ModemManager not yet available (count=18)
Wed May 14 02:59:47 2025 daemon.info ModemManager[717]: hotplug: ModemManager not yet available
Wed May 14 02:59:48 2025 daemon.info ModemManager[718]: hotplug: Checking if ModemManager is available...
Wed May 14 02:59:48 2025 daemon.info ModemManager[717]: hotplug: checking if ModemManager is available...
Wed May 14 02:59:48 2025 daemon.info ModemManager[717]: hotplug: ModemManager not yet available
Wed May 14 02:59:48 2025 daemon.info ModemManager[718]: hotplug: ModemManager not yet available (count=17)
Wed May 14 02:59:49 2025 daemon.notice netifd: omrvpn (17740): udhcpc: broadcasting discover
Wed May 14 02:59:49 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was 10.xxx.xxx.98)
Wed May 14 02:59:49 2025 daemon.info ModemManager[718]: hotplug: Checking if ModemManager is available...
Wed May 14 02:59:49 2025 daemon.info ModemManager[717]: hotplug: checking if ModemManager is available...
Wed May 14 02:59:49 2025 daemon.info ModemManager[718]: hotplug: ModemManager not yet available (count=16)
Wed May 14 02:59:49 2025 daemon.info ModemManager[717]: hotplug: ModemManager not yet available
Wed May 14 02:59:49 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Wed May 14 02:59:50 2025 daemon.err openvpn(omr)[17093]: event_wait : Interrupted system call (fd=-1,code=4)
Wed May 14 02:59:50 2025 daemon.notice openvpn(omr)[17093]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Wed May 14 02:59:50 2025 daemon.notice openvpn(omr)[17093]: net_addr_v4_del: 10.255.252.2 dev tun0
Wed May 14 02:59:50 2025 user.notice NET: hotplug (iface): action='remove' interface='tun0'
Wed May 14 02:59:50 2025 daemon.info ModemManager[20741]: hotplug: remove network interface tun0: event processed
Wed May 14 02:59:50 2025 daemon.notice openvpn(omr)[17093]: /usr/libexec/openvpn-hotplug down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Wed May 14 02:59:50 2025 daemon.notice openvpn(omr)[17093]: SIGTERM[hard,] received, process exiting
Wed May 14 02:59:50 2025 daemon.info ModemManager[718]: hotplug: Checking if ModemManager is available...
Wed May 14 02:59:50 2025 daemon.info ModemManager[717]: hotplug: checking if ModemManager is available...
Wed May 14 02:59:50 2025 daemon.info ModemManager[718]: hotplug: ModemManager not yet available (count=15)
Wed May 14 02:59:50 2025 daemon.notice openvpn(omr)[21021]: OpenVPN 2.6.13 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Wed May 14 02:59:50 2025 daemon.notice openvpn(omr)[21021]: library versions: OpenSSL 3.0.16 11 Feb 2025, LZO 2.10
Wed May 14 02:59:50 2025 daemon.notice openvpn(omr)[21021]: DCO version: N/A
Wed May 14 02:59:50 2025 daemon.warn openvpn(omr)[21021]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Wed May 14 02:59:50 2025 daemon.warn openvpn(omr)[21021]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Wed May 14 02:59:50 2025 daemon.info ModemManager[717]: hotplug: ModemManager not yet available
[... cut ...]
Wed May 14 07:29:27 2025 daemon.notice netifd: wan1 (9894): udhcpc: lease of 10.xxx.xxx.97 obtained from 10.xxx.xxx.98, lease time 1200
Wed May 14 07:39:27 2025 daemon.notice netifd: wan1 (9894): udhcpc: sending renew to server 10.xxx.xxx.98
Wed May 14 07:39:27 2025 daemon.notice netifd: wan1 (9894): udhcpc: lease of 10.xxx.xxx.97 obtained from 10.xxx.xxx.98, lease time 1200
Wed May 14 07:40:15 2025 daemon.err uhttpd[2804]: [info] luci: accepted login on /admin/status/overview for root from 192.168.100.134

moussa11, 2025-05-14:
Is MPTCP over VPN settings as Wireguard correct, or should it also be OpenVPN ?

[image]

moussa11, 2025-05-15:
switched to Glorytun VPN, and seems the issue is still there: 

Wed May 14 13:29:21 2025 daemon.err glorytun[25487]: 3.xxx.xxx.4.65001: connected
Wed May 14 13:29:21 2025 daemon.err glorytun[25487]: couldn't set socket option `TCP_USER_TIMEOUT'
Wed May 14 13:29:21 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Wed May 14 13:29:22 2025 user.notice post-tracking-003-up: Change in routes, set ip route replace default scope global nexthop via 10.xxx.xxx.141 dev eth1 weight 100 nexthop via 92.xxx.xxx.254 dev pppoe-wan2 weight 1 (omrvpn_intf: tun0)
Wed May 14 13:29:22 2025 daemon.notice ttyd[7199]: [2025/05/14 13:29:22:3535] N: rops_handle_POLLIN_netlink: DELADDR
Wed May 14 13:29:22 2025 user.notice OMR-Tracker: Reloading OMR-Tracker due to ifup of omrvpn (tun0)
Wed May 14 13:29:31 2025 daemon.err glorytun[25487]: 3.xxx.xxx.4.65001: key exchange failed
Wed May 14 13:29:31 2025 daemon.info glorytun[25487]: STOPPED tun0
Wed May 14 13:29:32 2025 daemon.err glorytun[25487]: couldn't set socket option `MPTCP_ENABLED (42)'
Wed May 14 13:29:32 2025 daemon.err glorytun[25487]: couldn't set socket option `MPTCP_ENABLED (26)'
Wed May 14 13:29:32 2025 daemon.err glorytun[25487]: 3.xxx.xxx.4.65001: connected
Wed May 14 13:29:32 2025 daemon.err glorytun[25487]: couldn't set socket option `TCP_USER_TIMEOUT'
Wed May 14 13:29:34 2025 daemon.info glorytun[25487]: STARTED tun0
Wed May 14 13:29:34 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.255.1 dev tun0 (was 10.xxx.xxx.141)
Wed May 14 13:29:52 2025 daemon.notice netifd: wan1 (31328): udhcpc: sending renew to server 10.xxx.xxx.141
Wed May 14 13:29:52 2025 daemon.notice netifd: wan1 (31328): udhcpc: lease of 10.xxx.xxx.140 obtained from 10.xxx.xxx.141, lease time 1200
Wed May 14 13:33:25 2025 daemon.info glorytun[25487]: STOPPED tun0
Wed May 14 13:33:26 2025 daemon.err glorytun[25487]: couldn't set socket option `MPTCP_ENABLED (42)'
Wed May 14 13:33:26 2025 daemon.err glorytun[25487]: couldn't set socket option `MPTCP_ENABLED (26)'
Wed May 14 13:33:26 2025 daemon.err glorytun[25487]: 3.xxx.xxx.4.65001: connected
Wed May 14 13:33:26 2025 daemon.err glorytun[25487]: couldn't set socket option `TCP_USER_TIMEOUT'
Wed May 14 13:33:36 2025 daemon.err glorytun[25487]: 3.xxx.xxx.4.65001: key exchange failed
Wed May 14 13:33:36 2025 daemon.info glorytun[25487]: STOPPED tun0
Wed May 14 13:33:37 2025 daemon.err glorytun[25487]: couldn't set socket option `MPTCP_ENABLED (42)'
Wed May 14 13:33:37 2025 daemon.err glorytun[25487]: couldn't set socket option `MPTCP_ENABLED (26)'
Wed May 14 13:33:37 2025 daemon.err glorytun[25487]: 3.xxx.xxx.4.65001: connected
Wed May 14 13:33:37 2025 daemon.err glorytun[25487]: couldn't set socket option `TCP_USER_TIMEOUT'
Wed May 14 13:33:38 2025 daemon.info glorytun[25487]: STARTED tun0
Wed May 14 13:36:02 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Wed May 14 13:36:05 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Wed May 14 13:36:06 2025 daemon.err procd: instance omr not found
Wed May 14 13:36:07 2025 daemon.info unbound: [27455:0] info: service stopped (unbound 1.21.0).
Wed May 14 13:36:07 2025 daemon.info unbound: [27455:0] info: server stats for thread 0: 0 queries, 0 answers from cache, 0 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Wed May 14 13:36:07 2025 daemon.info unbound: [27455:0] info: server stats for thread 0: requestlist max 0 avg 0 exceeded 0 jostled 0
Wed May 14 13:36:07 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Wed May 14 13:36:08 2025 daemon.notice unbound: [12600:0] notice: init module 0: respip
Wed May 14 13:36:08 2025 daemon.notice unbound: [12600:0] notice: init module 1: validator
Wed May 14 13:36:08 2025 daemon.notice unbound: [12600:0] notice: init module 2: iterator
Wed May 14 13:36:08 2025 daemon.info unbound: [12600:0] info: start of service (unbound 1.21.0).
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Wed May 14 13:36:08 2025 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Wed May 14 13:36:08 2025 daemon.info dnsmasq-dhcp[1]: DHCP, sockets bound exclusively to interface eth0
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5355
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Wed May 14 13:36:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
[... cut ...]
Thu May 15 03:18:18 2025 daemon.notice netifd: wan1 (31297): udhcpc: lease of 10.xxx.xxx.62 obtained from 10.xxx.xxx.61, lease time 1200
Thu May 15 03:27:18 2025 daemon.err uhttpd[2804]: [info] luci: accepted login on /admin/system/openmptcprouter/status for root from 192.168.100.134
Thu May 15 03:28:18 2025 daemon.notice netifd: wan1 (31297): udhcpc: sending renew to server 10.xxx.xxx.61
Thu May 15 03:28:18 2025 daemon.notice netifd: wan1 (31297): udhcpc: lease of 10.xxx.xxx.62 obtained from 10.xxx.xxx.61, lease time 1200

Ysurac, 2025-05-15:
Seems that glorytun key is not correct. In System->OpenMPTCProuter, "Wizard" tab, "Advenced settings" checkbox, Check "Force retrieve settings" and Save.

moussa11, 2025-05-15:
Seems the same issue remains after "Force retrieve setting":

Thu May 15 07:15:41 2025 user.notice OMR-Tracker: Reloading OMR-Tracker due to ifup of omrvpn (tun0)
Thu May 15 07:15:42 2025 user.notice omr-bypass: Starting OMR-ByPass...
Thu May 15 07:15:45 2025 daemon.info procd: Instance glorytun::instance1 pid 13294 not stopped on SIGTERM, sending SIGKILL instead
Thu May 15 07:15:45 2025 daemon.notice ttyd[7199]: [2025/05/15 07:15:45:4176] N: rops_handle_POLLIN_netlink: DELADDR
Thu May 15 07:15:45 2025 daemon.notice ttyd[7199]: [2025/05/15 07:15:45:4217] N: rops_handle_POLLIN_netlink: DELADDR
Thu May 15 07:15:45 2025 user.notice NET: hotplug (iface): action='remove' interface='tun0'
Thu May 15 07:15:45 2025 daemon.info ModemManager[22455]: hotplug: remove network interface tun0: event processed
Thu May 15 07:15:45 2025 daemon.notice netifd: Interface 'omrvpn' is now down
Thu May 15 07:15:45 2025 daemon.info glorytun[22510]: INITIALIZED tun0
Thu May 15 07:15:45 2025 user.notice NET: hotplug (iface): action='add' interface='tun0'
Thu May 15 07:15:45 2025 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu May 15 07:15:45 2025 daemon.notice netifd: Network device 'tun0' link is down
Thu May 15 07:15:45 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Thu May 15 07:15:45 2025 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu May 15 07:15:45 2025 daemon.notice netifd: Network device 'tun0' link is up
Thu May 15 07:15:45 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Thu May 15 07:15:45 2025 daemon.notice netifd: Interface 'omrvpn' is setting up now
Thu May 15 07:15:45 2025 daemon.notice netifd: Interface 'omrvpn' is now up
Thu May 15 07:15:45 2025 daemon.info ModemManager[22536]: hotplug: add network interface tun0: event processed
Thu May 15 07:15:46 2025 daemon.err glorytun[22510]: couldn't set socket option `MPTCP_ENABLED (42)'
Thu May 15 07:15:46 2025 daemon.err glorytun[22510]: couldn't set socket option `MPTCP_ENABLED (26)'
Thu May 15 07:15:46 2025 daemon.err glorytun[22510]: 3.xxx.xxx.4.65001: connected
Thu May 15 07:15:46 2025 daemon.err glorytun[22510]: couldn't set socket option `TCP_USER_TIMEOUT'
Thu May 15 07:15:46 2025 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Thu May 15 07:15:46 2025 user.notice post-tracking-003-up: Change in routes, set ip route replace default scope global nexthop via 10.xxx.xxx.61 dev eth1 weight 100 nexthop via 92.xxx.xxx.254 dev pppoe-wan2 weight 1 (omrvpn_intf: tun0)
Thu May 15 07:15:47 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu May 15 07:15:48 2025 daemon.notice ttyd[7199]: [2025/05/15 07:15:48:5542] N: rops_handle_POLLIN_netlink: DELADDR
Thu May 15 07:15:48 2025 user.notice OMR-Tracker: Reloading OMR-Tracker due to ifup of omrvpn (tun0)
Thu May 15 07:15:49 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu May 15 07:15:50 2025 user.notice omr-bypass: OMR-ByPass is running
Thu May 15 07:15:54 2025 user.notice cake-autorate.wan1: INFO: 1747293354.338691 Starting cake-autorate with PID: 17753 and config: /root/cake-autorate/config.wan1.sh
Thu May 15 07:15:56 2025 daemon.err glorytun[22510]: 3.xxx.xxx.4.65001: key exchange failed
Thu May 15 07:15:56 2025 daemon.info glorytun[22510]: STOPPED tun0
Thu May 15 07:15:57 2025 daemon.err glorytun[22510]: couldn't set socket option `MPTCP_ENABLED (42)'
Thu May 15 07:15:57 2025 daemon.err glorytun[22510]: couldn't set socket option `MPTCP_ENABLED (26)'
Thu May 15 07:15:57 2025 daemon.err glorytun[22510]: 3.xxx.xxx.4.65001: connected
Thu May 15 07:15:57 2025 daemon.err glorytun[22510]: couldn't set socket option `TCP_USER_TIMEOUT'
Thu May 15 07:16:00 2025 user.notice omr-schedule-010-services: Set firewall on server AWSlondon
Thu May 15 07:16:07 2025 daemon.err glorytun[22510]: 3.xxx.xxx.4.65001: key exchange failed
Thu May 15 07:16:07 2025 daemon.info glorytun[22510]: STOPPED tun0
Thu May 15 07:16:08 2025 daemon.err glorytun[22510]: couldn't set socket option `MPTCP_ENABLED (42)'
Thu May 15 07:16:08 2025 daemon.err glorytun[22510]: couldn't set socket option `MPTCP_ENABLED (26)'
Thu May 15 07:16:08 2025 daemon.err glorytun[22510]: 3.xxx.xxx.4.65001: connected
Thu May 15 07:16:08 2025 daemon.err glorytun[22510]: couldn't set socket option `TCP_USER_TIMEOUT'
Thu May 15 07:16:13 2025 user.notice post-tracking-002-error: omrvpn down because IPv4 gateway down
Thu May 15 07:16:14 2025 user.notice post-tracking-002-error: Glorytun VPN down, restart it
Thu May 15 07:16:14 2025 daemon.info glorytun: starting glorytun vpn instance vpn
Thu May 15 07:16:14 2025 user.notice post-tracking-002-error: DNS flush
Thu May 15 07:16:16 2025 user.notice post-tracking-003-up: Change in routes, set ip route replace default scope global nexthop via 10.xxx.xxx.61 dev eth1 weight 100 nexthop via 92.xxx.xxx.254 dev pppoe-wan2 weight 1 (omrvpn_intf: tun0)
Thu May 15 07:16:17 2025 daemon.info glorytun[22510]: STARTED tun0
Thu May 15 07:16:17 2025 daemon.info glorytun[22510]: STOPPED tun0
Thu May 15 07:16:17 2025 daemon.notice netifd: Network device 'tun0' link is down
Thu May 15 07:16:17 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Thu May 15 07:16:17 2025 daemon.notice netifd: Interface 'omrvpn' is now down
Thu May 15 07:16:17 2025 daemon.notice ttyd[7199]: [2025/05/15 07:16:17:4107] N: rops_handle_POLLIN_netlink: DELADDR
[... cut ...]
Thu May 15 07:17:15 2025 user.notice post-tracking-022-speedtest: Calculated wan1 upload speed: 31206
Thu May 15 07:18:21 2025 daemon.notice netifd: wan1 (31297): udhcpc: sending renew to server 10.xxx.xxx.61
Thu May 15 07:18:21 2025 daemon.notice netifd: wan1 (31297): udhcpc: lease of 10.xxx.xxx.62 obtained from 10.xxx.xxx.61, lease time 1200
Thu May 15 07:28:21 2025 daemon.notice netifd: wan1 (31297): udhcpc: sending renew to server 10.xxx.xxx.61
Thu May 15 07:28:21 2025 daemon.notice netifd: wan1 (31297): udhcpc: lease of 10.xxx.xxx.62 obtained from 10.xxx.xxx.61, lease time 1200

Ysurac, 2025-05-15:
Can you check if the key on the router via SSH ```uci show glorytun.key``` is the same on the VPS in ```/etc/glorytun-tcp/tun0.key``` ?
Can you also check the result, on the VPS, of ```journalctl -u omr-admin``` ?

moussa11, 2025-05-15:
key on the VPS is matching with key on router GUI. router via SSH `uci show glorytun.key` is giving `uci: Entry not found`

May 15 07:57:11 ip-172-26-11-214 omr-admin.py[1967]: INFO:     127.0.0.1:39355 - "GET / HTTP/1.1" 200 OK
May 15 07:57:15 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:52523 - "GET / HTTP/1.1" 200 OK
May 15 07:57:22 ip-172-26-11-214 omr-admin.py[1969]: INFO:     91.xxx.xxx.41:57165 - "GET / HTTP/1.1" 200 OK
May 15 07:57:25 ip-172-26-11-214 omr-admin.py[1967]: INFO:     127.0.0.1:46741 - "GET / HTTP/1.1" 200 OK
May 15 07:57:29 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:41473 - "GET / HTTP/1.1" 200 OK
May 15 07:57:36 ip-172-26-11-214 omr-admin.py[1969]: INFO:     91.xxx.xxx.41:41934 - "GET / HTTP/1.1" 200 OK
May 15 07:57:39 ip-172-26-11-214 omr-admin.py[1970]: INFO:     127.0.0.1:53971 - "GET / HTTP/1.1" 200 OK
May 15 07:57:43 ip-172-26-11-214 omr-admin.py[1968]: INFO:     91.xxx.xxx.41:33525 - "GET / HTTP/1.1" 200 OK
May 15 07:57:50 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:59258 - "GET / HTTP/1.1" 200 OK
May 15 07:57:53 ip-172-26-11-214 omr-admin.py[1967]: INFO:     127.0.0.1:34459 - "GET / HTTP/1.1" 200 OK
May 15 07:57:57 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:41955 - "GET / HTTP/1.1" 200 OK
May 15 07:58:04 ip-172-26-11-214 omr-admin.py[1969]: INFO:     91.xxx.xxx.41:44144 - "GET / HTTP/1.1" 200 OK
May 15 07:58:07 ip-172-26-11-214 omr-admin.py[1967]: INFO:     127.0.0.1:14895 - "GET / HTTP/1.1" 200 OK
May 15 07:58:11 ip-172-26-11-214 omr-admin.py[1969]: INFO:     91.xxx.xxx.41:48158 - "GET / HTTP/1.1" 200 OK
May 15 07:58:18 ip-172-26-11-214 omr-admin.py[1969]: INFO:     91.xxx.xxx.41:43386 - "GET / HTTP/1.1" 200 OK
May 15 07:58:25 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:58335 - "GET / HTTP/1.1" 200 OK
May 15 07:58:29 ip-172-26-11-214 omr-admin.py[1967]: INFO:     127.0.0.1:57999 - "GET / HTTP/1.1" 200 OK
May 15 07:58:32 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:39117 - "GET / HTTP/1.1" 200 OK
May 15 07:58:39 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:63286 - "GET / HTTP/1.1" 200 OK
May 15 07:58:42 ip-172-26-11-214 omr-admin.py[1969]: INFO:     92.xxx.xxx.89:48333 - "POST /token HTTP/1.1" 200 OK
May 15 07:58:42 ip-172-26-11-214 omr-admin.py[1967]: INFO:     92.xxx.xxx.89:48341 - "GET /config HTTP/1.1" 200 OK
May 15 07:58:44 ip-172-26-11-214 omr-admin.py[1968]: INFO:     92.xxx.xxx.89:48347 - "POST /token HTTP/1.1" 200 OK
May 15 07:58:44 ip-172-26-11-214 omr-admin.py[1967]: INFO:     127.0.0.1:44523 - "GET / HTTP/1.1" 200 OK
May 15 07:58:44 ip-172-26-11-214 omr-admin.py[1967]: INFO:     92.xxx.xxx.89:48353 - "POST /shorewalllist HTTP/1.1" 200 OK
May 15 07:58:45 ip-172-26-11-214 omr-admin.py[1969]: INFO:     92.xxx.xxx.89:48369 - "POST /shorewalllist HTTP/1.1" 200 OK
May 15 07:58:45 ip-172-26-11-214 omr-admin.py[1967]: INFO:     92.xxx.xxx.89:48373 - "POST /shorewalllist HTTP/1.1" 200 OK
May 15 07:58:46 ip-172-26-11-214 omr-admin.py[1967]: INFO:     92.xxx.xxx.89:48379 - "POST /shorewalllist HTTP/1.1" 200 OK
May 15 07:58:46 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:33674 - "GET / HTTP/1.1" 200 OK
May 15 07:58:46 ip-172-26-11-214 omr-admin.py[1967]: INFO:     92.xxx.xxx.89:48395 - "POST /token HTTP/1.1" 200 OK
May 15 07:58:47 ip-172-26-11-214 omr-admin.py[1967]: INFO:     92.xxx.xxx.89:48411 - "GET /backuplist HTTP/1.1" 200 OK
May 15 07:58:47 ip-172-26-11-214 omr-admin.py[1969]: INFO:     92.xxx.xxx.89:48413 - "GET /config HTTP/1.1" 200 OK
May 15 07:58:48 ip-172-26-11-214 omr-admin.py[1969]: INFO:     92.xxx.xxx.89:48415 - "GET /config HTTP/1.1" 200 OK
May 15 07:58:49 ip-172-26-11-214 omr-admin.py[1969]: INFO:     92.xxx.xxx.89:48429 - "GET /config HTTP/1.1" 200 OK
May 15 07:58:49 ip-172-26-11-214 omr-admin.py[1969]: INFO:     92.xxx.xxx.89:48435 - "GET /config HTTP/1.1" 200 OK
May 15 07:58:50 ip-172-26-11-214 omr-admin.py[1969]: INFO:     92.xxx.xxx.89:48447 - "GET /config HTTP/1.1" 200 OK
May 15 07:58:50 ip-172-26-11-214 omr-admin.py[1969]: INFO:     92.xxx.xxx.89:38301 - "POST /vpnips HTTP/1.1" 422 Unprocessable Entity
May 15 07:58:51 ip-172-26-11-214 omr-admin.py[1969]: INFO:     92.xxx.xxx.89:38307 - "POST /sipalg HTTP/1.1" 200 OK
May 15 07:58:53 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:42313 - "GET / HTTP/1.1" 200 OK
May 15 07:58:58 ip-172-26-11-214 omr-admin.py[1967]: INFO:     127.0.0.1:33481 - "GET / HTTP/1.1" 200 OK
May 15 07:59:00 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:50960 - "GET / HTTP/1.1" 200 OK
May 15 07:59:07 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:34368 - "GET / HTTP/1.1" 200 OK
May 15 07:59:12 ip-172-26-11-214 omr-admin.py[1969]: INFO:     127.0.0.1:10309 - "GET / HTTP/1.1" 200 OK
May 15 07:59:14 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:36547 - "GET / HTTP/1.1" 200 OK
May 15 07:59:21 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:54734 - "GET / HTTP/1.1" 200 OK
May 15 07:59:26 ip-172-26-11-214 omr-admin.py[1967]: INFO:     127.0.0.1:13621 - "GET / HTTP/1.1" 200 OK
May 15 07:59:28 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:38837 - "GET / HTTP/1.1" 200 OK
May 15 07:59:35 ip-172-26-11-214 omr-admin.py[1969]: INFO:     91.xxx.xxx.41:37337 - "GET / HTTP/1.1" 200 OK
May 15 07:59:40 ip-172-26-11-214 omr-admin.py[1967]: INFO:     127.0.0.1:34701 - "GET / HTTP/1.1" 200 OK
[... cut ...]
May 15 08:01:49 ip-172-26-11-214 omr-admin.py[1969]: INFO:     91.xxx.xxx.41:35743 - "GET / HTTP/1.1" 200 OK
May 15 08:01:56 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:55912 - "GET / HTTP/1.1" 200 OK
May 15 08:01:59 ip-172-26-11-214 omr-admin.py[1969]: INFO:     127.0.0.1:62833 - "GET / HTTP/1.1" 200 OK
May 15 08:02:03 ip-172-26-11-214 omr-admin.py[1969]: INFO:     91.xxx.xxx.41:51088 - "GET / HTTP/1.1" 200 OK
May 15 08:02:10 ip-172-26-11-214 omr-admin.py[1967]: INFO:     91.xxx.xxx.41:40318 - "GET / HTTP/1.1" 200 OK

moussa11, 2025-05-16:
Switched to "A Dead Simple VPN": seems a. bit better, there are still some errors. Can you please explain what is going on, and what is happening. Is it cause of R5S hardware or AWS VPS or the connections ? Thanks

One more thing, there is a significant increase on the data traffic consumption, 100 GB 4G package used to last 3 ~ 4 weeks, now its about 1 week. Is this normal ? 

On the other hand I don't see significant pickup on data consumption for DSL slower connection.

Fri May 16 00:05:02 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Fri May 16 00:05:05 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Fri May 16 00:05:05 2025 daemon.err procd: instance omr not found
Fri May 16 00:05:07 2025 daemon.info unbound: [32241:0] info: service stopped (unbound 1.21.0).
Fri May 16 00:05:07 2025 daemon.info unbound: [32241:0] info: server stats for thread 0: 0 queries, 0 answers from cache, 0 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Fri May 16 00:05:07 2025 daemon.info unbound: [32241:0] info: server stats for thread 0: requestlist max 0 avg 0 exceeded 0 jostled 0
Fri May 16 00:05:07 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Fri May 16 00:05:07 2025 daemon.notice unbound: [13595:0] notice: init module 0: respip
Fri May 16 00:05:07 2025 daemon.notice unbound: [13595:0] notice: init module 1: validator
Fri May 16 00:05:07 2025 daemon.notice unbound: [13595:0] notice: init module 2: iterator
Fri May 16 00:05:07 2025 daemon.info unbound: [13595:0] info: start of service (unbound 1.21.0).
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Fri May 16 00:05:08 2025 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Fri May 16 00:05:08 2025 daemon.info dnsmasq-dhcp[1]: DHCP, sockets bound exclusively to interface eth0
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5355
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri May 16 00:05:08 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri May 16 00:05:08 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri May 16 00:06:00 2025 user.notice omr-schedule-010-services: Set firewall on server AWSlondon
Fri May 16 00:09:26 2025 daemon.notice netifd: wan1 (5991): udhcpc: sending renew to server 10.xxx.xxx.178
Fri May 16 00:09:26 2025 daemon.notice netifd: wan1 (5991): udhcpc: lease of 10.xxx.xxx.177 obtained from 10.xxx.xxx.178, lease time 1200
Fri May 16 00:19:26 2025 daemon.notice netifd: wan1 (5991): udhcpc: sending renew to server 10.xxx.xxx.178
Fri May 16 00:19:26 2025 daemon.notice netifd: wan1 (5991): udhcpc: lease of 10.xxx.xxx.177 obtained from 10.xxx.xxx.178, lease time 1200
Fri May 16 00:29:26 2025 daemon.notice netifd: wan1 (5991): udhcpc: sending renew to server 10.xxx.xxx.178
Fri May 16 00:29:27 2025 daemon.notice netifd: wan1 (5991): udhcpc: lease of 10.xxx.xxx.177 obtained from 10.xxx.xxx.178, lease time 1200
Fri May 16 00:39:27 2025 daemon.notice netifd: wan1 (5991): udhcpc: sending renew to server 10.xxx.xxx.178
Fri May 16 00:39:27 2025 daemon.notice netifd: wan1 (5991): udhcpc: lease of 10.xxx.xxx.177 obtained from 10.xxx.xxx.178, lease time 1200
Fri May 16 00:43:02 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Fri May 16 00:43:05 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Fri May 16 00:43:06 2025 daemon.err procd: instance omr not found
Fri May 16 00:43:07 2025 daemon.info unbound: [13595:0] info: service stopped (unbound 1.21.0).
Fri May 16 00:43:07 2025 daemon.info unbound: [13595:0] info: server stats for thread 0: 0 queries, 0 answers from cache, 0 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Fri May 16 00:43:07 2025 daemon.info unbound: [13595:0] info: server stats for thread 0: requestlist max 0 avg 0 exceeded 0 jostled 0
Fri May 16 00:43:07 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Fri May 16 00:43:08 2025 daemon.notice unbound: [8482:0] notice: init module 0: respip
Fri May 16 00:43:08 2025 daemon.notice unbound: [8482:0] notice: init module 1: validator
Fri May 16 00:43:08 2025 daemon.notice unbound: [8482:0] notice: init module 2: iterator
Fri May 16 00:43:08 2025 daemon.info unbound: [8482:0] info: start of service (unbound 1.21.0).
[... cut ...]
Fri May 16 02:49:29 2025 daemon.notice netifd: wan1 (5991): udhcpc: lease of 10.xxx.xxx.177 obtained from 10.xxx.xxx.178, lease time 1200
Fri May 16 02:59:29 2025 daemon.notice netifd: wan1 (5991): udhcpc: sending renew to server 10.xxx.xxx.178
Fri May 16 02:59:29 2025 daemon.notice netifd: wan1 (5991): udhcpc: lease of 10.xxx.xxx.177 obtained from 10.xxx.xxx.178, lease time 1200
Fri May 16 03:04:41 2025 daemon.err uhttpd[5054]: [info] luci: accepted login on /admin/status/vnstat2 for root from 192.168.100.134

Ysurac, 2025-05-16:
For the consumption, this can be due to the multiple disconnection/connection that is sending data each times. There is also OMR-Tracher that is testing all connections, but it's the same on all interfaces.
For the problem I would say it's VPS if there is nothing in ```dmesg``` on the router.
I will try on AWS when I will have some free times.

moussa11, 2025-05-16:
mentioned [Amazon AWS](https://aws.amazon.com/) port range **0-65535** in UDP, TCP and ICMP must be opened (US, ZA, HK, CA, UK, BR, BH, IT, IE, DE, JP,...)
 While : 

[image]

I have opened all ports to all protocols to see whether this makes any difference. 

Whichh service provider you have experience with in UK. I can try moving the VPS.

Ysurac, 2025-05-16:
Currently that are available in UK, I have one at DigitalOcean, Vultr, OVH and Ionos.

moussa11, 2025-05-18:
switch over to Digital Ocean VPS, seems the issue is still there !

VPS logs: 

May 18 13:12:29 OMRVPS ovpn-tun0[480]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:10023
May 18 13:12:29 OMRVPS ovpn-tun0[480]: MANAGEMENT: CMD 'status'
May 18 13:12:29 OMRVPS ovpn-tun0[480]: MANAGEMENT: Client disconnected
May 18 13:12:50 OMRVPS ovpn-tun0[480]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:13561
May 18 13:12:50 OMRVPS ovpn-tun0[480]: MANAGEMENT: CMD 'status'
May 18 13:12:50 OMRVPS ovpn-tun0[480]: MANAGEMENT: Client disconnected
May 18 13:13:09 OMRVPS ovpn-tun0[480]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:45657
May 18 13:13:09 OMRVPS ovpn-tun0[480]: MANAGEMENT: CMD 'status'
May 18 13:13:09 OMRVPS ovpn-tun0[480]: MANAGEMENT: Client disconnected
May 18 13:16:46 OMRVPS ovpn-tun0[480]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:52285
May 18 13:16:46 OMRVPS ovpn-tun0[480]: MANAGEMENT: CMD 'status'
May 18 13:16:46 OMRVPS ovpn-tun0[480]: MANAGEMENT: Client disconnected
May 18 13:17:06 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:57981 Connection reset, restarting [0]
May 18 13:17:06 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:57981 SIGUSR1[soft,connection-reset] received, client-instanc>
May 18 13:17:07 OMRVPS ovpn-tun0[480]: WARNING: Failed to stat CRL file, not reloading CRL.
May 18 13:17:07 OMRVPS ovpn-tun0[480]: WARNING: Failed to stat CRL file, not reloading CRL.
May 18 13:17:07 OMRVPS ovpn-tun0[480]: TCP connection established with [AF_INET6]::ffff:91.xxx.xxx.157:43087
May 18 13:17:07 OMRVPS ovpn-tun0[480]: Socket flags: TCP_NODELAY=1 succeeded
May 18 13:17:07 OMRVPS ovpn-tun0[480]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Operation not sup>
May 18 13:17:07 OMRVPS ovpn-tun0[480]: Note: enable extended error passing on TCP/UDP socket failed (IPV6_RECVERR): Operation not s>
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 TLS: Initial packet from [AF_INET6]::ffff:91.xxx.xxx.157:43087, sid=246>
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 VERIFY OK: depth=1, CN=Easy-RSA CA
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 VERIFY OK: depth=0, CN=openmptcprouter
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 peer info: IV_VER=2.6.13
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 peer info: IV_PLAT=linux
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 peer info: IV_TCPNL=1
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 peer info: IV_MTU=1600
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 peer info: IV_NCP=2
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 peer info: IV_CIPHERS=AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 peer info: IV_PROTO=990
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 peer info: IV_LZO_STUB=1
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 peer info: IV_COMP_STUB=1
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 peer info: IV_COMP_STUBv2=1
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 TLS: move_session: dest=TM_ACTIVE src=TM_INITIAL reinit_src=1
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 WARNING: Failed to stat CRL file, not reloading CRL.
May 18 13:17:07 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 TLS: tls_multi_process: initial untrusted session promoted to trusted
May 18 13:17:08 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256_GCM_SHA384, peer c>
May 18 13:17:08 OMRVPS ovpn-tun0[480]: 91.xxx.xxx.157:43087 [openmptcprouter] Peer Connection Initiated with [AF_INET6]::ffff:91.23>
May 18 13:17:08 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:43087 MULTI_sva: pool returned IPv4=10.255.252.2, IPv6=(Not e>
May 18 13:17:08 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:43087 Could not access file 'ccd/openmptcprouter': Permission>
May 18 13:17:08 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:43087 Could not access file 'ccd/DEFAULT': Permission denied >
May 18 13:17:08 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:43087 MULTI: Learn: 10.255.252.2 -> openmptcprouter/91.232.10>
May 18 13:17:08 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:43087 MULTI: primary virtual IP for openmptcprouter/91.232.10>
May 18 13:17:08 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:43087 SENT CONTROL [openmptcprouter]: 'PUSH_REPLY,sndbuf 2621>
May 18 13:17:08 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:43087 PUSH: Received control message: 'PUSH_REQUEST'
May 18 13:17:09 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:43087 Data Channel: cipher 'AES-256-GCM', peer-id: 0
May 18 13:17:09 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:43087 Timers: ping 10, ping-restart 480
May 18 13:17:09 OMRVPS ovpn-tun0[480]: openmptcprouter/91.xxx.xxx.157:43087 Protocol options: protocol-flags cc-exit tls-ekm dyn-tl>
May 18 13:17:11 OMRVPS ovpn-tun0[480]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:15631
May 18 13:17:11 OMRVPS ovpn-tun0[480]: MANAGEMENT: CMD 'status'
May 18 13:17:11 OMRVPS ovpn-tun0[480]: MANAGEMENT: Client disconnected

----------------------------------------------

router logs: 

Sun May 18 11:32:59 2025 daemon.info ModemManager[682]: hotplug: ModemManager not yet available (count=20)
[... cut ...]
Sun May 18 13:17:09 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sun May 18 13:17:09 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was 10.xxx.xxx.178)
Sun May 18 13:17:11 2025 daemon.notice netifd: omrvpn (27750): udhcpc: broadcasting discover
Sun May 18 13:17:13 2025 daemon.notice openvpn(omr)[26941]: Initialization Sequence Completed
Sun May 18 13:17:14 2025 daemon.notice netifd: omrvpn (27750): udhcpc: broadcasting discover

moussa11, 2025-05-18:
rebooted both router & VPS @13:30. took the error messages at 13:45 with last message logged 13:37.

can you please confirm if everything is working as it should ?

VPS: 
May 18 13:30:41 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:59477 Connection reset, restarting [0]
May 18 13:30:41 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:59477 SIGUSR1[soft,connection-reset] received, client-instanc>
May 18 13:30:41 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 18 13:30:41 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 18 13:30:41 OMRVPS ovpn-tun0[490]: TCP connection established with [AF_INET6]::ffff:91.xxx.xxx.157:61312
May 18 13:30:41 OMRVPS ovpn-tun0[490]: Socket flags: TCP_NODELAY=1 succeeded
May 18 13:30:41 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Operation not sup>
May 18 13:30:41 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IPV6_RECVERR): Operation not s>
May 18 13:30:41 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 TLS: Initial packet from [AF_INET6]::ffff:91.xxx.xxx.157:61312, sid=966>
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 VERIFY OK: depth=1, CN=Easy-RSA CA
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 VERIFY OK: depth=0, CN=openmptcprouter
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 peer info: IV_VER=2.6.13
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 peer info: IV_PLAT=linux
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 peer info: IV_TCPNL=1
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 peer info: IV_MTU=1600
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 peer info: IV_NCP=2
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 peer info: IV_CIPHERS=AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 peer info: IV_PROTO=990
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 peer info: IV_LZO_STUB=1
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 peer info: IV_COMP_STUB=1
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 peer info: IV_COMP_STUBv2=1
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 TLS: move_session: dest=TM_ACTIVE src=TM_INITIAL reinit_src=1
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 WARNING: Failed to stat CRL file, not reloading CRL.
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 TLS: tls_multi_process: initial untrusted session promoted to trusted
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256_GCM_SHA384, peer c>
May 18 13:30:42 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:61312 [openmptcprouter] Peer Connection Initiated with [AF_INET6]::ffff:91.23>
May 18 13:30:42 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:61312 MULTI_sva: pool returned IPv4=10.255.252.3, IPv6=(Not e>
May 18 13:30:42 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:61312 Could not access file 'ccd/openmptcprouter': Permission>
May 18 13:30:42 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:61312 Could not access file 'ccd/DEFAULT': Permission denied >
May 18 13:30:42 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:61312 MULTI: Learn: 10.255.252.3 -> openmptcprouter/91.232.10>
May 18 13:30:42 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:61312 MULTI: primary virtual IP for openmptcprouter/91.232.10>
May 18 13:30:42 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:61312 SENT CONTROL [openmptcprouter]: 'PUSH_REPLY,sndbuf 2621>
May 18 13:30:43 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:61312 Data Channel: cipher 'AES-256-GCM', peer-id: 0
May 18 13:30:43 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:61312 Timers: ping 10, ping-restart 480
May 18 13:30:43 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:61312 Protocol options: protocol-flags cc-exit tls-ekm dyn-tl>
May 18 13:30:46 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:32945
May 18 13:30:46 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 18 13:30:46 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 18 13:30:53 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:63267
May 18 13:30:53 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 18 13:30:53 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 18 13:30:53 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:63277
May 18 13:30:53 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 18 13:30:53 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 18 13:30:54 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:63281
May 18 13:30:54 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 18 13:30:54 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 18 13:30:54 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:63295
May 18 13:30:54 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 18 13:30:54 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 18 13:30:55 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:63305
May 18 13:30:55 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 18 13:30:55 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
[... cut ...]
Sun May 18 13:31:23 2025 daemon.info omr-tracker-ss: Shadowsocks rust sss0 is up (can contact via http 77.88.55.77)
Sun May 18 13:32:01 2025 user.notice omr-schedule-010-services: Set firewall on server DigitalOceanVPS
Sun May 18 13:32:37 2025 daemon.err uhttpd[5050]: [info] luci: accepted login on / for root from 192.168.100.134
Sun May 18 13:40:22 2025 daemon.notice netifd: wan1 (5952): udhcpc: sending renew to server 10.xxx.xxx1.178
Sun May 18 13:40:22 2025 daemon.notice netifd: wan1 (5952): udhcpc: lease of 10.xxx.xxx1.177 obtained from 10.xxx.xxx1.178, lease time 1200

Ysurac, 2025-05-18:
"May 18 13:37:44 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:33845 [openmptcprouter] Inactivity timeout (--ping-restart)" <= this is a problem ans we don't have same part on router side.
Both are set on UTC timezone ? Date/Time is the same on both side ?

moussa11, 2025-05-18:
both on same UTC time:

[image]

Ysurac, 2025-05-19:
opvn-tun0 is MPTCP over VPN. Disable this.
Disable also SQM to test if it's the same.

moussa11, 2025-05-19:
[image]

--------
router: Mon May 19 05:12:57 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:12:58 2025 daemon.err openvpn(omr)[20258]: event_wait : Interrupted system call (fd=-1,code=4)
Mon May 19 05:12:58 2025 daemon.notice openvpn(omr)[20258]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Mon May 19 05:12:58 2025 daemon.notice openvpn(omr)[20258]: net_addr_v4_del: 10.255.252.2 dev tun0
Mon May 19 05:12:58 2025 daemon.warn openvpn(omr)[20258]: sitnl_send: rtnl: generic error (-99): Address not available
Mon May 19 05:12:58 2025 daemon.warn openvpn(omr)[20258]: Linux can't del IP from iface tun0
Mon May 19 05:12:58 2025 user.notice NET: hotplug (iface): action='remove' interface='tun0'
Mon May 19 05:12:58 2025 daemon.info ModemManager[3144]: hotplug: remove network interface tun0: event processed
Mon May 19 05:12:58 2025 daemon.notice openvpn(omr)[20258]: /usr/libexec/openvpn-hotplug down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Mon May 19 05:12:58 2025 daemon.notice openvpn(omr)[20258]: SIGTERM[hard,] received, process exiting
Mon May 19 05:12:58 2025 daemon.notice netifd: Interface 'omrvpn' is disabled
Mon May 19 05:12:58 2025 daemon.notice netifd: Network device 'tun0' link is down
Mon May 19 05:12:58 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Mon May 19 05:12:58 2025 daemon.notice netifd: omrvpn (20460): udhcpc: read error: Network is down, reopening socket
Mon May 19 05:12:58 2025 daemon.notice netifd: omrvpn (20460): udhcpc: bind: No such device
Mon May 19 05:12:58 2025 daemon.notice netifd: Interface 'omrvpn' is now down
Mon May 19 05:12:59 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:12:59 2025 user.notice OMR-VPS: Get config from server DigitalOceanVPS...
Mon May 19 05:12:59 2025 user.notice OMR-VPS: Glorytun restart...
Mon May 19 05:13:00 2025 user.notice OMR-VPS: OpenVPN restart...
Mon May 19 05:13:00 2025 user.notice OMR-VPS: Restart shadowsocks Rust...
Mon May 19 05:13:00 2025 user.notice omr-schedule-010-services: Can't find Shadowsocks Rust, restart it...
Mon May 19 05:13:01 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:13:03 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:13:04 2025 user.notice omr-tracker: Launching...
Mon May 19 05:13:05 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:13:06 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:13:09 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:13:12 2025 user.notice omr-tracker: Launched
Mon May 19 05:13:12 2025 daemon.info vnstatd[5281]: Info: SIGTERM received, exiting.
Mon May 19 05:13:12 2025 daemon.info vnstatd[6958]: Info: vnStat daemon 2.12 (pid:6958 uid:0 gid:0 64-bit, SQLite 3.46.1)
Mon May 19 05:13:12 2025 daemon.info vnstatd[6958]: Info: Data retention: 48 5MinuteHours, 4 HourlyDays, 62 DailyDays, 25 MonthlyMonths, -1 YearlyYears, 20 TopDayEntries
Mon May 19 05:13:12 2025 daemon.info vnstatd[6958]: Info: Monitoring (3): pppoe-wan2 (1000 Mbit) eth1 (1000 Mbit) eth0 (1000 Mbit)
Mon May 19 05:13:13 2025 daemon.info omr-tracker-ss: Shadowsocks rust sss0 is up (can contact via http 212.27.48.10)
Mon May 19 05:13:15 2025 user.notice post-tracking-003-up: Change in routes, set ip route replace default scope global nexthop via 10.xxx.xxx.178 dev eth1 weight 100 nexthop via 92.xxx.xxx.254 dev pppoe-wan2 weight 1 (omrvpn_intf: tun0)
Mon May 19 05:13:18 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:13:18 2025 user.notice omr-tracker: Launching...
Mon May 19 05:13:18 2025 user.notice omr-bypass: Starting OMR-ByPass...
Mon May 19 05:13:25 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:13:25 2025 user.notice omr-tracker: Launched
Mon May 19 05:13:25 2025 user.notice omr-bypass: OMR-ByPass is running
Mon May 19 05:13:28 2025 user.notice omr-bypass: Starting OMR-ByPass...
Mon May 19 05:13:34 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:13:34 2025 user.notice omr-bypass: OMR-ByPass is running
Mon May 19 05:13:36 2025 daemon.notice ttyd[356]: [2025/05/19 05:13:36:0798] N: rops_handle_POLLIN_netlink: DELADDR
Mon May 19 05:13:36 2025 daemon.notice ttyd[356]: [2025/05/19 05:13:36:1208] N: rops_handle_POLLIN_netlink: DELADDR
Mon May 19 05:14:01 2025 user.notice omr-schedule-010-services: Set firewall on server DigitalOceanVPS
Mon May 19 05:14:11 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:14:14 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 05:14:15 2025 daemon.err procd: instance omr not found
Mon May 19 05:14:16 2025 daemon.info unbound: [20973:0] info: service stopped (unbound 1.21.0).
Mon May 19 05:14:16 2025 daemon.info unbound: [20973:0] info: server stats for thread 0: 0 queries, 0 answers from cache, 0 recursions, 0 prefetch, 0 rejected by ip ratelimiting
[... cut ...]
May 19 05:13:11 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 19 05:13:11 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:35681
May 19 05:13:11 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 19 05:13:11 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 19 05:13:12 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:35683
May 19 05:13:12 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 19 05:13:12 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected

Ysurac, 2025-05-20:
Can you try a fresh OpenMPTCProuter on router side without keeping previous settings and only set WANs ip & gateway, nothing more ?

Ysurac, 2025-05-22:
If you use squashFS a factory reset should suffice.

moussa11, 2025-05-22:
I am using ext4 image. I can reload the firmware from the UI. It will be more challenging to arrange for SD card to be removed.

moussa11, 2025-05-24:
re-installed firmware ext4 0.63 without keeping the settings. configured VPS settings, eth0 as LAN, eth1 - WAN1, eth2 - WAN2 (PPPOE). No other settings changed. 

Sat May 24 08:15:07 2025 kern.info kernel: [    2.134303] dma-pl330 fe530000.dma-controller: 	DBUFF-128x8bytes Num_Chans-8 Num_Peri-32 Num_Events-16
Sat May 24 08:15:07 2025 kern.info kernel: [    2.138028] dma-pl330 fe550000.dma-controller: Loaded driver for PL330 DMAC-241330
Sat May 24 08:15:07 2025 kern.info kernel: [    2.138781] dma-pl330 fe550000.dma-controller: 	DBUFF-128x8bytes Num_Chans-8 Num_Peri-32 Num_Events-16
Sat May 24 08:15:07 2025 kern.info kernel: [    2.141436] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
Sat May 24 08:15:07 2025 kern.info kernel: [    2.149833] printk: console [ttyS2] disabled
Sat May 24 08:15:07 2025 kern.info kernel: [    2.150662] fe660000.serial: ttyS2 at MMIO 0xfe660000 (irq = 31, base_baud = 1500000) is a 16550A
Sat May 24 08:15:07 2025 kern.info kernel: [    2.151809] printk: console [ttyS2] enabled
Sat May 24 08:15:07 2025 kern.info kernel: [    2.152595] printk: bootconsole [uart8250] disabled
Sat May 24 08:15:07 2025 kern.notice kernel: [    2.159106] random: crng init done
Sat May 24 08:15:07 2025 kern.info kernel: [    2.165930] loop: module loaded
Sat May 24 08:15:07 2025 kern.info kernel: [    2.166236] mtip32xx Version 1.3.1
Sat May 24 08:15:07 2025 kern.info kernel: [    2.172515] rk_gmac-dwmac fe2a0000.ethernet: IRQ eth_lpi not found
Sat May 24 08:15:07 2025 kern.warn kernel: [    2.173509] rk_gmac-dwmac fe2a0000.ethernet: supply phy not found, using dummy regulator
Sat May 24 08:15:07 2025 kern.info kernel: [    2.174393] rk_gmac-dwmac fe2a0000.ethernet: clock input or output? (output).
Sat May 24 08:15:07 2025 kern.info kernel: [    2.175031] rk_gmac-dwmac fe2a0000.ethernet: TX delay(0x3c).
Sat May 24 08:15:07 2025 kern.info kernel: [    2.175536] rk_gmac-dwmac fe2a0000.ethernet: RX delay(0x2f).
Sat May 24 08:15:07 2025 kern.info kernel: [    2.176053] rk_gmac-dwmac fe2a0000.ethernet: integrated PHY? (no).
Sat May 24 08:15:07 2025 kern.info kernel: [    2.181700] rk_gmac-dwmac fe2a0000.ethernet: init for RGMII
Sat May 24 08:15:07 2025 kern.info kernel: [    2.182506] rk_gmac-dwmac fe2a0000.ethernet: User ID: 0x30, Synopsys ID: 0x51
Sat May 24 08:15:07 2025 kern.info kernel: [    2.183149] rk_gmac-dwmac fe2a0000.ethernet: 	DWMAC4/5
Sat May 24 08:15:07 2025 kern.info kernel: [    2.183611] rk_gmac-dwmac fe2a0000.ethernet: DMA HW capability register supported
Sat May 24 08:15:07 2025 kern.info kernel: [    2.184271] rk_gmac-dwmac fe2a0000.ethernet: RX Checksum Offload Engine supported
Sat May 24 08:15:07 2025 kern.info kernel: [    2.184930] rk_gmac-dwmac fe2a0000.ethernet: TX Checksum insertion supported
Sat May 24 08:15:07 2025 kern.info kernel: [    2.185552] rk_gmac-dwmac fe2a0000.ethernet: Wake-Up On Lan supported
Sat May 24 08:15:07 2025 kern.info kernel: [    2.186120] rk_gmac-dwmac fe2a0000.ethernet: TSO supported
Sat May 24 08:15:07 2025 kern.info kernel: [    2.186637] rk_gmac-dwmac fe2a0000.ethernet: Enable RX Mitigation via HW Watchdog Timer
Sat May 24 08:15:07 2025 kern.info kernel: [    2.187348] rk_gmac-dwmac fe2a0000.ethernet: Enabled RFS Flow TC (entries=10)
Sat May 24 08:15:07 2025 kern.info kernel: [    2.187981] rk_gmac-dwmac fe2a0000.ethernet: TSO feature enabled
Sat May 24 08:15:07 2025 kern.info kernel: [    2.188515] rk_gmac-dwmac fe2a0000.ethernet: Using 32/32 bits DMA host/device width
Sat May 24 08:15:07 2025 kern.info kernel: [    2.315162] usbcore: registered new interface driver usb-storage
Sat May 24 08:15:07 2025 kern.info kernel: [    2.315968] i2c_dev: i2c /dev entries driver
Sat May 24 08:15:07 2025 kern.info kernel: [    2.316614] ehci-platform fd800000.usb: EHCI Host Controller
Sat May 24 08:15:07 2025 kern.info kernel: [    2.316627] ehci-platform fd880000.usb: EHCI Host Controller
Sat May 24 08:15:07 2025 kern.info kernel: [    2.316663] ohci-platform fd8c0000.usb: Generic Platform OHCI controller
Sat May 24 08:15:07 2025 kern.info kernel: [    2.316696] ohci-platform fd8c0000.usb: new USB bus registered, assigned bus number 1
Sat May 24 08:15:07 2025 kern.info kernel: [    2.316869] ohci-platform fd8c0000.usb: irq 40, io mem 0xfd8c0000
Sat May 24 08:15:07 2025 kern.info kernel: [    2.316901] ohci-platform fd840000.usb: Generic Platform OHCI controller
Sat May 24 08:15:07 2025 kern.info kernel: [    2.316938] ohci-platform fd840000.usb: new USB bus registered, assigned bus number 2
Sat May 24 08:15:07 2025 kern.info kernel: [    2.317032] ohci-platform fd840000.usb: irq 39, io mem 0xfd840000
Sat May 24 08:15:07 2025 kern.info kernel: [    2.317144] ehci-platform fd800000.usb: new USB bus registered, assigned bus number 3
Sat May 24 08:15:07 2025 kern.info kernel: [    2.317644] ehci-platform fd880000.usb: new USB bus registered, assigned bus number 4
Sat May 24 08:15:07 2025 kern.info kernel: [    2.318371] ehci-platform fd800000.usb: irq 37, io mem 0xfd800000
Sat May 24 08:15:07 2025 kern.info kernel: [    2.318995] ehci-platform fd880000.usb: irq 38, io mem 0xfd880000
Sat May 24 08:15:07 2025 kern.info kernel: [    2.324304] fan53555-regulator 0-001c: FAN53555 Option[12] Rev[15] Detected!
Sat May 24 08:15:07 2025 kern.info kernel: [    2.346603] ehci-platform fd800000.usb: USB 2.0 started, EHCI 1.00
Sat May 24 08:15:07 2025 kern.info kernel: [    2.348173] hub 3-0:1.0: USB hub found
[... cut ...]
Sat May 24 08:59:26 2025 daemon.info unbound: [14455:0] info: generate keytag query _ta-4f66. NULL IN
Sat May 24 08:59:26 2025 daemon.notice netifd: omrvpn (14140): udhcpc: broadcasting discover
Sat May 24 08:59:28 2025 daemon.notice openvpn(omr)[13825]: Initialization Sequence Completed
Sat May 24 08:59:29 2025 daemon.notice netifd: omrvpn (14140): udhcpc: broadcasting discover
Sat May 24 08:59:29 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was 10.115.111.178)

moussa11, 2025-05-24:
VPS: 

May 24 08:59:00 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 24 08:59:00 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 24 08:59:00 OMRVPS ovpn-tun0[490]: TCP connection established with [AF_INET6]::ffff:91.xxx.xxx.157:47520
May 24 08:59:00 OMRVPS ovpn-tun0[490]: Socket flags: TCP_NODELAY=1 succeeded
May 24 08:59:00 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Operation not sup>
May 24 08:59:00 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IPV6_RECVERR): Operation not s>
May 24 08:59:00 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 TLS: Initial packet from [AF_INET6]::ffff:91.xxx.xxx.157:47520, sid=0c0>
May 24 08:59:00 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 VERIFY OK: depth=1, CN=Easy-RSA CA
May 24 08:59:00 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 VERIFY OK: depth=0, CN=openmptcprouter
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 peer info: IV_VER=2.6.13
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 peer info: IV_PLAT=linux
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 peer info: IV_TCPNL=1
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 peer info: IV_MTU=1600
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 peer info: IV_NCP=2
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 peer info: IV_CIPHERS=AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 peer info: IV_PROTO=990
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 peer info: IV_LZO_STUB=1
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 peer info: IV_COMP_STUB=1
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 peer info: IV_COMP_STUBv2=1
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 TLS: move_session: dest=TM_ACTIVE src=TM_INITIAL reinit_src=1
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 WARNING: Failed to stat CRL file, not reloading CRL.
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 TLS: tls_multi_process: initial untrusted session promoted to trusted
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256_GCM_SHA384, peer c>
May 24 08:59:01 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:47520 [openmptcprouter] Peer Connection Initiated with [AF_INET6]::ffff:91.23>
May 24 08:59:01 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:47520 MULTI_sva: pool returned IPv4=10.255.252.2, IPv6=(Not e>
May 24 08:59:01 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:47520 Could not access file 'ccd/openmptcprouter': Permission>
May 24 08:59:01 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:47520 Could not access file 'ccd/DEFAULT': Permission denied >
May 24 08:59:01 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:47520 MULTI: Learn: 10.255.252.2 -> openmptcprouter/91.232.10>
May 24 08:59:01 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:47520 MULTI: primary virtual IP for openmptcprouter/91.232.10>
May 24 08:59:01 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:47520 SENT CONTROL [openmptcprouter]: 'PUSH_REPLY,sndbuf 2621>
May 24 08:59:01 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:47520 PUSH: Received control message: 'PUSH_REQUEST'
May 24 08:59:02 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:47520 Data Channel: cipher 'AES-256-GCM', peer-id: 0
May 24 08:59:02 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:47520 Timers: ping 10, ping-restart 480
May 24 08:59:02 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:47520 Protocol options: protocol-flags cc-exit tls-ekm dyn-tl>
May 24 08:59:14 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:18045
May 24 08:59:14 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 24 08:59:14 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 24 08:59:14 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:18053
May 24 08:59:14 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 24 08:59:14 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 24 08:59:22 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 24 08:59:22 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 24 08:59:22 OMRVPS ovpn-tun0[490]: TCP connection established with [AF_INET6]::ffff:91.xxx.xxx.157:34342
May 24 08:59:22 OMRVPS ovpn-tun0[490]: Socket flags: TCP_NODELAY=1 succeeded
May 24 08:59:22 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Operation not sup>

root@OMRVPS:~# journalctl -u openvpn@tun0 --since "1 hour  ago"
May 24 08:59:00 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 24 08:59:00 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 24 08:59:00 OMRVPS ovpn-tun0[490]: TCP connection established with [AF_INET6]::ffff:91.xxx.xxx.157:47520
May 24 08:59:00 OMRVPS ovpn-tun0[490]: Socket flags: TCP_NODELAY=1 succeeded
May 24 08:59:00 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Operation not sup>
May 24 08:59:00 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IPV6_RECVERR): Operation not s>
[... cut ...]
May 24 09:03:27 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 24 09:03:47 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:51657
May 24 09:03:47 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 24 09:03:47 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 24 09:09:28 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:14835
May 24 09:09:28 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 24 09:09:28 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected

moussa11, 2025-05-25:
[image]

May 25 04:17:58 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:33273 Connection reset, restarting [0]
May 25 04:17:58 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:33273 SIGUSR1[soft,connection-reset] recei>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 25 04:17:59 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 25 04:17:59 OMRVPS ovpn-tun0[490]: TCP connection established with [AF_INET6]::ffff:91.xxx.xxx.157:50677
May 25 04:17:59 OMRVPS ovpn-tun0[490]: Socket flags: TCP_NODELAY=1 succeeded
May 25 04:17:59 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR)>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IPV6_RECVER>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 TLS: Initial packet from [AF_INET6]::ffff:91.xxx.xxx>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 VERIFY OK: depth=1, CN=Easy-RSA CA
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 VERIFY OK: depth=0, CN=openmptcprouter
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 peer info: IV_VER=2.6.13
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 peer info: IV_PLAT=linux
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 peer info: IV_TCPNL=1
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 peer info: IV_MTU=1600
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 peer info: IV_NCP=2
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 peer info: IV_CIPHERS=AES-256-GCM:AES-128-GCM:CHACHA>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 peer info: IV_PROTO=990
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 peer info: IV_LZO_STUB=1
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 peer info: IV_COMP_STUB=1
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 peer info: IV_COMP_STUBv2=1
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 TLS: move_session: dest=TM_ACTIVE src=TM_INITIAL rei>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 WARNING: Failed to stat CRL file, not reloading CRL.
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 TLS: tls_multi_process: initial untrusted session pr>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: 91.xxx.xxx.157:50677 [openmptcprouter] Peer Connection Initiated with [AF>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:50677 MULTI_sva: pool returned IPv4=10.255>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:50677 Could not access file 'ccd/openmptcp>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:50677 Could not access file 'ccd/DEFAULT':>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:50677 MULTI: Learn: 10.255.252.2 -> openmp>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:50677 MULTI: primary virtual IP for openmp>
May 25 04:17:59 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:50677 SENT CONTROL [openmptcprouter]: 'PUS>
May 25 04:18:00 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:50677 Data Channel: cipher 'AES-256-GCM', >
May 25 04:18:00 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:50677 Timers: ping 10, ping-restart 480
May 25 04:18:00 OMRVPS ovpn-tun0[490]: openmptcprouter/91.xxx.xxx.157:50677 Protocol options: protocol-flags cc->
May 25 04:18:04 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:55287
May 25 04:18:04 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 25 04:18:04 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 25 04:18:04 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:55289
May 25 04:18:04 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 25 04:18:04 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 25 04:18:05 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:55303
May 25 04:18:05 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 25 04:18:05 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 25 04:18:05 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:55319
May 25 04:18:05 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 25 04:18:05 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 25 04:18:06 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:55331
May 25 04:18:06 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 25 04:18:06 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 25 04:18:09 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:40225
May 25 04:18:09 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 25 04:18:09 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 25 04:18:15 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:40231
May 25 04:18:15 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 25 04:18:15 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 25 04:18:15 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:40247
May 25 04:18:15 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 25 04:18:15 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
[... cut ...]
May 25 04:18:31 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 25 04:18:32 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:45905
May 25 04:18:32 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 25 04:18:32 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 25 04:18:32 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:45917
May 25 04:18:32 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 25 04:18:32 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected

moussa11, 2025-05-25:
router: LBN, VPS: NLD

set wan 1 as master, disabled wan 2. below logs. 

VPS: 
root@OMRVPS:~# journalctl -u openvpn@tun0 --since "10 min  ago"
May 25 22:13:55 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 25 22:13:55 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 25 22:13:55 OMRVPS ovpn-tun0[490]: TCP connection established with [AF_INET6]::ffff:92.xxx.xxx.85:14951
May 25 22:13:55 OMRVPS ovpn-tun0[490]: Socket flags: TCP_NODELAY=1 succeeded
May 25 22:13:55 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR)>
May 25 22:13:55 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IPV6_RECVER>
May 25 22:13:55 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 TLS: Initial packet from [AF_INET6]::ffff:92.xxx.xxx.85>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 VERIFY OK: depth=1, CN=Easy-RSA CA
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 VERIFY OK: depth=0, CN=openmptcprouter
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 peer info: IV_VER=2.6.13
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 peer info: IV_PLAT=linux
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 peer info: IV_TCPNL=1
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 peer info: IV_MTU=1600
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 peer info: IV_NCP=2
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 peer info: IV_CIPHERS=AES-256-GCM:AES-128-GCM:CHACHA20>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 peer info: IV_PROTO=990
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 peer info: IV_LZO_STUB=1
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 peer info: IV_COMP_STUB=1
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 peer info: IV_COMP_STUBv2=1
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 TLS: move_session: dest=TM_ACTIVE src=TM_INITIAL reini>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 WARNING: Failed to stat CRL file, not reloading CRL.
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 TLS: tls_multi_process: initial untrusted session prom>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256_G>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14951 [openmptcprouter] Peer Connection Initiated with [AF_I>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: openmptcprouter/92.xxx.xxx.85:14951 MULTI_sva: pool returned IPv4=10.255.2>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: openmptcprouter/92.xxx.xxx.85:14951 Could not access file 'ccd/openmptcpro>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: openmptcprouter/92.xxx.xxx.85:14951 Could not access file 'ccd/DEFAULT': P>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: openmptcprouter/92.xxx.xxx.85:14951 MULTI: Learn: 10.255.252.3 -> openmptc>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: openmptcprouter/92.xxx.xxx.85:14951 MULTI: primary virtual IP for openmptc>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: openmptcprouter/92.xxx.xxx.85:14951 SENT CONTROL [openmptcprouter]: 'PUSH_>
May 25 22:13:56 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:33497
May 25 22:13:56 OMRVPS ovpn-tun0[490]: MANAGEMENT: CMD 'status'
May 25 22:13:56 OMRVPS ovpn-tun0[490]: MANAGEMENT: Client disconnected
May 25 22:13:57 OMRVPS ovpn-tun0[490]: openmptcprouter/92.xxx.xxx.85:14951 Connection reset, restarting [0]
May 25 22:13:57 OMRVPS ovpn-tun0[490]: openmptcprouter/92.xxx.xxx.85:14951 SIGUSR1[soft,connection-reset] receive>
May 25 22:13:57 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 25 22:13:57 OMRVPS ovpn-tun0[490]: WARNING: Failed to stat CRL file, not reloading CRL.
May 25 22:13:57 OMRVPS ovpn-tun0[490]: TCP connection established with [AF_INET6]::ffff:92.xxx.xxx.85:14953
May 25 22:13:57 OMRVPS ovpn-tun0[490]: Socket flags: TCP_NODELAY=1 succeeded
May 25 22:13:57 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR)>
May 25 22:13:57 OMRVPS ovpn-tun0[490]: Note: enable extended error passing on TCP/UDP socket failed (IPV6_RECVER>
May 25 22:13:57 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 TLS: Initial packet from [AF_INET6]::ffff:92.xxx.xxx.85>
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 VERIFY OK: depth=1, CN=Easy-RSA CA
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 VERIFY OK: depth=0, CN=openmptcprouter
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 peer info: IV_VER=2.6.13
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 peer info: IV_PLAT=linux
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 peer info: IV_TCPNL=1
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 peer info: IV_MTU=1600
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 peer info: IV_NCP=2
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 peer info: IV_CIPHERS=AES-256-GCM:AES-128-GCM:CHACHA20>
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 peer info: IV_PROTO=990
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 peer info: IV_LZO_STUB=1
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 peer info: IV_COMP_STUB=1
May 25 22:13:58 OMRVPS ovpn-tun0[490]: 92.xxx.xxx.85:14953 peer info: IV_COMP_STUBv2=1
[... cut ...]
Sun May 25 22:26:40 2025 daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules
Sun May 25 22:26:40 2025 daemon.info omr-tracker-ss: Rules still not set, restart Shadowsocks-Rust
Sun May 25 22:26:51 2025 daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules
Sun May 25 22:26:51 2025 daemon.info omr-tracker-ss: Rules still not set, restart Shadowsocks-Rust
Sun May 25 22:27:02 2025 daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules
Sun May 25 22:27:02 2025 daemon.info omr-tracker-ss: Rules still not set, restart Shadowsocks-Rust

moussa11, 2025-05-25:
wan 1 disabled, wan 2 master

seems same result expect on the OMR status WAN 1 was showing red X when it was the only one enabled. while when WAN 2 was enabled it was green check. 

router: Sun May 25 22:50:24 2025 user.notice post-tracking-002-error: wan1 (eth1) switched off because check error and ping from 10.xxx.xxx.177 error (4.2.2.1,8.8.8.8,80.67.169.12)
Sun May 25 22:50:24 2025 user.notice post-tracking-002-error: Delete default route to 134.xxx.xxx.58 dev eth1
Sun May 25 22:50:25 2025 user.notice post-tracking-002-error: DNS flush
Sun May 25 22:51:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sun May 25 22:51:00 2025 user.notice omr-tracker: Launching...
Sun May 25 22:51:04 2025 user.notice omr-tracker: Launched
Sun May 25 22:51:24 2025 user.notice post-tracking-002-error: DNS flush
Sun May 25 22:51:46 2025 daemon.notice ttyd[19001]: [2025/05/25 22:51:46:6453] N: rops_handle_POLLIN_netlink: DELADDR
Sun May 25 22:51:46 2025 daemon.notice ttyd[19001]: [2025/05/25 22:51:46:6456] N: rops_handle_POLLIN_netlink: DELADDR
Sun May 25 22:51:48 2025 user.notice MPTCP: Set pppoe-wan2 to master from pppoe-wan2 is deactivated
Sun May 25 22:51:49 2025 daemon.err openvpn(omr)[26185]: event_wait : Interrupted system call (fd=-1,code=4)
Sun May 25 22:51:49 2025 daemon.notice openvpn(omr)[26185]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Sun May 25 22:51:49 2025 daemon.notice openvpn(omr)[26185]: net_addr_v4_del: 10.255.252.2 dev tun0
Sun May 25 22:51:49 2025 daemon.warn openvpn(omr)[26185]: sitnl_send: rtnl: generic error (-99): Address not available
Sun May 25 22:51:49 2025 daemon.warn openvpn(omr)[26185]: Linux can't del IP from iface tun0
Sun May 25 22:51:49 2025 daemon.notice netifd: Network device 'tun0' link is down
Sun May 25 22:51:49 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sun May 25 22:51:49 2025 daemon.notice netifd: omrvpn (26638): udhcpc: read error: Network is down, reopening socket
Sun May 25 22:51:49 2025 user.notice NET: hotplug (iface): action='remove' interface='tun0'
Sun May 25 22:51:49 2025 daemon.info ModemManager[3797]: hotplug: remove network interface tun0: event processed
Sun May 25 22:51:49 2025 daemon.notice openvpn(omr)[26185]: /usr/libexec/openvpn-hotplug down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Sun May 25 22:51:49 2025 daemon.notice openvpn(omr)[26185]: SIGTERM[hard,] received, process exiting
Sun May 25 22:51:49 2025 daemon.notice openvpn(omr)[3911]: OpenVPN 2.6.13 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Sun May 25 22:51:49 2025 daemon.notice openvpn(omr)[3911]: library versions: OpenSSL 3.0.16 11 Feb 2025, LZO 2.10
Sun May 25 22:51:49 2025 daemon.notice openvpn(omr)[3911]: DCO version: N/A
Sun May 25 22:51:49 2025 daemon.warn openvpn(omr)[3911]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Sun May 25 22:51:49 2025 daemon.warn openvpn(omr)[3911]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sun May 25 22:51:49 2025 daemon.notice openvpn(omr)[3911]: TCP/UDP: Preserving recently used remote address: [AF_INET]134.xxx.xxx.58:65301
Sun May 25 22:51:49 2025 daemon.notice openvpn(omr)[3911]: Attempting to establish TCP connection with [AF_INET]134.xxx.xxx.58:65301
Sun May 25 22:51:50 2025 user.notice omr-tracker: Launching...
Sun May 25 22:51:50 2025 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun May 25 22:51:50 2025 daemon.notice netifd: omrvpn (26638): udhcpc: bind: No such device
Sun May 25 22:51:50 2025 daemon.notice netifd: Interface 'omrvpn' is now down
Sun May 25 22:51:54 2025 user.notice omr-tracker: Launched
Sun May 25 22:51:54 2025 user.notice post-tracking-002-error: omrvpn down
Sun May 25 22:51:54 2025 daemon.info vnstatd[13733]: Info: SIGTERM received, exiting.
Sun May 25 22:51:54 2025 daemon.info vnstatd[4634]: Info: vnStat daemon 2.12 (pid:4634 uid:0 gid:0 64-bit, SQLite 3.46.1)
Sun May 25 22:51:54 2025 daemon.info vnstatd[4634]: Info: Data retention: 48 5MinuteHours, 4 HourlyDays, 62 DailyDays, 25 MonthlyMonths, -1 YearlyYears, 20 TopDayEntries
Sun May 25 22:51:54 2025 daemon.info vnstatd[4634]: Info: Monitoring (2): wan2 (1000 Mbit) wan1 (1000 Mbit)
Sun May 25 22:51:54 2025 daemon.info vnstatd[4634]: Info: Interface "wan2" disabled.
Sun May 25 22:51:54 2025 daemon.info vnstatd[4634]: Info: Interface "wan1" disabled.
Sun May 25 22:51:54 2025 daemon.info xray: No xray instance enabled, skip transparent proxy.
Sun May 25 22:51:55 2025 user.notice post-tracking-002-error: OpenVPN down, restart it
Sun May 25 22:51:55 2025 daemon.notice openvpn(omr)[3911]: SIGTERM[hard,init_instance] received, process exiting
Sun May 25 22:51:55 2025 daemon.notice openvpn(omr)[5503]: OpenVPN 2.6.13 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Sun May 25 22:51:55 2025 daemon.notice openvpn(omr)[5503]: library versions: OpenSSL 3.0.16 11 Feb 2025, LZO 2.10
Sun May 25 22:51:55 2025 daemon.notice openvpn(omr)[5503]: DCO version: N/A
Sun May 25 22:51:55 2025 daemon.warn openvpn(omr)[5503]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
[... cut ...]
May 25 22:52:45 OMRVPS ovpn-tun0[494]: MANAGEMENT: Client disconnected
May 25 22:52:46 OMRVPS ovpn-tun0[494]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:12745
May 25 22:52:46 OMRVPS ovpn-tun0[494]: MANAGEMENT: CMD 'status'
May 25 22:52:46 OMRVPS ovpn-tun0[494]: MANAGEMENT: Client disconnected
May 25 22:52:46 OMRVPS ovpn-tun0[494]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:12753
May 25 22:52:46 OMRVPS ovpn-tun0[494]: MANAGEMENT: CMD 'status'
May 25 22:52:46 OMRVPS ovpn-tun0[494]: MANAGEMENT: Client disconnected

moussa11, 2025-05-26:
I have enabled MTCP over VPN for both WAN1 & WAN2. 

[image]

[image]

[image]

VPS: 
openvpn@tun0 
-- No Entries --
openvpn@tun1
May 25 22:41:03 OMRVPS systemd[1]: Stopping openvpn@tun1.service - OpenVPN connection to tun1...
May 25 22:41:03 OMRVPS ovpn-tun1[493]: event_wait : Interrupted system call (fd=-1,code=4)
May 25 22:41:03 OMRVPS ovpn-tun1[493]: Closing TUN/TAP interface
May 25 22:41:03 OMRVPS ovpn-tun1[493]: net_addr_v4_del: 10.255.250.1 dev tun1
May 25 22:41:03 OMRVPS ovpn-tun1[493]: SIGTERM[hard,] received, process exiting
May 25 22:41:03 OMRVPS systemd[1]: openvpn@tun1.service: Deactivated successfully.
May 25 22:41:03 OMRVPS systemd[1]: Stopped openvpn@tun1.service - OpenVPN connection to tun1.
May 25 22:41:04 OMRVPS systemd[1]: openvpn@tun1.service: Consumed 7.385s CPU time.
-- Boot 2550c4396cd742b193029997f3cfd403 --
May 25 22:41:28 OMRVPS systemd[1]: Starting openvpn@tun1.service - OpenVPN connection to tun1...
May 25 22:41:28 OMRVPS ovpn-tun1[496]: Note: --cipher is not set. OpenVPN versions before 2.5 defaulted to BF-CBC as fallback when ciph>
May 25 22:41:28 OMRVPS ovpn-tun1[496]: Note: Kernel support for ovpn-dco missing, disabling data channel offload.
May 25 22:41:28 OMRVPS ovpn-tun1[496]: OpenVPN 2.6.3 x86_64-pc-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [PKCS11] [MH/PKTINFO] [AEA>
May 25 22:41:28 OMRVPS ovpn-tun1[496]: library versions: OpenSSL 3.0.16 11 Feb 2025, LZO 2.10
May 25 22:41:28 OMRVPS ovpn-tun1[496]: DCO version: N/A
May 25 22:41:28 OMRVPS ovpn-tun1[496]: net_route_v4_best_gw query: dst 0.0.0.0
May 25 22:41:28 OMRVPS ovpn-tun1[496]: net_route_v4_best_gw result: via 134.xxx.xxx.1 dev eth0
May 25 22:41:28 OMRVPS ovpn-tun1[496]: Diffie-Hellman initialized with 2048 bit key
May 25 22:41:29 OMRVPS ovpn-tun1[496]: CRL: loaded 1 CRLs from file /etc/openvpn/ca/pki/crl.pem
May 25 22:41:29 OMRVPS ovpn-tun1[496]: TUN/TAP device tun1 opened
May 25 22:41:29 OMRVPS ovpn-tun1[496]: TUN/TAP TX queue length set to 1000
May 25 22:41:29 OMRVPS ovpn-tun1[496]: net_iface_mtu_set: mtu 1420 for tun1
May 25 22:41:29 OMRVPS ovpn-tun1[496]: net_iface_up: set tun1 up
May 25 22:41:29 OMRVPS ovpn-tun1[496]: net_addr_v4_add: 10.255.250.1/24 dev tun1
May 25 22:41:29 OMRVPS ovpn-tun1[496]: Socket Buffers: R=[212992->524288] S=[212992->524288]
May 25 22:41:29 OMRVPS ovpn-tun1[496]: setsockopt(IPV6_V6ONLY=0)
May 25 22:41:29 OMRVPS ovpn-tun1[496]: UDPv6 link local (bound): [AF_INET6][undef]:65301
May 25 22:41:29 OMRVPS ovpn-tun1[496]: UDPv6 link remote: [AF_UNSPEC]
May 25 22:41:29 OMRVPS ovpn-tun1[496]: MULTI: multi_init called, r=256 v=256
May 25 22:41:29 OMRVPS ovpn-tun1[496]: IFCONFIG POOL IPv4: base=10.255.250.2 size=253
May 25 22:41:29 OMRVPS ovpn-tun1[496]: Initialization Sequence Completed
May 25 22:41:29 OMRVPS systemd[1]: Started openvpn@tun1.service - OpenVPN connection to tun1.

root@OMRVPS:~# journalctl -u wg-quick@wg0.service 
May 25 22:41:03 OMRVPS systemd[1]: Stopping wg-quick@wg0.service - WireGuard via wg-quick(8) for wg0...
May 25 22:41:03 OMRVPS wg-quick[2400977]: [#] wg showconf wg0
May 25 22:41:03 OMRVPS wg-quick[2400919]: [#] ip link delete dev wg0
May 25 22:41:04 OMRVPS systemd[1]: wg-quick@wg0.service: Deactivated successfully.
May 25 22:41:04 OMRVPS systemd[1]: Stopped wg-quick@wg0.service - WireGuard via wg-quick(8) for wg0.
-- Boot 2550c4396cd742b193029997f3cfd403 --
May 25 22:41:29 OMRVPS systemd[1]: Starting wg-quick@wg0.service - WireGuard via wg-quick(8) for wg0...
May 25 22:41:29 OMRVPS wg-quick[544]: [#] ip link add wg0 type wireguard
May 25 22:41:29 OMRVPS wg-quick[544]: [#] wg setconf wg0 /dev/fd/63
May 25 22:41:29 OMRVPS wg-quick[544]: [#] ip -4 address add 10.xxx.xxx.1/24 dev wg0
May 25 22:41:29 OMRVPS wg-quick[544]: [#] ip link set mtu 1420 up dev wg0
May 25 22:41:29 OMRVPS systemd[1]: Finished wg-quick@wg0.service - WireGuard via wg-quick(8) for wg0.

router:
Mon May 26 02:40:14 2025 user.notice NET: hotplug (iface): action='add' interface='tun0'
Mon May 26 02:40:15 2025 daemon.notice netifd: omrvpn (29374): udhcpc: broadcasting discover
Mon May 26 02:40:15 2025 daemon.info ModemManager[29314]: hotplug: add network interface tun0: event processed
Mon May 26 02:40:16 2025 user.notice omr-bypass: Starting OMR-ByPass...
Mon May 26 02:40:18 2025 daemon.notice netifd: omrvpn (29374): udhcpc: broadcasting discover
Mon May 26 02:40:19 2025 daemon.notice openvpn(omr)[28661]: Initialization Sequence Completed
Mon May 26 02:40:20 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was 10.xxx.xxx.237)
Mon May 26 02:40:21 2025 daemon.notice netifd: omrvpn (29374): udhcpc: broadcasting discover
Mon May 26 02:40:22 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 26 02:40:23 2025 user.notice omr-bypass: OMR-ByPass is running
Mon May 26 02:49:31 2025 daemon.notice netifd: wan1 (11516): udhcpc: sending renew to server 10.xxx.xxx.237
Mon May 26 02:49:31 2025 daemon.notice netifd: wan1 (11516): udhcpc: lease of 10.xxx.xxx.238 obtained from 10.xxx.xxx.237, lease time 1200
Mon May 26 02:59:31 2025 daemon.notice netifd: wan1 (11516): udhcpc: sending renew to server 10.xxx.xxx.237
Mon May 26 02:59:31 2025 daemon.notice netifd: wan1 (11516): udhcpc: lease of 10.xxx.xxx.238 obtained from 10.xxx.xxx.237, lease time 1200
Mon May 26 03:09:31 2025 daemon.notice netifd: wan1 (11516): udhcpc: sending renew to server 10.xxx.xxx.237
[... cut ...]
Mon May 26 03:49:31 2025 daemon.notice netifd: wan1 (11516): udhcpc: lease of 10.xxx.xxx.238 obtained from 10.xxx.xxx.237, lease time 1200
Mon May 26 03:59:31 2025 daemon.notice netifd: wan1 (11516): udhcpc: sending renew to server 10.xxx.xxx.237
Mon May 26 03:59:31 2025 daemon.notice netifd: wan1 (11516): udhcpc: lease of 10.xxx.xxx.238 obtained from 10.xxx.xxx.237, lease time 1200
Mon May 26 04:09:32 2025 daemon.notice netifd: wan1 (11516): udhcpc: sending renew to server 10.xxx.xxx.237
Mon May 26 04:09:32 2025 daemon.notice netifd: wan1 (11516): udhcpc: lease of 10.xxx.xxx.238 obtained from 10.xxx.xxx.237, lease time 1200

Ysurac, 2025-05-26:
MPTCP over VPN is only when provider filter MPTCP or some country firewall. It's better or not ? You didn't answered in which country you are.
It shouldn't be needed in China, Iran seems to be a bit more difficult and I have nobody really available for tests here, I didn't know any others country with problem for now.

Ysurac, 2025-05-26:
Sorry, I didn't see that.
Yes I don't think there is filtering lebanon, there is no real reason to have that. You tried with VPS in another country ?
openvpn@tun1 is for MPTCP over VPN when using OpenVPN for that and not Wireguard. From the screenshot, you are using wireguard, so you should have nothing for tun1.

moussa11, 2025-05-26:
Initial VPS was on AWS london. That didn't work. 
Prior to switching on MPTCP over VPN for both WANs the router kept giving repeatedly:

```
user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
user.notice omr-tracker: Launching...
user.notice omr-tracker: Launched
```

How to confirm whether everything is working well now? If its preferred not to use MPTCP over VPN, what else could we try ?

Ysurac, 2025-05-27:
Strange log, I will check how this can be possible.
To test, only go to Network->MPTCP, "Bandwidth" tab to check if more than one interface is used when you download (to download you can use "omr-test-speed" via SSH).
MPTCP over VPN is not preferred when this work without it. In your case this seems to be an issue with the VPN, maybe you can try XRay as proxy (System->OpenMPTCProuter, wizard tab, advanced settings chekbox) with UDP enabled (System->OpenMPTCProuter, Advanced settings tab, V2Ray/XRay UDP checkbox) and siable the VPN.

moussa11, 2025-05-27:
With MPTCP over VPN turned ON: omr-test-speed shows traffic utlisation on both wan. I can see that download speeds and access to camera is much faster. 

Tested with V2Ray/XRay UDP checked > with the following proxy : XRay socks, XRay VLESS, XRay Shadowsocks 2022. All same results: 

router: 
Tue May 27 09:20:28 2025 daemon.info xray[22215]: Xray 25.4.30 (Xray, Penetrates Everything.) Custom (go1.24.1 linux/arm64)
Tue May 27 09:20:28 2025 daemon.info xray[22215]: A unified platform for anti-censorship.
Tue May 27 09:20:28 2025 daemon.info xray[22215]: 2025/05/27 09:20:28.049890 [Info] infra/conf/serial: Reading config: &{Name:/var/etc/xray/xray.main.json Format:json}
Tue May 27 09:20:32 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Tue May 27 09:20:32 2025 user.notice omr-bypass: Starting OMR-ByPass...
Tue May 27 09:20:35 2025 daemon.info omr-tracker-xray: xray is up (can contact via http 151.101.129.164)
Tue May 27 09:20:38 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Tue May 27 09:20:39 2025 user.notice omr-bypass: OMR-ByPass is running
Tue May 27 09:20:39 2025 user.notice omr-tracker: Launching...
Tue May 27 09:20:47 2025 user.notice omr-tracker: Launched
Tue May 27 09:20:47 2025 daemon.info omr-tracker-xray: xray is up (can contact via http 212.27.48.10)
Tue May 27 09:20:52 2025 user.notice omr-bypass: Starting OMR-ByPass...
Tue May 27 09:20:53 2025 user.notice post-tracking-020-status: New public ip detected for wan2 (pppoe-wan2): 92.62.167.224 (First public IP detected)
Tue May 27 09:20:57 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Tue May 27 09:20:57 2025 user.notice omr-bypass: OMR-ByPass is running
Tue May 27 09:21:01 2025 user.notice omr-schedule-010-services: Set firewall on server vps
Tue May 27 09:29:55 2025 daemon.notice netifd: wan1 (11516): udhcpc: sending renew to server 10.xxx.xxx.237
Tue May 27 09:29:55 2025 daemon.notice netifd: wan1 (11516): udhcpc: lease of 10.xxx.xxx.238 obtained from 10.xxx.xxx.237, lease time 1200

VPS:
May 27 09:34:26 OMRVPS ovpn-tun0[486]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:44141
May 27 09:34:26 OMRVPS ovpn-tun0[486]: MANAGEMENT: CMD 'status'
May 27 09:34:26 OMRVPS ovpn-tun0[486]: MANAGEMENT: Client disconnected

moussa11, 2025-05-28:
excellent! much appreciated.

Ysurac, 2025-05-28:
I would need, via SSH on the router, the result of ```uci show omr-tracker``` and of ```uci show shadowsocks-libev``` (remove the key).

moussa11, 2025-05-29:
MPTCP over VPN is enabled. let me know whether you need the results with it disabled. 

root@OpenMPTCProuter:~# uci show omr-tracker
omr-tracker.defaults=defaults
omr-tracker.defaults.enabled='1'
omr-tracker.defaults.hosts='4.2.2.1' '8.8.8.8' '80.67.169.12' '8.8.4.4' '9.9.9.9' '1.0.0.1' '114.114.115.115' '1.2.4.8' '80.67.169.40' '114.114.114.114' '1.1.1.1' '208.67.222.222' '208.67.222.220'
omr-tracker.defaults.hosts6='2606:4700:4700::1111' '2606:4700:4700::1001' '2620:fe::fe' '2620:fe::9' '2001:4860:4860::8888' '2001:4860:4860::8844' '2620:119:35::35' '2620:119:53::53'
omr-tracker.defaults.timeout='2'
omr-tracker.defaults.count='2'
omr-tracker.defaults.tries='3'
omr-tracker.defaults.interval='2'
omr-tracker.defaults.interval_tries='1'
omr-tracker.defaults.type='ping'
omr-tracker.defaults.wait_test='0'
omr-tracker.defaults.server_http_test='0'
omr-tracker.defaults.server_test='0'
omr-tracker.defaults.restart_down='0'
omr-tracker.defaults.mail_alert='0'
omr-tracker.defaults.initial_state='online'
omr-tracker.defaults.family='ipv4'
omr-tracker.defaults.reliability='1'
omr-tracker.defaults.failure_interval='20'
omr-tracker.defaults.tries_up='5'
omr-tracker.proxy=proxy
omr-tracker.proxy.enabled='1'
omr-tracker.proxy.hosts='212.27.48.10' '198.27.92.1' '151.101.129.164' '77.88.55.77' '1.1.1.1' '74.82.42.42' '198.41.212.162'
omr-tracker.proxy.hosts6='2606:4700:4700::1111' '2606:4700:4700::1001' '2620:119:35::35' '2620:119:53::53'
omr-tracker.proxy.timeout='10'
omr-tracker.proxy.tries='3'
omr-tracker.proxy.wait_test='0'
omr-tracker.proxy.interval_tries='1'
omr-tracker.proxy.interval='10'
omr-tracker.proxy.mail_alert='0'
omr-tracker.proxy.initial_state='online'
omr-tracker.proxy.family='ipv4ipv6'
omr-tracker.server=server
omr-tracker.server.enabled='1'
omr-tracker.server.tries='3'
omr-tracker.server.timeout='10'
omr-tracker.server.wait_test='0'
omr-tracker.server.interval='5'
omr-tracker.server.mail_alert='0'
omr-tracker.server.initial_state='online'
omr-tracker.server.type='apiping'
omr-tracker.omrvpn=interface
omr-tracker.omrvpn.type='none'
omr-tracker.omrvpn.timeout='10'
omr-tracker.omrvpn.tries='2'
omr-tracker.omrvpn.interval='5'
omr-tracker.omrvpn.mail_alert='0'
omr-tracker.omrvpn.enabled='1'
omr-tracker.omrvpn.server_http_test='1'
omr-tracker.omrvpn.restart_down='0'
omr-tracker.omrvpn.hosts='4.2.2.1' '8.8.8.8'
omr-tracker.omrvpn.initial_state='online'
omr-tracker.omrvpn.family='ipv4'
omr-tracker.omrvpn.reliability='1'
omr-tracker.omrvpn.count='1'
omr-tracker.omrvpn.failure_interval='3'
omr-tracker.omrvpn.tries_up='1'

root@OpenMPTCProuter:~# uci show shadowsocks-libev
shadowsocks-libev.hi1=ss_redir
shadowsocks-libev.hi1.server='sss0'
shadowsocks-libev.hi1.local_address='::'
shadowsocks-libev.hi1.local_port='1100'
shadowsocks-libev.hi1.mode='tcp_and_udp'
shadowsocks-libev.hi1.timeout='1000'
shadowsocks-libev.hi1.fast_open='1'
shadowsocks-libev.hi1.verbose='0'
[... 66 lines cut ...]
shadowsocks-libev.hi3.no_delay='1'
shadowsocks-libev.hi3.verbose='0'
shadowsocks-libev.hi3.disabled='1'
shadowsocks-libev.hi3.ebpf='0'
shadowsocks-libev.hi4=ss_redir
shadowsocks-libev.hi4.server='sss1'
shadowsocks-libev.hi4.local_address='::'
shadowsocks-libev.hi4.local_port='1101'
shadowsocks-libev.hi4.mode='tcp_and_udp'
shadowsocks-libev.hi4.timeout='1000'
shadowsocks-libev.hi4.fast_open='1'
shadowsocks-libev.hi4.reuse_port='1'
shadowsocks-libev.hi4.syslog='1'
shadowsocks-libev.hi4.mptcp='1'
shadowsocks-libev.hi4.no_delay='1'
shadowsocks-libev.hi4.verbose='0'
shadowsocks-libev.hi4.ebpf='0'
shadowsocks-libev.sss1=server
shadowsocks-libev.sss1.server_port='65101'
shadowsocks-libev.sss1.method='aes-256-gcm'
shadowsocks-libev.sss1.obfs='0'
shadowsocks-libev.sss1.disabled='1'
shadowsocks-libev.sss1.key='<key_removed>'
shadowsocks-libev.sss1.obfs_host='134.xxx.xxx.58'
shadowsocks-libev.sss1.server='<my_domain_removed>'

moussa11, 2025-06-03:
I am guessing when using MPTCP over VPN shadowsocks get disabled! not sure. but here I have disabled MPTCP over VPN: 

root@OpenMPTCProuter:~# uci show omr-tracker
omr-tracker.defaults=defaults
omr-tracker.defaults.enabled='1'
omr-tracker.defaults.hosts='4.2.2.1' '8.8.8.8' '80.67.169.12' '8.8.4.4' '9.9.9.9' '1.0.0.1' '114.114.115.115' '1.2.4.8' '80.67.169.40' '114.114.114.114' '1.1.1.1' '208.67.222.222' '208.67.222.220'
omr-tracker.defaults.hosts6='2606:4700:4700::1111' '2606:4700:4700::1001' '2620:fe::fe' '2620:fe::9' '2001:4860:4860::8888' '2001:4860:4860::8844' '2620:119:35::35' '2620:119:53::53'
omr-tracker.defaults.timeout='2'
omr-tracker.defaults.count='2'
omr-tracker.defaults.tries='3'
omr-tracker.defaults.interval='2'
omr-tracker.defaults.interval_tries='1'
omr-tracker.defaults.type='ping'
omr-tracker.defaults.wait_test='0'
omr-tracker.defaults.server_http_test='0'
omr-tracker.defaults.server_test='0'
omr-tracker.defaults.restart_down='0'
omr-tracker.defaults.mail_alert='0'
omr-tracker.defaults.initial_state='online'
omr-tracker.defaults.family='ipv4'
omr-tracker.defaults.reliability='1'
omr-tracker.defaults.failure_interval='20'
omr-tracker.defaults.tries_up='5'
omr-tracker.proxy=proxy
omr-tracker.proxy.enabled='1'
omr-tracker.proxy.hosts='212.27.48.10' '198.27.92.1' '151.101.129.164' '77.88.55.77' '1.1.1.1' '74.82.42.42' '198.41.212.162'
omr-tracker.proxy.hosts6='2606:4700:4700::1111' '2606:4700:4700::1001' '2620:119:35::35' '2620:119:53::53'
omr-tracker.proxy.timeout='10'
omr-tracker.proxy.tries='3'
omr-tracker.proxy.wait_test='0'
omr-tracker.proxy.interval_tries='1'
omr-tracker.proxy.interval='10'
omr-tracker.proxy.mail_alert='0'
omr-tracker.proxy.initial_state='online'
omr-tracker.proxy.family='ipv4ipv6'
omr-tracker.server=server
omr-tracker.server.enabled='1'
omr-tracker.server.tries='3'
omr-tracker.server.timeout='10'
omr-tracker.server.wait_test='0'
omr-tracker.server.interval='5'
omr-tracker.server.mail_alert='0'
omr-tracker.server.initial_state='online'
omr-tracker.server.type='apiping'
omr-tracker.omrvpn=interface
omr-tracker.omrvpn.type='none'
omr-tracker.omrvpn.timeout='10'
omr-tracker.omrvpn.tries='2'
omr-tracker.omrvpn.interval='5'
omr-tracker.omrvpn.mail_alert='0'
omr-tracker.omrvpn.enabled='1'
omr-tracker.omrvpn.server_http_test='1'
omr-tracker.omrvpn.restart_down='0'
omr-tracker.omrvpn.hosts='4.2.2.1' '8.8.8.8'
omr-tracker.omrvpn.initial_state='online'
omr-tracker.omrvpn.family='ipv4'
omr-tracker.omrvpn.reliability='1'
omr-tracker.omrvpn.count='1'
omr-tracker.omrvpn.failure_interval='3'
omr-tracker.omrvpn.tries_up='1'

root@OpenMPTCProuter:~# uci show shadowsocks-libev
shadowsocks-libev.hi1=ss_redir
shadowsocks-libev.hi1.server='sss0'
shadowsocks-libev.hi1.local_address='::'
shadowsocks-libev.hi1.local_port='1100'
shadowsocks-libev.hi1.mode='tcp_and_udp'
shadowsocks-libev.hi1.timeout='1000'
shadowsocks-libev.hi1.fast_open='1'
shadowsocks-libev.hi1.verbose='0'
[... 66 lines cut ...]
shadowsocks-libev.hi3.no_delay='1'
shadowsocks-libev.hi3.verbose='0'
shadowsocks-libev.hi3.disabled='1'
shadowsocks-libev.hi3.ebpf='0'
shadowsocks-libev.hi4=ss_redir
shadowsocks-libev.hi4.server='sss1'
shadowsocks-libev.hi4.local_address='::'
shadowsocks-libev.hi4.local_port='1101'
shadowsocks-libev.hi4.mode='tcp_and_udp'
shadowsocks-libev.hi4.timeout='1000'
shadowsocks-libev.hi4.fast_open='1'
shadowsocks-libev.hi4.reuse_port='1'
shadowsocks-libev.hi4.syslog='1'
shadowsocks-libev.hi4.mptcp='1'
shadowsocks-libev.hi4.no_delay='1'
shadowsocks-libev.hi4.verbose='0'
shadowsocks-libev.hi4.ebpf='0'
shadowsocks-libev.sss1=server
shadowsocks-libev.sss1.server_port='65101'
shadowsocks-libev.sss1.method='aes-256-gcm'
shadowsocks-libev.sss1.obfs='0'
shadowsocks-libev.sss1.disabled='1'
shadowsocks-libev.sss1.key='mykey'
shadowsocks-libev.sss1.obfs_host='134.xxx.xxx.58'
shadowsocks-libev.sss1.server='mydomain'

moussa11, 2025-06-20:
Hi, any update?

Ysurac, 2025-06-20:
Can you update to latest snapshot and try Shadowsocks-Rust ?

moussa11, 2025-06-20:
The version I have running is: [openmptcprouter v0.63-snapshot-6.6 r0+28627-0b392b925f]. Yet I see the on the folder that the date was updated to June 19, 2025. Should I be re-installing the same version?

moussa11, 2025-06-20:
downloaded snapshot dated June 19,2025 and installed with keep current settings. 

seems the issue still there: 

Fri Jun 20 09:28:30 2025 daemon.notice openvpn(omr)[9012]: OpenVPN 2.6.14 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Fri Jun 20 09:28:30 2025 daemon.notice openvpn(omr)[9012]: library versions: OpenSSL 3.0.16 11 Feb 2025, LZO 2.10
Fri Jun 20 09:28:30 2025 daemon.notice openvpn(omr)[9012]: DCO version: N/A
Fri Jun 20 09:28:30 2025 daemon.warn openvpn(omr)[9012]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Fri Jun 20 09:28:30 2025 daemon.warn openvpn(omr)[9012]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Fri Jun 20 09:28:30 2025 user.notice post-tracking-002-error: DNS flush
Fri Jun 20 09:28:30 2025 daemon.notice openvpn(omr)[9012]: TCP/UDP: Preserving recently used remote address: [AF_INET]134.xxx.xxx.58:65301
Fri Jun 20 09:28:30 2025 daemon.notice openvpn(omr)[9012]: Attempting to establish TCP connection with [AF_INET]134.xxx.xxx.58:65301
Fri Jun 20 09:28:30 2025 daemon.notice openvpn(omr)[9012]: TCP connection established with [AF_INET]134.xxx.xxx.58:65301
Fri Jun 20 09:28:30 2025 daemon.warn openvpn(omr)[9012]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Not supported (errno=95)
Fri Jun 20 09:28:30 2025 daemon.notice openvpn(omr)[9012]: TCPv4_CLIENT link local: (not bound)
Fri Jun 20 09:28:30 2025 daemon.notice openvpn(omr)[9012]: TCPv4_CLIENT link remote: [AF_INET]134.xxx.xxx.58:65301
Fri Jun 20 09:28:31 2025 daemon.notice netifd: Interface 'omrvpn' is disabled
Fri Jun 20 09:28:31 2025 daemon.notice netifd: Network device 'tun0' link is down
Fri Jun 20 09:28:31 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Fri Jun 20 09:28:31 2025 daemon.notice netifd: omrvpn (19072): udhcpc: read error: Network is down, reopening socket
Fri Jun 20 09:28:31 2025 daemon.notice netifd: omrvpn (19072): udhcpc: bind: No such device
Fri Jun 20 09:28:31 2025 daemon.notice netifd: Interface 'omrvpn' is now down
Fri Jun 20 09:28:31 2025 daemon.notice openvpn(omr)[9012]: [server] Peer Connection Initiated with [AF_INET]134.xxx.xxx.58:65301
Fri Jun 20 09:28:31 2025 daemon.notice netifd: Interface 'omrvpn' is enabled
Fri Jun 20 09:28:31 2025 daemon.notice netifd: Network device 'tun0' link is up
Fri Jun 20 09:28:31 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Fri Jun 20 09:28:31 2025 daemon.notice netifd: Interface 'omrvpn' is setting up now
Fri Jun 20 09:28:31 2025 daemon.notice openvpn(omr)[9012]: TUN/TAP device tun0 opened
Fri Jun 20 09:28:31 2025 daemon.notice openvpn(omr)[9012]: net_iface_mtu_set: mtu 1420 for tun0
Fri Jun 20 09:28:31 2025 daemon.notice openvpn(omr)[9012]: net_iface_up: set tun0 up
Fri Jun 20 09:28:31 2025 daemon.notice openvpn(omr)[9012]: net_addr_v4_add: 10.xxx.xxx.3/24 dev tun0
Fri Jun 20 09:28:31 2025 daemon.notice openvpn(omr)[9012]: /usr/libexec/openvpn-hotplug up omr tun0 1420 0 10.xxx.xxx.3 255.255.255.0 init
Fri Jun 20 09:28:31 2025 daemon.notice netifd: omrvpn (9172): udhcpc: started, v1.36.1
Fri Jun 20 09:28:31 2025 user.notice NET: hotplug (iface): action='add' interface='tun0'
Fri Jun 20 09:28:31 2025 daemon.notice netifd: omrvpn (9172): udhcpc: broadcasting discover
Fri Jun 20 09:28:31 2025 daemon.info ModemManager[9124]: hotplug: add network interface tun0: event processed
Fri Jun 20 09:28:33 2025 user.notice post-tracking-003-up: Change in routes, set ip route replace default scope global nexthop via 10.117.22.120 dev eth1 weight 100 nexthop via 92.62.172.254 dev pppoe-wan2 weight 1 (omrvpn_intf: tun0)
Fri Jun 20 09:28:34 2025 daemon.notice netifd: omrvpn (9172): udhcpc: broadcasting discover
Fri Jun 20 09:28:34 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.xxx.xxx.1 dev tun0 (was 10.117.22.120)
Fri Jun 20 09:28:36 2025 daemon.notice openvpn(omr)[9012]: Initialization Sequence Completed
Fri Jun 20 09:28:36 2025 daemon.notice netifd: omrvpn (9172): udhcpc: broadcasting discover
Fri Jun 20 09:29:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Fri Jun 20 09:29:00 2025 user.notice omr-tracker: Launching...
Fri Jun 20 09:29:07 2025 user.notice omr-tracker: Launched
Fri Jun 20 09:30:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Fri Jun 20 09:30:00 2025 user.notice omr-tracker: Launching...
Fri Jun 20 09:30:07 2025 user.notice omr-tracker: Launched

Ysurac, 2025-06-20:
For MPTCP over VPN: none
It only add support of OpenVPN for anything that is not TCP, all the "openvpn(omr)" entries you can see in log.

moussa11, 2025-06-21:
update with newest compile: 

Sat Jun 21 01:13:55 2025 daemon.info urandom_seed[30624]: Seed saved (/etc/urandom.seed)
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5591] N: ttyd 1.7.3 (libwebsockets 4.3.3-unknown)
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5592] N: tty configuration:
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5593] N:   start command: /bin/login
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5593] N:   close signal: SIGHUP (1)
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5594] N:   terminal type: xterm-256color
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5645] N:    /usr/lib/libwebsockets-evlib_uv.so
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5647] N: lws_create_context: LWS: 4.3.3-unknown, NET CLI SRV H1 H2 WS ConMon IPV6-off
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5649] N: elops_init_pt_uv:  Using foreign event loop...
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5650] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5652] N: __lws_lc_tag:  ++ [vh|0|netlink] (1)
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5654] N: __lws_lc_tag:  ++ [vh|1|default|eth0|eth0|7681] (2)
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5664] N: [vh|1|default|eth0|eth0|7681]: lws_socket_bind: source ads 192.168.100.1
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5666] N: __lws_lc_tag:  ++ [wsi|1|listen|default|eth0|7681] (2)
Sat Jun 21 01:13:55 2025 daemon.notice ttyd[30578]: [2025/06/21 01:13:55:5667] N:  Listening on port: 7681
Sat Jun 21 01:13:55 2025 daemon.info xray: No xray instance enabled, skip transparent proxy.
Sat Jun 21 01:13:55 2025 daemon.notice procd: /etc/rc.d/S99xray: [info] No xray instance enabled, skip transparent proxy.
Sat Jun 21 01:13:55 2025 daemon.info procd: - init complete -
Sat Jun 21 01:14:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sat Jun 21 01:14:00 2025 user.notice omr-tracker: Launching...
Sat Jun 21 01:14:06 2025 user.notice omr-tracker: Launched
Sat Jun 21 01:14:07 2025 daemon.info unbound: [29712:0] info: generate keytag query _ta-4f66. NULL IN
Sat Jun 21 01:14:10 2025 user.notice post-tracking-003-up: Interface route not yet set, set route ip r add default via 10.xxx.xxx.120 dev eth1 metric 16
Sat Jun 21 01:14:10 2025 user.notice post-tracking-003-up: Local IP changed for pppoe-wan2 (wan2) reload Multipath
Sat Jun 21 01:14:10 2025 user.notice post-tracking-003-up: Reload MPTCP config for pppoe-wan2
Sat Jun 21 01:14:10 2025 user.notice post-tracking-003-up: Multipath eth1 (wan1) switched to on (from off)
Sat Jun 21 01:14:13 2025 user.notice post-tracking-020-status: New public ip detected for wan2 (pppoe-wan2): 92.62.167.14 (First public IP detected)
Sat Jun 21 01:15:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sat Jun 21 01:15:00 2025 user.notice omr-tracker: Launching...
Sat Jun 21 01:15:06 2025 user.notice omr-tracker: Launched
Sat Jun 21 01:16:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sat Jun 21 01:16:00 2025 user.notice omr-tracker: Launching...
Sat Jun 21 01:16:06 2025 user.notice omr-tracker: Launched
Sat Jun 21 01:17:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sat Jun 21 01:17:00 2025 user.notice omr-tracker: Launching...
Sat Jun 21 01:17:06 2025 user.notice omr-tracker: Launched
Sat Jun 21 01:18:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sat Jun 21 01:18:00 2025 user.notice omr-tracker: Launching...
Sat Jun 21 01:18:06 2025 user.notice omr-tracker: Launched
Sat Jun 21 01:19:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sat Jun 21 01:19:00 2025 user.notice omr-tracker: Launching...
Sat Jun 21 01:19:06 2025 user.notice omr-tracker: Launched
Sat Jun 21 01:20:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sat Jun 21 01:20:00 2025 user.notice omr-tracker: Launching...
Sat Jun 21 01:20:06 2025 user.notice omr-tracker: Launched
Sat Jun 21 01:21:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sat Jun 21 01:21:00 2025 user.notice omr-tracker: Launching...
Sat Jun 21 01:21:06 2025 user.notice omr-tracker: Launched
Sat Jun 21 01:22:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sat Jun 21 01:22:00 2025 user.notice omr-tracker: Launching...
Sat Jun 21 01:22:06 2025 user.notice omr-tracker: Launched
Sat Jun 21 01:22:52 2025 daemon.notice netifd: wan1 (10111): udhcpc: sending renew to server 10.xxx.xxx.120
Sat Jun 21 01:22:52 2025 daemon.notice netifd: wan1 (10111): udhcpc: lease of 10.xxx.xxx.119 obtained from 10.xxx.xxx.120, lease time 1200
[... cut ...]
Sat Jun 21 01:24:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sat Jun 21 01:24:00 2025 user.notice omr-tracker: Launching...
Sat Jun 21 01:24:06 2025 user.notice omr-tracker: Launched
Sat Jun 21 01:25:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Sat Jun 21 01:25:00 2025 user.notice omr-tracker: Launching...
Sat Jun 21 01:25:06 2025 user.notice omr-tracker: Launched

moussa11, 2025-06-21:
root@OpenMPTCProuter:~# uci show xray
xray.main=xray
xray.main.xray_file='/usr/bin/xray'
xray.main.mem_percentage='0'
xray.main.loglevel='error'
xray.main.access_log='/dev/null'
xray.main.error_log='/dev/null'
xray.main.enabled='0'
xray.main.outbounds='omrout' 'omrexit'
xray.main.inbounds='omr' 'omrtest' 'omr6'
xray.main_dns=dns
xray.main_dns.hosts='example.com|127.0.0.1'
xray.main_dns.enabled='0'
xray.main_policy=policy
xray.main_policy.enabled='1'
xray.main_policy.levels='policy_level_0'
xray.policy_level_0=policy_level
xray.policy_level_0.level='0'
xray.policy_level_0.handshake='4'
xray.policy_level_0.conn_idle='1200'
xray.policy_level_0.uplink_only='0'
xray.policy_level_0.downlink_only='0'
xray.policy_level_0.buffer_size='512'
xray.main_transparent_proxy=transparent_proxy
xray.main_transparent_proxy.proxy_mode='default'
xray.main_transparent_proxy.apnic_delegated_mirror='apnic'
xray.main_transparent_proxy.gfwlist_mirror='github'
xray.main_transparent_proxy.redirect_udp='0'
xray.main_transparent_proxy.redirect_port='1100'
xray.omrout=outbound
xray.omrout.tag='omrout_tunnel'
xray.omrout.protocol='shadowsocks'
xray.omrout.s_vmess_port='65250'
xray.omrout.s_vmess_user_security='aes-128-gcm'
xray.omrout.s_vmess_user_alter_id='0'
xray.omrout.s_vless_port='65248'
xray.omrout.s_vless_user_security='aes-128-gcm'
xray.omrout.s_vless_user_encryption='none'
xray.omrout.s_vless_user_alter_id='0'
xray.omrout.s_trojan_port='65249'
xray.omrout.s_trojan_user_security='aes-128-gcm'
xray.omrout.s_trojan_user_encryption='none'
xray.omrout.s_trojan_user_alter_id='0'
xray.omrout.s_socks_port='65251'
xray.omrout.s_socks_user_security='aes-128-gcm'
xray.omrout.s_socks_user_encryption='none'
xray.omrout.s_socks_user_alter_id='0'
xray.omrout.ss_network='tcp'
xray.omrout.ss_security='tls'
xray.omrout.ss_tls_allow_insecure='1'
xray.omrout.ss_tls_disable_system_root='1'
xray.omrout.ss_tls_cert_usage='verify'
xray.omrout.ss_tls_cert_file='/etc/luci-uploads/client.crt'
xray.omrout.ss_tls_key_file='/etc/luci-uploads/client.key'
xray.omrout.s_shadowsocks_port='65252'
xray.omrout.mux_concurrency='8'
xray.omrout.ss_sockopt_mptcp='1'
xray.omrout.s_vless_reality_port='443'
xray.omrout.s_vless_reality_flow='xtls-rprx-vision'
xray.omrout.s_vless_reality_user_security='aes-128-gcm'
xray.omrout.s_vless_reality_user_encryption='none'
xray.omrout.s_vless_reality_user_alter_id='0'
xray.omrout.s_shadowsocks_email='openmptcprouter'
xray.omrout.s_vmess_address='134.xxx.xxx.58'
xray.omrout.s_vless_address='134.xxx.xxx.58'
xray.omrout.s_vless_reality_address='134.xxx.xxx.58'
xray.omrout.s_trojan_address='134.xxx.xxx.58'
xray.omrout.s_socks_address='134.xxx.xxx.58'
xray.omrout.s_shadowsocks_address='134.xxx.xxx.58'
xray.omrout.s_shadowsocks_method='2022-blake3-aes-256-gcm'
[... 118 lines cut ...]
shadowsocks-rust.tracker_sss0.keep_alive='15'
shadowsocks-rust.tracker_sss0.fast_open='0'
shadowsocks-rust.tracker_sss0.syslog='0'
shadowsocks-rust.tracker_sss0.reuse_port='1'
shadowsocks-rust.tracker_sss0.mptcp='1'
shadowsocks-rust.tracker_sss0.verbose='0'
shadowsocks-rust.mptcpovervpn=server
shadowsocks-rust.mptcpovervpn.server_port='65280'
shadowsocks-rust.mptcpovervpn.password='password'
shadowsocks-rust.mptcpovervpn.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.mptcpovervpn.server='10.xxx.xxx.1'
shadowsocks-rust.mptcpovervpn.disabled='0'
shadowsocks-rust.hivpn=ss_redir
shadowsocks-rust.hivpn.server='mptcpovervpn'
shadowsocks-rust.hivpn.local_address='0.0.0.0'
shadowsocks-rust.hivpn.local_port='1101'
shadowsocks-rust.hivpn.mode='tcp_and_udp'
shadowsocks-rust.hivpn.timeout='1000'
shadowsocks-rust.hivpn.fast_open='1'
shadowsocks-rust.hivpn.verbose='0'
shadowsocks-rust.hivpn.syslog='1'
shadowsocks-rust.hivpn.reuse_port='1'
shadowsocks-rust.hivpn.mptcp='1'
shadowsocks-rust.hivpn.ipv6_first='1'
shadowsocks-rust.hivpn.no_delay='1'

Ysurac, 2025-07-11:
It's not solved. I'm not able to reproduce this on several platform.
Interesting that you also have a R5S on kernel 6.6, did you try with latest snapshot using kernel 6.12 (it's now supported) ? https://snapshots.openmptcprouter.com/6.12/r5s/targets/rockchip/armv8/

Nicras, 2025-07-11:
I tested a bit yesterday and with v0.63-6.6 it was not as bad as before but still, when I put load onto the aggregation it works well until the moment when the speedtest switches to upload. then the system gets unresponsive for a few seconds and the tunnel drops. after a few seconds the system gets back. 
Will try kernel 6.12 now and report back.
As I dont have a clue, is 6.12 better than 6.6?

Ysurac, 2025-07-11:
Yes. And maybe can solve some hardware issues on R5S...

Nicras, 2025-07-11:
kernel 6.12 is up and running and the problem is gone while I still have high latency during speed test. I didn't change anything regarding proxy or vpn.

[image]

I have starlink as master and telekom 5G enabled for MPTCP

Since it is working and I read you recommend Shadowsocks Rust 2022 as proxy, I think I will leave it as it is, although it is 100Mbps slower than with XRay VLESS.

Ysurac, 2025-07-11:
Should work. What problem you have ? on a domain or ip ?

Ysurac, 2025-07-11:
It's a bug, will be fixed in next snapshot.

moussa11, 2025-09-04:
Hi, 

I have finally got around start with fresh image on the SD card for R5S. Completed only basic settings for LAN, and both WANs. 

I can't figure out what could be the issue on my connections: 

91.232.xxx.xx2: is wan2 - pppoe dsl non-master 

VPS: 

Sep 04 08:09:34 OMRVPS ovpn-tun0[492]: 91.232.xxx.xx2:35273 TLS: move_session: dest=TM_ACTIVE src=TM_INITIAL rei>
Sep 04 08:09:34 OMRVPS ovpn-tun0[492]: 91.232.xxx.xx2:35273 WARNING: Failed to stat CRL file, not reloading CRL.
Sep 04 08:09:34 OMRVPS ovpn-tun0[492]: 91.232.1xxx.xx2:35273 TLS: tls_multi_process: initial untrusted session pr>
Sep 04 08:09:34 OMRVPS ovpn-tun0[492]: 91.232.xxx.xx2:35273 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256>
Sep 04 08:09:34 OMRVPS ovpn-tun0[492]: 91.232.xxx.xx2:35273 [openmptcprouter] Peer Connection Initiated with [AF>
Sep 04 08:09:34 OMRVPS ovpn-tun0[492]: openmptcprouter/91.232.xxx.xx2:35273 MULTI_sva: pool returned IPv4=10.255>
Sep 04 08:09:34 OMRVPS ovpn-tun0[492]: openmptcprouter/91.232.xxx.xx2:35273 Could not access file 'ccd/openmptcp>
Sep 04 08:09:34 OMRVPS ovpn-tun0[492]: openmptcprouter/91.232.xxx.xx2:35273 Could not access file 'ccd/DEFAULT':>
Sep 04 08:09:34 OMRVPS ovpn-tun0[492]: openmptcprouter/91.232.xxx.xx2:35273 MULTI: Learn: 10.255.252.2 -> openmp>
Sep 04 08:09:34 OMRVPS ovpn-tun0[492]: openmptcprouter/91.232.xxx.xx2:35273 MULTI: primary virtual IP for openmp>
Sep 04 08:09:34 OMRVPS ovpn-tun0[492]: openmptcprouter/91.232.xxx.xx2:35273 SENT CONTROL [openmptcprouter]: 'PUS>
Sep 04 08:09:35 OMRVPS ovpn-tun0[492]: openmptcprouter/91.232.xxx.xx2:35273 Data Channel: cipher 'AES-256-GCM', >
Sep 04 08:09:35 OMRVPS ovpn-tun0[492]: openmptcprouter/91.232.xxx.xx2:35273 Timers: ping 10, ping-restart 480
Sep 04 08:09:35 OMRVPS ovpn-tun0[492]: openmptcprouter/91.232.xxx.xx2:35273 Protocol options: protocol-flags cc->
Sep 04 08:09:49 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:44233
Sep 04 08:09:49 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:09:49 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:09:53 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:62683
Sep 04 08:09:53 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:09:53 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:09:54 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:62685
Sep 04 08:09:54 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:09:54 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:09:55 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:62695
Sep 04 08:09:55 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:09:55 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:09:55 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:62709
Sep 04 08:09:55 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:09:55 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:09:56 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:62715
Sep 04 08:09:56 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:09:56 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:10:45 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:54181
Sep 04 08:10:45 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:10:45 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:10:53 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:50965
Sep 04 08:10:53 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:10:53 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:11:14 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:45343
Sep 04 08:11:14 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:11:14 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:13:04 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:49933
Sep 04 08:13:04 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:13:04 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:13:25 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:32913
Sep 04 08:13:25 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:13:25 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:13:45 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:22603
Sep 04 08:13:45 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:13:45 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:15:03 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:21341
Sep 04 08:15:03 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:15:03 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
Sep 04 08:15:24 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:53843
Sep 04 08:15:24 OMRVPS ovpn-tun0[492]: MANAGEMENT: CMD 'status'
Sep 04 08:15:24 OMRVPS ovpn-tun0[492]: MANAGEMENT: Client disconnected
lines 56-100/100 (END)

OMR log: 
Thu Sep  4 08:21:28 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Sep  4 08:21:29 2025 user.notice omr-bypass: OMR-ByPass is running
Thu Sep  4 08:21:29 2025 daemon.notice ttyd[25159]: [2025/09/04 08:21:29:3913] N: ttyd 1.7.7-unknown (libwebsockets 4.3.3-unknown)
[... cut ...]
Thu Sep  4 08:22:00 2025 user.notice omr-schedule-010-services: Set firewall on server vps
Thu Sep  4 08:29:55 2025 authpriv.info dispatcher.uc: luci: accepted login on / for root from 192.168.100.134
Thu Sep  4 08:30:02 2025 daemon.info unbound: [23367:0] info: generate keytag query _ta-4f66. NULL IN
Thu Sep  4 08:30:48 2025 daemon.notice netifd: wan1 (5516): udhcpc: sending renew to server 10.119.xx.xx5
Thu Sep  4 08:30:49 2025 daemon.notice netifd: wan1 (5516): udhcpc: lease of 10.119.xx.xx6 obtained from 10.119.xx.xx5, lease time 1200

closed 2025-09-19 by moussa11