Ysurac/openmptcprouter #3944: Speed test nott fully bonded

mvmaksnet, 2025-05-14:
Hello.
I have screenshot of separate 3 connections download and upload, and mptco speedtest which in sum is lower than it should be if you sum 3 operator in total dwonload and upload should be more than in mptcp speedtest.
see picture
Is this normall for mptcp?
I have 1 Gbps up and down on VPS.

[image]

Ysurac, 2025-05-14:
Not that much. You should get between at least 90% (due to encryption for most part).
Can you complete template ? I need OMR version, VPS script version, arch used,... and also a screenshot of System->OpenMPTCProuter, "Status" tab

Can you also try with ```omr-test-speed``` from OMR ?

And if possible try latest snapshot from https://snapshots.openmptcprouter.com/6.6/

mvmaksnet, 2025-05-14:
server:
6.12.15-x64v3-xanmod1 0.1032-test

server Linux:
Debian 11 

router:
v0.62rc1-6.6 r0+28431-92e020

OMR speedtest:

root@OpenMPTCProuter:~# omr-test-speed
Select best test server...
host: scaleway.testdebit.info - ping: 74
ping: bordeaux.testdebit.info: Name does not resolve
host: bordeaux.testdebit.info - ping:
ping: aix-marseille.testdebit.info: Name does not resolve
host: aix-marseille.testdebit.info - ping:
ping: lyon.testdebit.info: Name does not resolve
host: lyon.testdebit.info - ping:
ping: lille.testdebit.info: Name does not resolve
host: lille.testdebit.info - ping:
ping: paris.testdebit.info: Name does not resolve
host: paris.testdebit.info - ping:
host: appliwave.testdebit.info - ping: 65
host: speedtest.frankfurt.linode.com - ping: 65
host: speedtest.tokyo2.linode.com - ping: 301
host: speedtest.singapore.linode.com - ping: 304
host: speedtest.newark.linode.com - ping: 136
host: speedtest.atlanta.linode.com - ping: 152
host: speedtest.dallas.linode.com - ping: 169
host: speedtest.fremont.linode.com - ping: 200
ping: ipv4.bouygues.testdebit.info: Name does not resolve
host: ipv4.bouygues.testdebit.info - ping:
host: par.download.datapacket.com - ping: 54
host: nyc.download.datapacket.com - ping: 128
host: ams.download.datapacket.com - ping: 65
host: fra.download.datapacket.com - ping: 52
host: lon.download.datapacket.com - ping: 64
host: mad.download.datapacket.com - ping: 80
host: prg.download.datapacket.com - ping: 56
host: sto.download.datapacket.com - ping: 74
host: vie.download.datapacket.com - ping: 52
host: war.download.datapacket.com - ping: 54
host: atl.download.datapacket.com - ping: 155
host: chi.download.datapacket.com - ping: 155
host: lax.download.datapacket.com - ping: 198
host: mia.download.datapacket.com - ping: 200
host: nyc.download.datapacket.com - ping: 125
host: speedtest.milkywan.fr - ping: 82
host: rbx.proof.ovh.net - ping: 63
host: gra.proof.ovh.net - ping: 70
host: sbg.proof.ovh.net - ping: 53
host: bhs.proof.ovh.ca - ping: 149
host: vin.proof.ovh.us - ping: 143
host: hil.proof.ovh.us - ping: 214
host: speedtest.tele2.net - ping: 95
host: la.speedtest.clouvider.net - ping: 202
host: atl.speedtest.clouvider.net - ping: 171
host: nyc.speedtest.clouvider.net - ping: 152
host: lon.speedtest.clouvider.net - ping: 67
host: man.speedtest.clouvider.net - ping: 79
host: ams.speedtest.clouvider.net - ping: 53
host: fra.speedtest.clouvider.net - ping: 55
host: dal.speedtest.clouvider.net - ping: 182
host: ash.speedtest.clouvider.net - ping: 148
host: phx.speedtest.clouvider.net - ping: 199
host: chi.speedtest.clouvider.net - ping: 148
Best server is http://fra.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 10 9536M   10 1006M    0     0  5359k      0  0:30:22  0:03:12  0:27:10 6320k^C

[image]