Ysurac/openmptcprouter #3946: Wifi Crash on Intel PCI Wi-Fi 6 AX iwlwifi

gsxryan, 2025-05-19:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

Support for modern Wireless chip sets will help future proof use cases for modern, fast wifi needs, supporting AC, and AX.

[openmptcprouter v0.62-6.6 r0+28431-92e020b50f](https://github.com/ysurac/openmptcprouter)
Kernel Version：6.6.73

**Describe why you are asking for this support?**

I cannot get the Intel Corporation Wi-Fi 6 AX210/AX211/AX411 160Mhz (rev 1a) to work.  It is attached on an x86 system over PCI.  If i attempt to Enable the wifi AP on the AX interface, it crashes both wireless interfaces.

[image]

**Additional context**

I have tested a USB 2.0 with more legacy Wifi N.  It works fine.

The following software modules have been installed
iwlwifi-firmware-ax210	20241110-r1
iwlwifi-firmware-ax411	20241110-r1
kmod-iwlwifi	6.6.73.6.12.6-r1

System Log: 
```
Mon May 19 02:43:48 2025 kern.info kernel: [ 1439.242170] rt2800usb 9-1:1.0 phy1-ap0: entered allmulticast mode
Mon May 19 02:43:48 2025 kern.info kernel: [ 1439.242803] rt2800usb 9-1:1.0 phy1-ap0: entered promiscuous mode
Mon May 19 02:43:48 2025 daemon.notice netifd: Wireless device 'radio2' is now up
Mon May 19 02:43:48 2025 daemon.info ModemManager[3256]: hotplug: add network interface phy1-ap0: event processed
Mon May 19 02:43:48 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 02:43:48 2025 user.notice omr-bypass: OMR-ByPass is running
Mon May 19 02:43:48 2025 kern.info kernel: [ 1439.651935] hostapd[3067]: segfault at 0 ip 000055beb75b4c7c sp 00007ffc6c674768 error 4 in wpad[55beb74a0000+117000] likely on CPU 0 (core 0, socket 0)
Mon May 19 02:43:48 2025 kern.info kernel: [ 1439.653082] Code: 0a 0f b7 d5 48 89 df c7 44 24 0a ff ff ff ff 66 c7 44 24 0e ff ff e8 a9 f6 ff ff 48 89 df e8 b7 1d f1 ff 48 83 c4 18 5b 5d c3 <48> 8b 07 48 89 7e 08 48 89 06 48 89 70 08 48 89 37 c3 48 8b 47 10
Mon May 19 02:43:49 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon May 19 02:43:49 2025 daemon.warn netifd: Wireless device 'radio2' setup failed, retry=3
Mon May 19 02:43:49 2025 daemon.notice netifd: radio2 (3994): Command failed: Not found
Mon May 19 02:43:49 2025 daemon.notice wpa_supplicant[5035]: Set new config for phy phy1
Mon May 19 02:43:49 2025 kern.info kernel: [ 1440.353762] br-lan: port 2(phy1-ap0) entered disabled state
Mon May 19 02:43:49 2025 kern.info kernel: [ 1440.390733] rt2800usb 9-1:1.0 phy1-ap0 (unregistering): left allmulticast mode
Mon May 19 02:43:49 2025 kern.info kernel: [ 1440.391545] rt2800usb 9-1:1.0 phy1-ap0 (unregistering): left promiscuous mode
Mon May 19 02:43:49 2025 kern.info kernel: [ 1440.392308] br-lan: port 2(phy1-ap0) entered disabled state
Mon May 19 02:43:49 2025 user.notice NET: hotplug (iface): action='remove' interface='phy1-ap0'
Mon May 19 02:43:49 2025 daemon.info ModemManager[4026]: hotplug: remove network interface phy1-ap0: event processed
Mon May 19 02:43:49 2025 daemon.notice [21805]: <msg> [base-manager] port phy1-ap0 released by device '/sys/devices/pci0000:00/0000:00:1e.0/0000:05:02.0/0000:07:1b.0/usb9/9-1'
Mon May 19 02:43:49 2025 daemon.notice [21805]: <msg> [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:1e.0/0000:05:02.0/0000:07:1b.0/usb9/9-1': Operation was cancelled
Mon May 19 02:43:49 2025 daemon.notice netifd: Wireless device 'radio2' is now down
Mon May 19 02:43:49 2025 daemon.notice netifd: radio2 (4051): WARNING: Variable 'data' does not exist or is not an array/object
Mon May 19 02:43:49 2025 daemon.notice netifd: radio2 (4051): Command failed: Not found
Mon May 19 02:43:49 2025 daemon.notice wpa_supplicant[5035]: Set new config for phy phy1
Mon May 19 02:43:49 2025 daemon.notice wpa_supplicant[5035]: Set new config for phy phy1
Mon May 19 02:43:49 2025 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Mon May 19 02:43:49 2025 daemon.notice hostapd: Restart interface for phy phy1
Mon May 19 02:43:49 2025 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 hw_mode=a beacon_int=100 stationary_ap=1 chanlist=36 #num_global_macaddr=1 #macaddr_base= ieee80211n=1 ht_coex=0 ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][RX-STBC1] channel=36  interface=phy1-ap0 bssid=00:c0:ca:8f:ff:59 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 auth_algs=1 wpa=0 ssid=OpenWrt bridge=br-lan wds_bridge= snoop_iface=br-lan qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56 #default_macaddr nas_identifier=00c0ca8fff59  (phy phy1) --> new PHY
Mon May 19 02:43:49 2025 kern.info kernel: [ 1440.748385] br-lan: port 2(phy1-ap0) entered blocking state
Mon May 19 02:43:49 2025 kern.info kernel: [ 1440.749086] br-lan: port 2(phy1-ap0) entered disabled state
```

Kernel Log: 
```
[ 1439.242170] rt2800usb 9-1:1.0 phy1-ap0: entered allmulticast mode
[ 1439.242803] rt2800usb 9-1:1.0 phy1-ap0: entered promiscuous mode
[ 1439.651935] hostapd[3067]: segfault at 0 ip 000055beb75b4c7c sp 00007ffc6c674768 error 4 in wpad[55beb74a0000+117000] likely on CPU 0 (core 0, socket 0)
[ 1439.653082] Code: 0a 0f b7 d5 48 89 df c7 44 24 0a ff ff ff ff 66 c7 44 24 0e ff ff e8 a9 f6 ff ff 48 89 df e8 b7 1d f1 ff 48 83 c4 18 5b 5d c3 <48> 8b 07 48 89 7e 08 48 89 06 48 89 70 08 48 89 37 c3 48 8b 47 10
[ 1440.353762] br-lan: port 2(phy1-ap0) entered disabled state
[ 1440.390733] rt2800usb 9-1:1.0 phy1-ap0 (unregistering): left allmulticast mode
[ 1440.391545] rt2800usb 9-1:1.0 phy1-ap0 (unregistering): left promiscuous mode
[ 1440.392308] br-lan: port 2(phy1-ap0) entered disabled state
[ 1440.748385] br-lan: port 2(phy1-ap0) entered blocking state
[ 1440.749086] br-lan: port 2(phy1-ap0) entered disabled state
```

gsxryan, 2025-05-19:
I will give these versions a try next, thank you for the guidance.

gsxryan, 2025-05-19:
I have exhausted initial testing on the other versions.  I will add my notes here for further investigation if needed.  Surprisingly, the first test above is the closest to being compatible with both of the intel adapters i have connected AX210, and N.  Are there any recommended USB or PCI A/E key Wifi adapters you support in your environment that you would recommend?

**63-66** \
DRAFT Newest Kernel Latest Version \
Both Wireless drivers are Generic Unknown. \
AX211 and 410 are installed, Kmod wlwifi is missing from opkg list.

**61-66** \
Used SYSUPGRADE to downgrade to 61-6.6
Wifi now scans, but will not host a hotspot for AP mode.
Reinstalled Software mods.
ax411 firmware is not available.

```
[  108.192579] br-lan: port 2(phy0-ap0) entered blocking state
[  108.193200] br-lan: port 2(phy0-ap0) entered disabled state
[  108.193797] iwlwifi 0000:06:10.0 phy0-ap0: entered allmulticast mode
[  108.194498] iwlwifi 0000:06:10.0 phy0-ap0: entered promiscuous mode
[  108.196213] iwlwifi 0000:06:10.0 phy0-ap0: left allmulticast mode
[  108.196892] iwlwifi 0000:06:10.0 phy0-ap0: left promiscuous mode
[  108.197625] br-lan: port 2(phy0-ap0) entered disabled state
[  108.226432] iwlwifi 0000:06:10.0: WRT: Invalid buffer destination
[  108.382503] iwlwifi 0000:06:10.0: WFPM_UMAC_PD_NOTIFICATION: 0x20
[  108.383126] iwlwifi 0000:06:10.0: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[  108.383718] iwlwifi 0000:06:10.0: WFPM_AUTH_KEY_0: 0x90
[  108.384221] iwlwifi 0000:06:10.0: CNVI_SCU_SEQ_DATA_DW9: 0x0
[  108.475658] br-lan: port 2(phy0-ap0) entered blocking state
[  108.476230] br-lan: port 2(phy0-ap0) entered disabled state
```

**63-612** \
Latest Version Older Kernel \
Both wifi adapters are Generic Unknown \
opkg is not set up for available packages. \
Attempted to manually add paths to download wifi driver dependencies \
apk --allow-untrusted add https://packages.openmptcprouter.com/v0.63-snapshot-6.12/x86_64/base/iwlwifi-firmware-ax210-20241110-r1.apk \
https://packages.openmptcprouter.com/v0.63-snapshot-6.12/x86_64/base/iwlwifi-firmware-ax411-20241110-r1.apk \
can't find kmod wlwifi package for this kernel version, attempted to use IPK from openWRT, but i don't understand how to proceed here.  \
 https://downloads.openwrt.org/releases/packages-23.05/x86_64/base/iwlwifi-firmware-iwl7265d_20230804-1_x86_64.ipk

gsxryan, 2025-05-20:
May have to point this one to upstream support.  I read reports of AP mode not working with many AX adapters in openWRT.  The reports were years old, but it looks like some may still be in this state.  Running in 'STP' mode or Wireless client mode with the AX210 firmware works, on 62-6.6, just not in AP mode.  I actually can workaround this for the time being, and as this is not in your support realm I am closing the request.

closed 2025-05-20 by gsxryan

gsxryan, 2025-05-21:
Will add some more products here I purchased for testing for Potential support for AP, and Client mode.
Afterward i found an article highlighting mainstream linux support where it may be a good place to start to look for compatible openWRT wifi modules:
https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Chipsets.md
https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md
docs.alfa.com.tw/Support/Linux/RT3572
openwrt.org/toh/views/toh_available_16128_ax-wifi

** Alfa AWUS051nh v2 Ralink RT3572
No additional Drivers Installed
AP Mode Successful
Client Mode Successful

** TP-Link USB2.0 AC600 \
Installed All Realtek RTL8821* drivers (firmware + kmod) \
AP Mode FAILED \
Client Mode Successful
I might keep this for Client Mode operations only until the Kernel 6.12 mainstream is released

** Realtek RTL8812BU AC1300 1200m USB 3.0 5dBi Antenna \
Installed Drivers for RTL8812A* family \
Drivers are not related enough for any functionality
BU driver may be custom built from 3rd party sources working on this
AP Mode FAILED
Client Mode FAILED
Only mainstream support for Kernel 6.14+

** EDUP AC600m AC1607gs Realtek RTL8811cu
8811cu is mentioned in Kernel 6.12 support
However, due to the current state of OMR opkg in this snapshot I am unable to test efficiently
I did not unbox this product, but I may keep this on hand for testing in the future.

Ysurac, 2025-05-21:
What you mean with current opkg state ? Should work now.

gsxryan, 2025-05-21:
Thank you, i'll test again if i decide to unbox that one.  Otherwise i'll update my comment.  did not notice the updates on 3947