Ysurac/openmptcprouter #3947: v0.63-snapshot-6.6 urls missing - return 404

datapharmer, 2025-05-20:
## Expected Behavior

Entries in /etc/opkg/distfeeds.conf should point to a valid location

## Current Behavior

currently the urls in /etc/opkg/distfeeds.conf point to locations that are invalid. 

## Possible Solution

unknown
## Steps to Reproduce the Problem

  1. run opkg update
  2. get error instead of it being successful

## Context (Environment)

trying to install fan control packages for rpi4 gpio controlled fan but they aren't available.

## Specifications

  - OpenMPTCProuter version: 
  openmptcprouter v0.63-snapshot-6.6 r0+286
  - OpenMPTCProuter VPS version: 
  NA
  - OpenMPTCProuter VPS provider:
  NA
  - OpenMPTCProuter platform: 
NA
  - Country: 
USA

datapharmer, 2025-05-20:
> What are the errors ?

It gives error 8 for each url on update and ends with an error 5 If you
visit the url in a browser you can see the directories are missing for .63

Ysurac, 2025-05-20:
Should be solved (or will be after cache update).

datapharmer, 2025-05-20:
Great, I confirmed this is working in 0.63 now. Note that I had to opkg install wget-ssl before updating as the package isn't in the base. You may want to consider including it by default in the future. This issue can be closed, just wanted to share the note about wget-ssl.