Ysurac/openmptcprouter #3953: Very slow speeds and instability when aggregating interfaces

danausx, 2025-05-24:
## Expected Behavior
Aggregated speeds and stability are reasonable.

## Current Behavior
On a fresh start the speeds and stability seem OK, yet after some time usually a few minutes, the speeds seem to degrade greatly, specially the upload speeds, websites and speed tests cannot load correctly. Fore example accessing RDP services remotely that are within the network through Zerotier slow down to a crawl and become unstable, so do any similar services even when accessed directly, they frequently disconnect, this happen with or without users on the network. I have attempted using different proxies and both GloryTUN UDP and TCP for UDP traffic, OpenVPN. Tried a single interface over MPTCP over VPN with Wireguard, used different SQM, tried with a single WAN. Anything connected to it just doesn't work as intended. The individual fiber lines work correctly and have same speeds and latencies. I also attempted using different VPS providers but the issue persists.

System Log:
https://pastebin.com/QA6iwjtw

[image]

## Specifications

  - OpenMPTCProuter version: 0.62-6.6
  - OpenMPTCProuter VPS version: 0.1032 6.12.15-x64v1-xanmod1
  - OpenMPTCProuter VPS provider: SolaDrive, Debian 12
  - OpenMPTCProuter platform: x86_64 running on Proxmox host, ext4 image

danausx, 2025-05-26:
> Never use MPTCP over VPN is aggregation is working. It's only a workaround when a network block MPTCP. Can you try 0.63 snapshot: https://snapshots.openmptcprouter.com/6.6/ and use Shadowsocks Rust proxy. You have a quite high latency with the VPS.

Thanks, it seems to be better now, at least the speed tests seem to be going through most of the time. I still haven't tested for longer period of time or under load.

What effect does the higher latency have in this case, what's the maximum recommended for this? Depending on the route my response time can be 70-90ms 

I still do have issues with Zerotier which I believe uses UDP on a random port. Would the OMR bypass help with this at all, if so how can I set it up or should I even do that?

I am able to use Zerotier with higher latency for RDP connections which in turn run over TCP within ZT even while tunneled through regular Wireguard but I am not able to here for some reason. The available bandwidth is very low 0.30Mbps tops, doesn't get any higher than this as if something is limiting it.

Ysurac, 2025-05-26:
Higher latency make aggregation slower in many case. For UDP this use the VPN, it's often slower. You can use OMR-ByPass by setting Zerotier server in the ByPass.

danausx, 2025-05-26:
Thanks.

Unfortunately still getting socket errors on speed tests that use the network after some time it has been on, sites don't load anymore as well, it seemed to be stable for a bit more time but eventually it reverted to this again.

I remember this used to work well with the same VPS and connections around a year or so ago.

Would it be possible to bypass the entire UDP flow into the main WAN without the VPN? I tried setting some rules for UDP traffic into the Ports Source section with all ports on OMR but that didn't seem to work for me, the traffic seems to still go through the VPS, tried the same with ICMP to test but does't work for me. The test machine is running the DNS from the OpenMPTCP.

closed 2025-09-19 by github-actions[bot]