Ysurac/openmptcprouter #3956: Not working on Azure

DominicBoettger, 2025-05-25:
I tried to install openmptcprouter on Azure but after a restart I am not able to ssh into the machine. The machine seems to be up and running and the debug show also that sshd is started.

I also tried to remove the line where the port is replaced to 65222 but this also did not change anything. 

My network security group is adjusted to allow port 22 ( which works before the installation ) and ports between 65000 and 65535 are allowed.

Maybe it's related to some Azure specific settings in the Debian image. Did somebody ever get the VPS up and running on Azure? I am interested to run the machine in Azure because of the trust of the IP addresses and because everything else is hosted there... 

I only tried with KERNEL="6.12" and not with the older 5.4.

Ysurac, 2025-05-26:
If you change SSH port you also need to allow it in ```/etc/shorewall/rules``` on the VPS.

vempire-ghost, 2025-05-26:
I have 2 VPS on Azure, and they work perfectly.

DominicBoettger, 2025-05-26:
I also tried the defaults, but it did not work. Maybe I did something wrong in my configuation.
This is my bicep config.

After I execute the original the script I am unable to connect via ssh to port 65222 but I am able to ping the machine.

``
// naming convetion https://docs.microsoft.com/en-us/azure/cloud-adoption-framework/ready/azure-best-practices/resource-abbreviations
/**
 * Bicep configuration file for OpenMPTCProuter
 * @author: Dominic Böttger
 * @date: 2025-05-23
 */
targetScope = 'subscription'

@description('The name of the current environment.')
param environment string = 'production'

@description('The location for OpenMPTCProuter resources.')
param openmptcprouterlocation string = 'germanywestcentral'

@secure()
@description('The password for the OpenMPTCProuter virtual machine')
param openmptcprouterPass string

// env variable which says prod or dev depending on the environment
var env = (environment == 'production') ? 'prod' : 'dev'

//#region OpenMPTCProuter

// create a resource group for the OpenMPTCProuter
module openMPTCProuterResourceGroup '../../modules/bicep/ResourceGroup.bicep' =  {
  name: 'rg-openmptcprouter-${environment}-001'
  params: {
    name: 'rg-openmptcprouter-${environment}-001'
    location: openmptcprouterlocation
    deleteable: true
  }
}

/**
* Create a virtual network for the OpenMPTCProuter
*/
module openMPTCProuterVNet 'br/public:avm/res/network/virtual-network:0.1.8' = {
  name: 'vnet-openmptcprouter-${environment}-001'
  scope: resourceGroup(openMPTCProuterResourceGroup.name)
  params: {
    name: 'vnet-openmptcprouter-${environment}-001'
    location: openmptcprouterlocation
    addressPrefixes: [
      '172.16.0.0/16'
    ]
    subnets: [
      {
        name: 'snet-openmptcprouter-${environment}-001'
        addressPrefix: '172.16.1.0/24'
      }
    ]
  }
}

/**
* Create Network Security Group for OpenMPTCProuter with SSH access
*/
module openMPTCProuterNSG 'br/public:avm/res/network/network-security-group:0.5.0' = {
  name: 'nsg-openmptcprouter-${environment}-001'
  scope: resourceGroup(openMPTCProuterResourceGroup.name)
  params: {
    name: 'nsg-openmptcprouter-${environment}-001'
    location: openmptcprouterlocation
    securityRules: [
      {
        name: 'AllowSSH'
        properties: {
          protocol: 'Tcp'
          sourcePortRange: '*'
          destinationPortRange: '22'
          sourceAddressPrefix: '*'
          destinationAddressPrefix: '*'
          access: 'Allow'
          priority: 1000
          direction: 'Inbound'
        }
      }
      {
        name: 'AllowSSHCustom'
        properties: {
          protocol: 'Tcp'
          sourcePortRange: '*'
          destinationPortRange: '65222'
          sourceAddressPrefix: '*'
          destinationAddressPrefix: '*'
          access: 'Allow'
          priority: 1001
          direction: 'Inbound'
        }
      }
      {
        name: 'AllowOpenMPTCProuterTCP'
        properties: {
          protocol: 'Tcp'
          sourcePortRange: '*'
          destinationPortRange: '65000-65535'
          sourceAddressPrefix: '*'
          destinationAddressPrefix: '*'
          access: 'Allow'
          priority: 1010
          direction: 'Inbound'
        }
      }
      {
        name: 'AllowOpenMPTCProuterUDP'
        properties: {
          protocol: 'Udp'
          sourcePortRange: '*'
          destinationPortRange: '65000-65535'
          sourceAddressPrefix: '*'
          destinationAddressPrefix: '*'
          access: 'Allow'
          priority: 1020
          direction: 'Inbound'
        }
      }
      {
        name: 'AllowXrayHTTPS'
        properties: {
          protocol: 'Tcp'
          sourcePortRange: '*'
          destinationPortRange: '443'
          sourceAddressPrefix: '*'
          destinationAddressPrefix: '*'
          access: 'Allow'
          priority: 1030
          direction: 'Inbound'
        }
      }
      {
        name: 'AllowICMP'
        properties: {
          protocol: 'Icmp'
          sourcePortRange: '*'
          destinationPortRange: '*'
          sourceAddressPrefix: '*'
          destinationAddressPrefix: '*'
          access: 'Allow'
          priority: 1040
          direction: 'Inbound'
        }
      }
    ]
  }
}

/**
* Create the OpenMPTCProuter VM with Debian 12 and public IP only
*/
module openMPTCProuter 'br/public:avm/res/compute/virtual-machine:0.11.0' = {
  name: 'vmopenmptcprouter${env}001'
  scope: resourceGroup(openMPTCProuterResourceGroup.name)
  params: {
    name: 'vmopenmptcprouter${env}001'
    location: openmptcprouterlocation
    vmSize: 'Standard_B1ls'
    osType: 'Linux'
    zone: 0
    adminUsername: 'azure'
    adminPassword: openmptcprouterPass
    encryptionAtHost: true
    secureBootEnabled: false
    vTpmEnabled: true
    securityType: 'TrustedLaunch'
    managedIdentities: {
      systemAssigned: true
    }
    osDisk: {
      diskSizeGB: 30
      createOption: 'FromImage'
      managedDisk: {
        storageAccountType: 'Premium_LRS'
      }
    }
    imageReference: {
      publisher: 'Debian'
      offer: 'debian-12'
      sku: '12-gen2'
      version: 'latest'
    }
    nicConfigurations: [
      {
        enableAcceleratedNetworking: false
        networkSecurityGroupResourceId: openMPTCProuterNSG.outputs.resourceId
        ipConfigurations: [
          {
            name: 'ipconfig1'
            pipConfiguration: {
              publicIpNameSuffix: '-pip-01'
              publicIPAllocationMethod: 'Static'
              skuName: 'Standard'
              zones: [
                1
                2
                3
              ]
            }
            subnetResourceId: openMPTCProuterVNet.outputs.subnetResourceIds[0]
            privateIPAllocationMethod: 'Dynamic'
            primary: true
          }
        ]
      }
    ]
  }
}

//#endregion

// Outputs
output resourceGroupName string = openMPTCProuterResourceGroup.name
output vmName string = openMPTCProuter.name
``

closed 2025-08-29 by github-actions[bot]