Ysurac/openmptcprouter #3962: QMI Cellular Interface Disabled After Ping Failure Even with Ping Checks Turned Off

w453y, 2025-05-29:
## Expected Behavior

When configuring cellular WAN interfaces using the QMI protocol (e.g., `/dev/cdc-wdm0` and `/dev/cdc-wdm2`) with proper APN and network settings, the interface should stay active and maintain stable connectivity. The route should not be deleted or interface disabled solely due to ping check failures when the modem and SIM are operational.

## Current Behavior

The interface comes up successfully and connects through the proxy. However, after a short duration (usually 30 seconds), the interface goes down and becomes unusable. The status page begins loading indefinitely, and only upload traffic appears on the bandwidth graph — no download traffic is observed.

Upon examining system logs, I consistently find:

```
Tue May 28 10:20:15 2025 user.notice post-tracking-002-error: wan1 (wwan0) switched off because check error and ping from 100.98.x.x error (9.9.9.9,1.0.0.1,114.114.115.115)
Tue May 28 10:20:15 2025 user.notice post-tracking-002-error: Delete default route to 34.131.x.x dev wwan0
```

Actions attempted:

* Disabled ping checks in **System > OpenMPTCProuter > Advanced Settings**, but interface still gets disabled.
* Commented out route deletion code in `/usr/share/omr/lib/common-post-tracking.sh`. This prevents route deletion but **does not prevent the interface from being disabled**.
* Removed `modemmanager` entirely to avoid segmentation faults and suspected conflicts when using QMI.
* Disabled `omr-tracker` for the default interface.
* Reflashed firmware and reconfigured interfaces cleanly — same issue returns upon load or speed testing.

When trying `modemmanager` as an alternative:

* It initially connects, but restarts every 30 seconds.
* The logs show it repeatedly requesting location info.
* GUI (Status > Cellular Network) **does show the correct location**, so the restart behavior appears unnecessary and disruptive.

## Specifications

  - OpenMPTCProuter version: `v0.62-6.6 r0+28431-92e020b50f`
  - OpenMPTCProuter VPS version: `OpenMPTCProuter VPS 0.1031 with 6.6.36-x64v3-xanmod1`
  - OpenMPTCProuter VPS provider: `GCP`
  - OpenMPTCProuter platform: `x86_64` (Silicom 90500-0151-E14 with Intel(R) Atom C3558 @ 2.20GHz)

## Additional Details

* Router has **two LTE modules** supporting four SIM slots.
* SIMs inserted in **slot 1 and 3**, corresponding to `/dev/cdc-wdm0` and `/dev/cdc-wdm2`.
* Interfaces configured using **QMI protocol**, APN correctly set, IPv4-only, and assigned to **WAN** firewall zone.
* `wan0` interface set as **MPTCP master** via GUI.
* Connection works briefly, but always drops due to ping check logic.

---

### Request for Help

I’ve spent over **two weeks debugging this setup** and am currently stuck. My goal is to use OpenMPTCProuter reliably on an LTE-based x86 router. I would really appreciate any help in identifying:

* Where the **interface disabling logic** lives (after ping failure, even when route deletion is disabled),
* How to **cleanly disable all ping-based tracking** (both UI and backend),
* Or any suggestions to **achieve a stable LTE setup** without unintended interface shutdown.

Thank you for all the great work on OMR — I’d love to help test patches or improvements if needed.

Ysurac, 2025-05-29:
Please use ModemManager (far more tested), and show le system log.
I would also need the modem model.

lhc1993, 2025-06-01:
I have the same problem. How can I permanently turn off these tracker tests?

lhc1993, 2025-06-01:
[image]
This tracker is also very resource intensive.

Ysurac, 2025-06-01:
Same answer, and not really CPU intensive, only using available CPU.

lhc1993, 2025-06-01:
How to cleanly disable all ping-based tracking (both UI and backend),
Or any suggestions to achieve a stable LTE setup without unintended interface shutdown.

Could you please give me the code location here? It would be very grateful!

lhc1993, 2025-06-01:
ModemManager does not automatically recognize pcie devices, I don't know where to add them manually, and I can't do more testing with it. Any suggestions for this problem with pcie devices?

Ysurac, 2025-06-01:
It's not related to this issue about QMI (but you can change timeout in Services->OMR-Tracker Manager)