Ysurac/openmptcprouter #3963: Меняются интерфейсы после перезагрузки

Maximguselnikov, 2025-05-29:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2025-06-01:
Please use the template.
I don't understand what you mean with the OpenMPTCProuter checkbox

Maximguselnikov, 2025-06-02:
Hello. I have a problem with openmptcprouter v0.62-6.6 on Raspberry Pi 3 Model B Rev 1.2. After connecting and configuring modems and a server on which I have Debian 12 installed and the openmptcprouter service, my interfaces change or are renamed when I reboot the system. I tried to enter the mac address in the interfaces and then the device, it does not help, then I unchecked the "automatic interface renaming" in the settings, it also does not help. What else can be done?

[image]
[image]
[image]

Ysurac, 2025-06-04:
@helixcn2024 maybe because issue was not open on the right repo, with information not the same on each answer and you said "interface not recognized" and then a screenshot with 4 interfaces... Not easy to understand what you mean...

@Maximguselnikov Did you try System->OpenMPTCProuter, "Advanced settings" tab, uncheck "Disable interfaces auto rename" ? I don't see any try to rename interface in the log.

Ysurac, 2025-06-06:
I would need the result of ```uci show network``` via SSH on OMR please.

Maximguselnikov, 2025-06-17:
Hello. I am sending you the logs that you requested. It turns out that the first log is when all modems are connected, and the second log is when the interface has changed on one of the modems.

Maximguselnikov, 2025-06-17:
root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.device='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.metric='7'
network.loopback.ip4table='7'
network.loopback.ip6table='67'
network.globals=globals
network.globals.ula_prefix='fd4a:ea7c:6f8a::/48'
network.globals.packet_steering='0'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='4'
network.globals.mptcp_subflows='8'
network.globals.mptcp_add_addr_accepted='8'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_pm_type='0'
network.globals.mptcp_disable_initial_config='0'
network.globals.mptcp_force_multipath='1'
network.globals.mptcpd_enable='0'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.device='eth0'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.ifname='eth0'
network.lan.metric='9999'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.addlatency='0'
network.lan.txqueuelen='2000'
network.lan.multipath='off'
network.lan.ip4table='9999'
network.lan.ip6table='69999'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan.modalias='usb:v0424pEC00d0200dcFFdsc00dp01icFFisc00ipFFin00-b8:27:eb                                                                                                             :4c:9f:7a'
network.lan.product='424/ec00/200'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='dhcp'
network.wan1.device='eth1'
network.wan1.multipath='on'
network.wan1.addlatency='0'
network.wan1.metric='5'
network.wan1.ip4table='5'
network.wan1.ip6table='65'
network.wan1.defaultroute='0'
network.wan1.peerdns='0'
network.wan1.ipv6='0'
network.wan1.modalias='usb:v12D1p14DCd0102dc02dsc00dp00ic02isc06ip00in00-02:df:8                                                                                                             b:0a:63:0a'
network.wan1.product='12d1/14dc/102'
network.omrvpn=interface
network.omrvpn.device='tun0'
network.omrvpn.proto='dhcp'
network.omrvpn.ip4table='1500'
network.omrvpn.multipath='off'
[... 113 lines cut ...]
network.eth0_fw_rule.lookup='9999'
network.eth1_fw_rule=rule
network.eth1_fw_rule.priority='1'
network.eth1_fw_rule.mark='0x45395'
network.eth1_fw_rule.lookup='5'
network.tun0_fw_rule=rule
network.tun0_fw_rule.priority='1'
network.tun0_fw_rule.mark='0x45391500'
network.tun0_fw_rule.lookup='1500'
network.eth6_fw_rule=rule
network.eth6_fw_rule.priority='1'
network.eth6_fw_rule.mark='0x453913'
network.eth6_fw_rule.lookup='13'
network.eth5_fw_rule=rule
network.eth5_fw_rule.priority='1'
network.eth5_fw_rule.mark='0x453918'
network.eth5_fw_rule.lookup='18'
network.ppp0_fw_rule=rule
network.ppp0_fw_rule.priority='1'
network.ppp0_fw_rule.mark='0x453920'
network.ppp0_fw_rule.lookup='20'
network.all_fw_rule=rule
network.all_fw_rule.priority='1'
network.all_fw_rule.mark='0x4539'
network.all_fw_rule.lookup='991337'

closed 2025-11-23 by github-actions[bot]