Ysurac/openmptcprouter #3964: No network bonding no redundance

alchool, 2025-05-29:
## Expected Behavior
2 or more network linked. I expected almost the sum of their bandwith instead of the result of the main.
Also if I shutdown the main connection I should have not any disconnection from VPN.

## Current Behavior

If I shut down the one that I see in establish connection I lose the tunnel, it changes status in FIN_WAIT and then it raise the other connection to ESTB and forget the previous one. No matter if I change master interface, or multipath scheduler or path-manager.
I also tried different VPN or Proxy with or without crypto.
I built omr image also in different hardware but with the same result

## Specifications
(I also tried v0.58, with different vm version, and I can handle the interface disconnection changing from "balancing" to "dynamic")

  - OpenMPTCProuter version: v0.61 or v0.62 
  - OpenMPTCProuter VPS version: 6.1.0-35-amd64 0.1032 
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: x86-64 (64-bits)

Ysurac, 2025-05-29:
VPS kernel is not the good one, should be a 6.12

Ysurac, 2025-06-03:
OMR 6.12 is still in test.
Please stay on Shadowsocks-Rust for test. What do you have in System->OpenMPTCProuter, "status" tab ?