Ysurac/openmptcprouter #3965: v2ray still have many problem at the latest omr

fareign, 2025-06-01:
settings:
proxy:v2ray vless
VPN:mlvpn
most time  first boot  v2ray not work, have low speed, the traffic transport with mlvpn
need /etc/init.d/v2ray reload. then everything is OK
sometime need restart VPS V2ray to use portfoward. 

## Specifications

  - OpenMPTCProuter version: 0.63rc1
  - OpenMPTCProuter VPS version: 0.1032
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: rpi4
  - Country:

Ysurac, 2025-06-01:
Did you try XRay ? do you have same issue with XRay ?

fareign, 2025-06-01:
yes,the same

fareign, 2025-06-02:
> I just updated server-test script to use latest *Ray versions. Possible client was not working well on previous server version.

I updated VPS version by myself, still have the problem. 
by route side,I can reload it in rc.local.
but,if reboot, port forward also not work. need restart.

fareign, 2025-06-03:
02:00:00 2025 user.notice omr-schedule-010-services: Set firewall on server vps

after this log .port forward works

Ysurac, 2025-06-06:
I forced firewall on server in latest snapshot, not sure if this is better ?

fareign, 2025-06-09:
> I forced firewall on server in latest snapshot, not sure if this is better ?

yes,much better.