Ysurac/openmptcprouter #3971: [BUG] MPTCP does not recover subflows on dynamic IP change (OMR 0.63 snapshot)

darkman1983, 2025-06-08:
### Expected Behavior
When the WAN interface gets a new public IP (e.g., via DHCP renewal or PPPoE reconnect), MPTCP should remove the old subflows and automatically add the new IP address, allowing new MPTCP connections and subflows to be created.

### Current Behavior
After a public IP change on the WAN interface, the system routing and firewall rules are reloaded correctly, but MPTCP does not create new subflows based on the new IP.
The old subflows remain in place, and no new ones are established.
New outgoing MPTCP connections fail to establish full subflow paths unless the path manager is manually restarted via:

/etc/init.d/mptcp pathmanager restart

System log shows no errors related to MPTCP, and no ADD_ADDR events are triggered automatically.

### Possible Solution

Add hotplug integration or netifd trigger to restart MPTCP path manager when WAN IP changes

Dynamically clean up old subflows and add the new IP using ip mptcp add address

Optional: integrate with mptcpd or implement equivalent behavior in userspace

### Steps to Reproduce the Problem

1. Use OpenMPTCProuter 0.63 snapshot (6.6 kernel) with only one WAN interface

2. Let WAN receive public IP via DHCP or PPPoE

3. Force an IP change (reconnect, Zwangstrennung, or DHCP release/renew)

4. Check /proc/net/mptcp_net/mptcp_subflows — stale entries remain, new ones are not created

5. Observe degraded MPTCP performance and fewer active subflows

### Context (Environment)
This causes degraded or broken MPTCP functionality after a public IP change, especially with only one WAN active. It affects overall connection reliability and performance, as MPTCP fails to recover dynamically.
A manual restart of the MPTCP path manager resolves the issue temporarily.

### Specifications

OpenMPTCProuter version: 0.63-snapshot-6.6

OpenMPTCProuter VPS version: 0.1032-test 6.12.15-x64v3-xanmod1

OpenMPTCProuter VPS provider: 1blu.de

OpenMPTCProuter platform: x86_64

Country: Germany

Ysurac, 2025-06-09:
There is no '"/etc/init.d/mptcp pathmanager restart" command and mptcp subflow should be updated when IP change.

Ysurac, 2025-06-10:
There is also no ```/proc/net/mptcp_net/mptcp_subflows``` in kernel 6.6. How are you checking active subflows ?

closed 2025-09-16 by github-actions[bot]