Ysurac/openmptcprouter #3984: slower (400Mbits) then only my Cable (800Mbits)

iEddie-cmd, 2025-06-22:
## Expected Behavior
Faster with 2 WAN

## Current Behavior
I have a Gigabit Cable + 100Mbit 5g Connection.
But with openmptcprouter i am slower Download Speeds (400Mbits) then only my Cable (800Mbits).

-> Upload speeds got increased just fine. From 53Mbits + 80Mbits -> 120 Mbits.
Any ideas why and how to solve it?

## Specifications

  - OpenMPTCProuter version: <openmptcprouter v0.62-6.6 r0+28431-92e020b50f
  - OpenMPTCProuter VPS version: 6.12.15-x64v3-xanmod1 0.1032
  - OpenMPTCProuter VPS provider: IONOS, VPS XS 1 Core
  - OpenMPTCProuter platform: Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz
  - Country: Germany

iEddie-cmd, 2025-06-23:
Yes. Cable is master.
 Speedtest done by Ookla® with cable conneciton directly inside the asus router behind openmptcproouter.

YcarusHospital, 2025-06-29:
Can you try, via SSH on the router, with ```omr-test-speed <interface>``` (you can find interface via ```multipath``` command) on each interface ?

iEddie-cmd, 2025-06-29:
eth1
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M    0 1024M    0     0   107M      0 --:--:--  0:00:09 --:--:--  109M

eth2
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  344M    0  344M    0     0  15.2M      0 --:--:--  0:00:22 --:--:-- 16.5M^C

YcarusHospital, 2025-06-30:
And without interface specified ?

iEddie-cmd, 2025-07-04:
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M    0 1024M    0     0  66.3M      0 --:--:--  0:00:15 --:--:-- 68.5M

Ysurac, 2025-07-05:
Can you check during a ```omr-test-speed``` without interfaces in Network->MPTCP, Bandwidth tab if more than one interface is used ? It's to check if aggregation is working.

iEddie-cmd, 2025-07-24:
It seems like.
wan2 (LTE) | Current: | 33.39 Mbit/s(4.17 MB/s) | Average: | 33.83 Mbit/s(4.23 MB/s) | Peak: | 50.52 Mbit/s(6.31 MB/s)
-- | -- | -- | -- | -- | -- | --
wan1 (Cable) | Current: | 430.8 Mbit/s(53.85 MB/s) | Average: | 453.66 Mbit/s(56.71 MB/s) | Peak: | 591.43 Mbit/s(73.93 MB/s)
total | Current: | 442.61 Mbit/s(55.33 MB/s) | Average: | 475.68 Mbit/s(59.46 MB/s) | Peak: | 598.95 Mbit/s(74.87 MB/s)