Ysurac/openmptcprouter #3985: IPv6 Not working.

iamfreaky, 2025-06-22:
## Expected Behavior
IPv6 Internet Connection should work without a problem. VPS has a /48 extra Network, which should be  routed to a OPNsense behind OpenMPTCP. (BuyVM)

## Current Behavior
Not even a IPv6 Connection is Possible, from OpenMPTCP itself. IPv4 works. omr6in4 is labeld as Up. In the Systemlog there is nothing that could help. 

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.62-6.6 r0+28431-92e020b5
  - OpenMPTCProuter VPS version: < OpenMPTCProuter VPS 0.1032 > 6.12.15-x64v3-xanmod1 #0~20250219.g6e42b4c
  - OpenMPTCProuter VPS provider: BuyVM, Manuell Install of Debian 12
  - OpenMPTCProuter platform: x86_64_efi

Ysurac, 2025-06-22:
IPv6 is working from the VPS ? Was already configured before installation of VPS script ?
What do you have in ```ip a``` from the VPS ? ```ip a``` and ```ip -6 r``` from the router ?

YcarusHospital, 2025-06-29:
Here you should have IPv6 Unique Local Range working, seems not ;) did you try to set your range in Network->Interface,  tab that I dont remember name

iamfreaky, 2025-06-29:
Funfact ? If i tgry to ping the Local IPv6 Gateways on the router, i get a Network Unreachable. The First IPv6 is the Gateway Itself Does work. 

[image]

@Ysurac @YcarusHospital Also get well soon.

Ysurac, 2025-07-27:
It's in my todo list. Seems that a route is not correct/stable.