Ysurac/openmptcprouter #3987: DNS Forwardings

useridivan, 2025-06-23:
## Expected Behavior

DNS forwarding settings should be saved and persist across system reboots and firewall configuration changes.

## Current Behavior

DNS forwarding settings reset to 127.0.0.1#53 after rebooting the system or modifying firewall settings (e.g., enabling/disabling port forwarding rules).

## Possible Solution

## Steps to Reproduce the Problem

  1. Insert an IP address on DNS Forwardings in Network / DHCP and DNS / General tab and apply.
  2.  Reboot the system or change a firewall settings
  3. DNS forwardings resets itself at 127.0.0.1#53

## Context (Environment)

The system was upgraded incrementally from v0.59 to v0.63rc1-snapshot-6.6, moving from one version to the next as each was released. I noticed differences in the OpenWrt interface between older and newer versions, which might be related to the issue.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.63-snapshot-6.6 r0+28627-0b392b925f (but also the stable 0.62 was affected
  - OpenMPTCProuter VPS version:  0.1032-test 6.12.15-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: Aruba Cloud 
  - OpenMPTCProuter platform: x86_64 on a virtual machine (Proxmox 8.4.1)
  - Country: Italy

useridivan, 2025-06-29:
Thank you for reviewing the issue. My default VPN is "A Dead Simple VPN". I configure the DNS forwardings IP to point to a local Lancache server, which in turn forwards DNS queries to a PiHole installation. Both Lancache and PiHole are running in separate local containers on the same Proxmox.

flimo44, 2025-07-17:
hello,
i have the same problem. But I have Pihole on the VPS. But now need forward since a pihole update to 10.255.255.1#53
After each reboot 10.255.255.1#53 lost.

Ysurac, 2025-07-17:
@flimo44 did you use the PiHole install script from OpenMPTCProuter ?

flimo44, 2025-07-17:
> [@flimo44](https://github.com/flimo44) did you use the PiHole install script from OpenMPTCProuter ?

yes