Ysurac/openmptcprouter #3989: Please help me exceed the maximum WAN Links of 8, I have more than 10 WAN links

skullface101, 2025-06-24:
## Specifications

  - OpenMPTCProuter version: 0.1032
  - OpenMPTCProuter VPS version: v0.62-6.6-r0+28431
  - OpenMPTCProuter VPS provider: Conatbo
  - OpenMPTCProuter platform: x86_64 VM

Please help me exceed the maximum WAN Links of 8, I have more than 10 WAN links

skullface101, 2025-07-05:
Please help me build a custom build that supports up to 16 WAN links, I can follow your instructions

skullface101, 2025-07-06:
whats the file name to be edited?

skullface101, 2025-07-09:
@Ysurac reminder

skullface101, 2025-07-10:
I need step by step instructions to make set the max to 16 instead of 8.
@Ysurac

skullface101, 2025-07-13:
@Ysurac this is very urgent and important, your distribution is the best, I need step by step instructions to make set the max to 16 instead of 8.

Ysurac, 2025-07-13:
There is no "step by step" instructions, a patch need to be done. As I can't test that for now and there is many changes in kernel >6.12 on this part, no sure it's worth it.

skullface101, 2025-07-13:
I can pay you for this patch

Ysurac, 2025-07-16:
The mattbe patch is now in last snapshot.

skullface101, 2025-07-16:
Thanks
So I need to build develop branch right?

skullface101, 2025-07-16:
Thanks, I will test it and let you know the results, I can help you with the tests you do, I have X86-64 on proxmox and I have WRT3200ACM

skullface101, 2025-07-16:
> You can use the snapshot (https://snapshots.openmptcprouter.com/6.12/) and change the value in Network->MPTCP, "Max subflows"

I changed it to 16 and 64 but it;s still limited to 8

Ysurac, 2025-07-16:
What is the result of ```ip mptcp limits``` ?

skullface101, 2025-07-16:
[image]

Ysurac, 2025-07-16:
Strange result...
Try a ```ip mptcp limits set add_addr_accepted 8 subflows 16```

skullface101, 2025-07-16:
```
root@OpenMPTCProuter:~# ip mptcp limits set add_addr_accepted 8 subflows 16
root@OpenMPTCProuter:~# ip mptcp limits
add_addr_accepted 8 subflows 16

```
Does this output mean that it works.

skullface101, 2025-07-16:
so how can I remove this other limit?

skullface101, 2025-07-17:
@Ysurac 
root@omr-vps:~# ip mptcp limits set add_addr_accepted 8 subflows 16
Error: limit greater than maximum.

VPS script needs to be patched too

skullface101, 2025-07-22:
Please help me patch the VPS
@Ysurac

skullface101, 2025-07-25:
Reminder @Ysurac

skullface101, 2025-08-24:
@Ysurac  I edited this file https://github.com/multipath-tcp/mptcp_net-next/blob/export/net/mptcp/pm_kernel.c
Now, how can I build this?

Note: this is for the VPS

skullface101, 2025-09-04:
> > Note: this is for the VPS
> 
> [@skullface101](https://github.com/skullface101), exactly which Linux distribution (including version) do you run on the VPS service?

Debian

skullface101, 2025-09-18:
Will rebuilding the VPS kernel make it work?

sskras, 2025-09-18:
If it's really the KVM-based VPS, it should.

Have you tried googling about that? Eg: 

* https://cyberpanel.net/blog/kvm-vps#:~:text=KVM%20supports%20custom%20kernels
  
  > Unlike container based virtualization, KVM supports custom kernels. This is especially useful for developers or users that need specific kernel modules. 

* https://www.hostinger.com/vps-hosting#:~:text=What%20is%20KVM%3F%20What%20are%20the%20benefits%20of%20KVM%3F
  
  > **What is KVM? What are the benefits of KVM?**
  >
  > [...]
  >
  > Unlike other virtualization technologies, KVM gives you a higher level of isolation, reliability, and customizability – ideal if your development projects demand plenty of power and flexibility.
  > 
  > With KVM VPS hosting services, you’ll be in complete control – freely customize the OS, control panel, and software to suit your needs. While you’ll be responsible for your virtual private servers, we’ll take care of the speed, uptime, and stability of our physical servers.

sskras, 2025-09-18:
> > exactly which Linux distribution (including version) do you run on the VPS service?
> 
> Debian 12

Eg: https://www.siberoloji.com/how-to-compile-the-linux-kernel-from-source-on-debian-12-bookworm/

Ysurac, 2025-09-18:
You need to run again VPS script: it will update kernel and everything (even XRay).

skullface101, 2025-09-18:
> You need to run again VPS script: it will update kernel and everything (even XRay).

I think I should run the vps script then rebuild the kernel

skullface101, 2025-09-19:
> Why do you want to rebuild a kernel and not use provided kernel ?

I need to patch the VPS kernel to allow up to 64 subflows, can you provide a vps script that included the patch?

skullface101, 2025-12-27:
Hi I added an additional IP thats only purpose is backup IP if the main ip is filtered, I dont care about exit IP, is there any needed confog for that?

skullface101, 2026-11-02:
Please add back the patch to snapshots

its gonna take forever for OpenWRT to use Kernel 7.2/7.3