Ysurac/openmptcprouter #3993: QMI and MBIM modem configuration failed

dmkuzmin, 2025-06-27:
I use the image openmptcprouter-v0.63-snapshot-6.12-r0+30095-6af9476b8a-bcm27xx-bcm2711-rpi-4-squashfs-sysupgrade.img.gz
In an LTE modem, I can choose to work using the MBIM or QMI protocol.

## Expected Behavior
When using the QMI protocol, the output of the "dmesg | grep qmi" command is as follows:
[   16.125921] usbcore: registered new interface driver qmi_wwan
[  488.175321] qmi_wwan 1-1.2.1:1.4: probe with driver qmi_wwan failed with error -22

When I select the MBIM protocol, the device appears.
When creating the MBIM interface and specifying the interface itself "/dev/cdc-wdm0" from the list of suggested ones and installing the mobile operator's APN, nothing happens.
When creating the Modem Manager interface, my modem is missing from the device list.

Have you encountered such problems and is there a solution for using QMI and MBIM protocols on a modem? If additional information is needed, I will try to provide everything necessary.

Thank you in advance.

dmkuzmin, 2025-06-27:
When choosing the ECM protocol, everything works fine in the modem, but there is another problem. The DHCP server on the modem side issues IP addresses from the same subnet, and there may be a conflict when connecting multiple modems. In addition, if all modems are on the same subnet, there is no flow summation. According to the configuration instructions, it was necessary to specify static addresses by the modem in different subnets, but due to the presence of a dhcp server in the modems, addresses of the same subnet are issued. Can you give me any recommendations about this?

Ycarusout, 2025-06-27:
Did you try with ModemManager and modem set as MBIM?

YcarusHospital, 2025-06-29:
Configured as ModemManager, what is the result of ```mmcli -L``` ?

dmkuzmin, 2025-07-04:
As a result, it turned out that QMI and MBIM in one type of LTE modem work fine, but not in the other. At the same time, both work fine in Windows.

Ysurac, 2025-07-07:
So this issue can be close ?
What are the LTE modem models ?

dmkuzmin, 2025-07-07:
Do you have any idea why one LTE modem might work failed in OpenMPTCProuter, but fine in Windows? Although there are no problems with another modem in either case.

Problematic modem: miniPCIe MeiG SLM750 (LTE Cat4) with USB-miniPCIe adapter
Working modem: miniPCIe MeiG SLM828 (LTE Cat6) with USB-miniPCIe adapter

dmkuzmin, 2025-07-18:
Are the OpenWRT developers implementing the QMI and MBIM driver?

Ysurac, 2025-07-18:
QMI & MBIM is implemented by ModemManager when this protocol is used.
Modem can have or not a dedicated module. If not, this can be a firmware issue or something else.
I never used "MeiG" modems.

dmkuzmin, 2025-07-18:
I wrote this in the first place, but there is no information there yet.

closed 2025-10-21 by github-actions[bot]