Ysurac/openmptcprouter #3996: Slow Webui Page Load on ZBT Z8102AX v0.62-6.6

chubx, 2025-06-29:
## Expected Behavior

Page load should be just quick when accessing the System > OpenMPTCProuter page
## Current Behavior

When you login to the admin webui, and then go to:
System > OpenMPTCProuter page

The page take longer that expected.

If you compare it with the other tabs/pages, it is significantly slow.

So it is very annoying to work on the router, especially if you are testing configurations, and that you are using the web ui.

## Possible Solution

Maybe we find what it causing the slow down
## Steps to Reproduce the Problem

  1. login to the admin webui
  2. go to System > OpenMPTCProuter page
  3. setting wizard tab is slow loading

## Context (Environment)

The issue is counter productive especially when you are testing out several settings, and on different scenarios. It loads so slow and testing becomes daunting.

I have mentioned this in an issue, and a user also confirmed it with his ZBT

System > OpenMPTCProuter page is slow loading.

If you go to other pages/tabs or menus on the OMR admin web ui, it is fine, but this page.

## Specifications

  - OpenMPTCProuter version:  `v0.62-6.6 r0+28431-92e020b50f`
  - OpenMPTCProuter VPS version:  ` OpenMPTCProuter VPS 0.1031`
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

YcarusHospital, 2025-06-29:
I'm not able to reproduce the issue. I would say it's a problem with ModemManager, but I have no problem with same config. Something slow in ```mmcli -L``` and ```mmcli -m ....```

chubx, 2025-08-27:
> I'm not able to reproduce the issue. I would say it's a problem with ModemManager, but I have no problem with same config. Something slow in `mmcli -L` and `mmcli -m ....`

@Ysurac I have now two units of ZBT Z8102AX, the latest unit with 4G/5G Module removed. Disabled the modem manager and the interface entries from the OpenMPTCP Wizard page.

Still slow when doing to the page:
System > OpenMPTCP Router 

any advise where to look on, or investigate on?

closed 2025-12-09 by github-actions[bot]