Ysurac/openmptcprouter #3998: Problems activating Multipath TCP

szycho, 2025-06-29:
Hello, I'm using this great project for SRT video transmissions and I'm encountering several issues. I'm from Argentina and I've tested it with up to 4 interfaces.

A. I can't enable Multipath TCP.
B. When I start streaming and the bandwidth exceeds 4 Mbps, the system becomes unstable.
C. Bonding doesn't seem to be working — when I disconnect one interface, I lose connectivity.

Specifications

  - OpenMPTCProuter version: openmptcprouter v0.62-6.6 r0+28431-92e020b50f
  - OpenMPTCProuter VPS version: 6.12.15-x64v3-xanmod1 0.1032
  - OpenMPTCProuter VPS provider: Amazon 
  - OpenMPTCProuter platform: RP4

[image]
[image]
[image]
[image]
[image]
[image]

YcarusHospital, 2025-06-29:
I would need Status->System log. It seems that you have an issue with VPN part

szycho, 2025-06-29:
Sat Jun 28 14:44:45 2025 daemon.info ModemManager[482]: hotplug: ModemManager not yet available (count=36)
Sat Jun 28 14:44:45 2025 daemon.info ModemManager[481]: hotplug: checking if ModemManager is available...
Sat Jun 28 14:44:45 2025 daemon.info ModemManager[481]: hotplug: ModemManager not yet available
Sat Jun 28 14:44:46 2025 daemon.info ModemManager[482]: hotplug: Checking if ModemManager is available...
Sat Jun 28 14:44:46 2025 daemon.info ModemManager[482]: hotplug: ModemManager not yet available (count=35)
Sat Jun 28 14:44:46 2025 daemon.info ModemManager[481]: hotplug: checking if ModemManager is available...
Sat Jun 28 14:44:46 2025 daemon.info ModemManager[481]: hotplug: ModemManager not yet available
Sat Jun 28 14:44:47 2025 daemon.info ModemManager[482]: hotplug: Checking if ModemManager is available...
Sat Jun 28 14:44:47 2025 daemon.info ModemManager[482]: hotplug: ModemManager not yet available (count=34)
Sat Jun 28 14:44:47 2025 daemon.info ModemManager[481]: hotplug: checking if ModemManager is available...
Sat Jun 28 14:44:47 2025 daemon.info ModemManager[481]: hotplug: ModemManager not yet available
Sat Jun 28 14:44:48 2025 user.notice omr-tracker: Launched
Sat Jun 28 14:44:48 2025 daemon.info ModemManager[482]: hotplug: Checking if ModemManager is available...
Sat Jun 28 14:44:48 2025 daemon.info ModemManager[482]: hotplug: ModemManager not yet available (count=33)
Sat Jun 28 14:44:48 2025 daemon.info ModemManager[481]: hotplug: checking if ModemManager is available...
Sat Jun 28 14:44:48 2025 daemon.info ModemManager[481]: hotplug: ModemManager not yet available
Sat Jun 28 14:44:48 2025 daemon.warn openvpn(omr)[15508]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations.
Sat Jun 28 14:44:48 2025 daemon.notice openvpn(omr)[15508]: OpenVPN 2.6.13 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Sat Jun 28 14:44:48 2025 daemon.notice openvpn(omr)[15508]: library versions: OpenSSL 3.0.15 3 Sep 2024, LZO 2.10
Sat Jun 28 14:44:48 2025 daemon.notice openvpn(omr)[15508]: DCO version: N/A
Sat Jun 28 14:44:48 2025 daemon.warn openvpn(omr)[15508]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Sat Jun 28 14:44:48 2025 daemon.warn openvpn(omr)[15508]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sat Jun 28 14:44:48 2025 daemon.notice openvpn(omr)[15508]: TCP/UDP: Preserving recently used remote address: [AF_INET]3.222.186.221:65301
Sat Jun 28 14:44:48 2025 daemon.notice openvpn(omr)[15508]: Attempting to establish TCP connection with [AF_INET]3.222.186.221:65301
Sat Jun 28 14:44:48 2025 user.notice post-tracking-002-error: omrvpn down
Sat Jun 28 14:44:49 2025 daemon.err usbmuxd[15766]: [17:44:49.078][3] usbmuxd v2023.10.11~360619c5 starting up
Sat Jun 28 14:44:49 2025 daemon.err usbmuxd[15766]: [17:44:49.079][3] Using libusb 1.0.27
Sat Jun 28 14:44:49 2025 daemon.err usbmuxd[15766]: [17:44:49.079][3] Initialization complete
Sat Jun 28 14:44:49 2025 daemon.err usbmuxd[15766]: [17:44:49.079][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Sat Jun 28 14:44:49 2025 daemon.notice openvpn(omr)[15508]: TCP connection established with [AF_INET]3.222.186.221:65301
Sat Jun 28 14:44:49 2025 daemon.warn openvpn(omr)[15508]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Not supported (errno=95)
Sat Jun 28 14:44:49 2025 daemon.notice openvpn(omr)[15508]: TCPv4_CLIENT link local: (not bound)
Sat Jun 28 14:44:49 2025 daemon.notice openvpn(omr)[15508]: TCPv4_CLIENT link remote: [AF_INET]3.222.186.221:65301
Sat Jun 28 14:44:49 2025 daemon.info ModemManager[482]: hotplug: Checking if ModemManager is available...
Sat Jun 28 14:44:49 2025 daemon.info ModemManager[482]: hotplug: ModemManager not yet available (count=32)
Sat Jun 28 14:44:49 2025 daemon.info ModemManager[481]: hotplug: checking if ModemManager is available...
Sat Jun 28 14:44:49 2025 user.notice post-tracking-002-error: OpenVPN down, restart it
Sat Jun 28 14:44:49 2025 daemon.info ModemManager[481]: hotplug: ModemManager not yet available
Sat Jun 28 14:44:49 2025 daemon.err openvpn(omr)[15508]: event_wait : Interrupted system call (fd=-1,code=4)
Sat Jun 28 14:44:49 2025 daemon.notice openvpn(omr)[15508]: SIGTERM[hard,] received, process exiting
Sat Jun 28 14:44:49 2025 daemon.warn openvpn(omr)[16219]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations.
Sat Jun 28 14:44:49 2025 daemon.notice openvpn(omr)[16219]: OpenVPN 2.6.13 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Sat Jun 28 14:44:49 2025 daemon.notice openvpn(omr)[16219]: library versions: OpenSSL 3.0.15 3 Sep 2024, LZO 2.10
Sat Jun 28 14:44:49 2025 daemon.notice openvpn(omr)[16219]: DCO version: N/A
Sat Jun 28 14:44:49 2025 daemon.warn openvpn(omr)[16219]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Sat Jun 28 14:44:49 2025 daemon.warn openvpn(omr)[16219]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
[... cut ...]
Sun Jun 29 12:32:44 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sun Jun 29 12:32:44 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Sun Jun 29 12:32:44 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sun Jun 29 12:32:45 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sun Jun 29 12:32:45 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Sun Jun 29 12:32:45 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses

derMarienkaefer, 2025-07-01:
Which encryption setting do you have on advanced page? FOr me this error message seems weird, maybe something wrong with Chacha20, i use AES-256-CGM

`Sat Jun 28 14:44:48 2025 daemon.warn openvpn(omr)[15508]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations.`

Pimmal, 2025-07-01:
Which encryption setting do you have on advanced page? FOr me this error message seems weird, maybe something wrong with Chacha20, i use AES-256-CGM

`Sat Jun 28 14:44:48 2025 daemon.warn openvpn(omr)[15508]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations.`

esquilo444, 2025-09-03:
Pour le moment ce projet n'est pas adapté aux flux SRT .
Dans ce cas le srt passe par de l'UDP, ce qui n'est pas la force majeur de ce projet ..
J'ai passé près d'un an à me pencher sur le même projet .
J'espère un jour moi aussi ..