Ysurac/openmptcprouter #4000: VPN tunnel DOWN during speed test

dmkuzmin, 2025-07-03:
## Current Behavior
When I run the speed test through the SpeedTest service, I see that the VPN tunnel is turned off during or after the download process - the VPN tunnel DOWN message appears and everything returns to normal after a couple of seconds. In this case, the connection with the SpeedTest service is lost.

[image]
[image]

## Possible Solution
Since all WAN interfaces are LTE modems, access to the Internet may be unstable from time to time. You probably need to configure something in MPTCP so that this instability does not affect the restart of the VPN tunnel. Maybe there are some recommendations or what to look for to optimize the operation of a router with multiple LTE modems?

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.63-snapshot-6.12-r0+30095-6af9476b8a-bcm27xx-bcm2711-rpi-4
  - OpenMPTCProuter VPS version: 0.1032 6.12.15-x64v3-xanmod1
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: RP4
  - Country: RU

Ysurac, 2025-07-03:
You should enable SQM for this.

dmkuzmin, 2025-07-03:
It has already been enabled on every WAN interface

[image]

Ysurac, 2025-07-03:
To have SQM work, you need also to set download/upload speed if fixed speed or autorate.

dmkuzmin, 2025-07-03:
Nothing has changed.

[image]

dmkuzmin, 2025-07-03:
**So far, everything is also unchanged. I turned off the SQM autorate and set a speed limit of 1 Mbit/s everywhere.**

root@OpenMPTCProuter:~# logread -f
Thu Jul  3 10:44:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Thu Jul  3 10:44:00 2025 user.notice omr-tracker: Launching...
Thu Jul  3 10:44:11 2025 user.notice omr-tracker: Launched
Thu Jul  3 10:44:32 2025 user.notice post-tracking-002-error: DNS flush
Thu Jul  3 10:44:33 2025 daemon.notice netifd: Network device 'usb3' link is down
Thu Jul  3 10:44:33 2025 daemon.notice netifd: Interface 'wan4' has link connectivity loss
Thu Jul  3 10:44:33 2025 daemon.notice netifd: wan4 (9375): udhcpc: received SIGTERM
Thu Jul  3 10:44:33 2025 daemon.notice netifd: wan4 (9375): udhcpc: unicasting a release of 100.83.69.214 to 169.254.0.1
Thu Jul  3 10:44:33 2025 daemon.notice netifd: wan4 (9375): udhcpc: sending release
Thu Jul  3 10:44:33 2025 daemon.notice netifd: wan4 (9375): udhcpc: entering released state
Thu Jul  3 10:44:33 2025 daemon.notice netifd: wan4 (9375): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan4" } (Permission denied)
Thu Jul  3 10:44:33 2025 daemon.notice netifd: Interface 'wan4' is now down
Thu Jul  3 10:44:33 2025 daemon.notice ttyd[4454]: [2025/07/03 10:44:33:9164] N: rops_handle_POLLIN_netlink: DELADDR
Thu Jul  3 10:45:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Thu Jul  3 10:45:00 2025 user.notice omr-tracker: Launching...
Thu Jul  3 10:45:11 2025 user.notice omr-tracker: Launched
Thu Jul  3 10:45:11 2025 user.notice post-tracking-002-error: DNS flush
Thu Jul  3 10:45:33 2025 user.notice post-tracking-002-error: omrvpn down because IPv4 gateway down
Thu Jul  3 10:45:34 2025 user.notice post-tracking-002-error: OpenVPN down, restart it
Thu Jul  3 10:45:34 2025 daemon.err openvpn(omr)[19195]: event_wait : Interrupted system call (fd=-1,code=4)
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[19195]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[19195]: net_addr_v4_del: 10.255.252.2 dev tun0
Thu Jul  3 10:45:34 2025 daemon.notice ttyd[4454]: [2025/07/03 10:45:34:3465] N: rops_handle_POLLIN_netlink: DELADDR
Thu Jul  3 10:45:34 2025 daemon.notice ttyd[4454]: [2025/07/03 10:45:34:3501] N: rops_handle_POLLIN_netlink: DELADDR
Thu Jul  3 10:45:34 2025 user.notice NET: hotplug (iface): action='remove' interface='tun0'
Thu Jul  3 10:45:34 2025 daemon.info ModemManager[6379]: hotplug: remove network interface tun0: event processed
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[19195]: /usr/libexec/openvpn-hotplug down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[19195]: SIGTERM[hard,] received, process exiting
Thu Jul  3 10:45:34 2025 user.notice post-tracking-002-error: DNS flush
Thu Jul  3 10:45:34 2025 daemon.warn openvpn(omr)[6547]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations.
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[6547]: OpenVPN 2.6.14 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[6547]: library versions: OpenSSL 3.5.0 8 Apr 2025, LZO 2.10
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[6547]: DCO version: N/A
Thu Jul  3 10:45:34 2025 daemon.warn openvpn(omr)[6547]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul  3 10:45:34 2025 daemon.warn openvpn(omr)[6547]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[6547]: TCP/UDP: Preserving recently used remote address: [AF_INET]62.113.114.59:65301
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[6547]: Attempting to establish TCP connection with [AF_INET]62.113.114.59:65301
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[6547]: TCP connection established with [AF_INET]62.113.114.59:65301
Thu Jul  3 10:45:34 2025 daemon.warn openvpn(omr)[6547]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Not supported (errno=95)
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[6547]: TCPv4_CLIENT link local: (not bound)
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[6547]: TCPv4_CLIENT link remote: [AF_INET]62.113.114.59:65301
Thu Jul  3 10:45:34 2025 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu Jul  3 10:45:34 2025 daemon.notice netifd: Network device 'tun0' link is down
Thu Jul  3 10:45:34 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Thu Jul  3 10:45:34 2025 daemon.notice netifd: omrvpn (19618): udhcpc: read error: Network is down, reopening socket
Thu Jul  3 10:45:34 2025 daemon.notice netifd: omrvpn (19618): udhcpc: bind: No such device
Thu Jul  3 10:45:34 2025 daemon.notice netifd: Interface 'omrvpn' is now down
Thu Jul  3 10:45:34 2025 daemon.notice openvpn(omr)[6547]: [server] Peer Connection Initiated with [AF_INET]62.113.114.59:65301
[... cut ...]
Thu Jul  3 10:50:52 2025 daemon.notice openvpn(omr)[1454]: Initialization Sequence Completed
Thu Jul  3 10:50:53 2025 daemon.notice netifd: omrvpn (1725): udhcpc: broadcasting discover
Thu Jul  3 10:51:00 2025 user.notice omr-schedule-010-services: Can't find omr-tracker-ss for Shadowsocks Rust, restart omr-tracker...
Thu Jul  3 10:51:00 2025 user.notice omr-tracker: Launching...
Thu Jul  3 10:51:11 2025 user.notice omr-tracker: Launched
Thu Jul  3 10:51:11 2025 user.notice post-tracking-002-error: DNS flush

dmkuzmin, 2025-07-03:
Fixed the issue of changing the proxy. Instead of the default proxy Shadowsocks Rust 2022, I chose Shadowsocks. I don't know if this is correct or if there are any problems in Shadowsocks, but for now it's a working solution.

[image]

dmkuzmin, 2025-07-03:
I also don't understand how the flow is distributed, since each interface has different speeds and seems to change by a "ladder" of ~ 25 percent, starting from the master. The result is a rather weak aggregation.

[image]

I tried running a speed test through omr-test-speed, and the result is roughly similar to what SpeedTest gives. But I can't test individual WAN interfaces - the speed drops quickly to zero and consistently shows zero.

Ysurac, 2025-07-07:
The USB device is on a powered hub ?

dmkuzmin, 2025-07-18:
Can the following optimizations from the list help?
- reduced MTU
- Suppressing a channel with a bad RTT
- Under **System –> OpenMPTCPRouter –> Advanced Settings** enable **TCP Low Latency**
- Maybe there is something for optimization

closed 2025-10-21 by github-actions[bot]