Ysurac/openmptcprouter #4007: OMR Bypass not working

Pimmal, 2025-07-11:
## Expected Behavior
When typing in Source MAC or IP the traffic needs to be routed outside the Bonding

## Current Behavior
I type in my MAC and my source IP but traffic still gets routed through VPS IP

## Steps to Reproduce the Problem

  1. Type in source Ip into OMR Bypass
  2. Check my IP

## Context (Environment)
I can not watch RTL online or my Gaming results into high ping

## Specifications

  - OpenMPTCProuter version: v0.63-snapshot-6.12 r0+30292-77b9393d2f
  - OpenMPTCProuter VPS version: 0.1032 6.12.15-x64v3-xanmod
  - OpenMPTCProuter VPS provider: Netcup
  - OpenMPTCProuter platform: x86_64
  - Country: Germany

chubx, 2025-07-22:
> know issue, this will be fixed in next snapshot.

@Ysurac hi! can i please ask if where can we download this version as mentioned in this thread: `OpenMPTCProuter version: v0.63-snapshot-6.12 r0+30292-77b9393d2f`
Or can anyone please point me to the right direction? thank you!

Ysurac, 2025-07-22:
Snapshots are in https://snapshots.openmptcprouter.com/

Ysurac, 2026-01-13:
Try latest snapshot: https://snapshots.openmptcprouter.com/