Ysurac/openmptcprouter #4010: No data on the Network overview tab

falconsk9565, 2025-07-15:
[image]
[image]
[syslog.txt](https://github.com/user-attachments/files/21224844/syslog.txt)
[image]

Ysurac, 2025-07-15:
The "no data" can happen when some tests fails. It's a know bugs when multiples tests timeout.

falconsk9565, 2025-07-15:
[image]
Now I get another error message. What can I do ?

Ysurac, 2025-07-15:
Try to log out and login again maybe.
Difficult to help more at it seems to be an old fork...

falconsk9565, 2025-07-15:
[image]

falconsk9565, 2025-07-20:
Are there any other suggestions ? 
I'm disappointed to lose the functionality of the router. 
Or all I ought to do is throw away this useless device?