Ysurac/openmptcprouter #4011: Router not reachable via portforward after OpenVPN down

max93776, 2025-07-17:
## Expected Behavior
Port forward works permanently..

## Current Behavior
After OpenVPN Tunnel is down a new IP (10.255.255.3) is used for the Router part. My monitoring then shows that from this point on neither the ssh port forwarding to the router nor the https port forwarding works. See syslog (before and after OpenVPN renewal there is just some dhcp stuff, I deleted it)

## Specifications

  - OpenMPTCProuter version: v0.61
  - OpenMPTCProuter VPS version: 0.1032 6.12.15-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: 
  - OpenMPTCProuter platform: x86_64

```
...
Jul 17 01:26:13 OpenMPTCProuter daemon.err openvpn(omr)[16901]: read TCPv4_CLIENT []: Network unreachable (fd=5,code=101)
Jul 17 01:26:13 OpenMPTCProuter daemon.err openvpn(omr)[16901]: read TCPv4_CLIENT []: Network unreachable (fd=5,code=101)
Jul 17 01:26:36 OpenMPTCProuter user.notice post-tracking-002-error: omrvpn down because gateway down
Jul 17 01:26:37 OpenMPTCProuter user.notice post-tracking-002-error: OpenVPN down, restart it
Jul 17 01:26:37 OpenMPTCProuter daemon.err openvpn(omr)[16901]: event_wait : Interrupted system call (fd=-1,code=4)
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[16901]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[16901]: net_addr_v4_del: 10.255.252.2 dev tun0
Jul 17 01:26:37 OpenMPTCProuter daemon.notice ttyd[14436]: [2025/07/17 01:26:37:0177] N: rops_handle_POLLIN_netlink: DELADDR
Jul 17 01:26:37 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Jul 17 01:26:37 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Jul 17 01:26:37 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Jul 17 01:26:37 OpenMPTCProuter daemon.notice ttyd[14436]: [2025/07/17 01:26:37:0181] N: rops_handle_POLLIN_netlink: DELADDR
Jul 17 01:26:37 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='tun0'
Jul 17 01:26:37 OpenMPTCProuter daemon.info ModemManager[8406]: hotplug: remove network interface tun0: event processed
Jul 17 01:26:37 OpenMPTCProuter user.notice post-tracking-002-error: Restart omrvpn
Jul 17 01:26:37 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[16901]: /usr/libexec/openvpn-hotplug down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[16901]: SIGTERM[hard,] received, process exiting
Jul 17 01:26:37 OpenMPTCProuter daemon.warn openvpn(omr)[8704]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: OpenVPN 2.6.10 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: library versions: OpenSSL 3.0.15 3 Sep 2024, LZO 2.10
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: DCO version: N/A
Jul 17 01:26:37 OpenMPTCProuter daemon.warn openvpn(omr)[8704]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Jul 17 01:26:37 OpenMPTCProuter daemon.warn openvpn(omr)[8704]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: TCP/UDP: Preserving recently used remote address: [AF_INET]xxxxx
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: Attempting to establish TCP connection with [AF_INET]xxxxxxx
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: TCP connection established with [AF_INET]xxxxxxx
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: TCPv4_CLIENT link local: (not bound)
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: TCPv4_CLIENT link remote: [AF_INET]xxxxx
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: [server] Peer Connection Initiated with [AF_INET]xxxxxxx
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: TUN/TAP device tun0 opened
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: net_iface_mtu_set: mtu 1420 for tun0
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: net_iface_up: set tun0 up
Jul 17 01:26:37 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Jul 17 01:26:37 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: net_addr_v4_add: 10.255.252.3/24 dev tun0
Jul 17 01:26:37 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: /usr/libexec/openvpn-hotplug up omr tun0 1420 0 10.255.252.3 255.255.255.0 init
Jul 17 01:26:37 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='tun0'
Jul 17 01:26:37 OpenMPTCProuter daemon.info ModemManager[8722]: hotplug: add network interface tun0: event processed
Jul 17 01:26:42 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Jul 17 01:26:42 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Jul 17 01:26:42 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Jul 17 01:26:42 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Jul 17 01:26:42 OpenMPTCProuter daemon.notice openvpn(omr)[8704]: Initialization Sequence Completed
Jul 17 01:26:43 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 17 01:26:44 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 17 01:26:44 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 17 01:26:48 OpenMPTCProuter user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was 192.168.11.2)
Jul 17 01:26:50 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 17 01:26:50 OpenMPTCProuter user.notice post-tracking-020-status: New public ip detected for omrvpn (tun0): 10.255.252.3 (previous: 10.255.252.2)
Jul 17 01:26:50 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 17 01:27:00 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Jul 17 01:27:00 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Jul 17 01:27:03 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 17 01:27:03 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Jul 17 01:27:08 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
... 
```

I changed the vpn from OpenVPN to Glorytun TCP and see if this will work better the next days. 

[image]

Also VPS logs show no abnormalities with `journalctl -u openvpn@tun0` and `journalctl -u omr-admin`.

Ysurac, 2025-07-17:
What do you have on VPS in ```/etc/shorewall/params.vpn``` when it's not working ?

max93776, 2025-07-17:
Changed to OpenVPN TCP again:

- from VPS 10.255.252.2 is pingable 
- portforwardings again not working

```
# cat /etc/shorewall/params.vpn
VPS_ADDR=10.255.252.1
OMR_ADDR=10.255.255.2
OMR_ADDR=10.255.255.2
VPS_IFACE=tun0
OMR_ADDR=10.255.255.2
```

closed 2025-07-18 by max93776