Ysurac/openmptcprouter #4016: Ping drops when primary WAN interface fails, despite MPTCP aggregation

marioroox, 2025-07-27:
I would like to understand if there is a specific configuration required to avoid the issue I’m experiencing, which is described below.

When aggregating two WAN interfaces using MPTCP and Shadowsocks, the system is expected to maintain continuous connectivity (with no ping loss), even if one of the interfaces temporarily goes down. Since bandwidth is aggregated, the transition between WAN interfaces should be seamless and traffic should not be interrupted.

In my setup, I have two WAN interfaces aggregated through Shadowsocks, with MPTCP active. One WAN interface is set as the master, and the second one is enabled as a secondary path.

The connection works as expected under normal conditions, and MPTCP is active. However, I encounter an issue when the primary (master) WAN interface, which was initially used to establish the tunnel, goes down for a few seconds. During this time, the system experiences packet loss (ping fails). Once the traffic switches over to the secondary WAN, ping resumes normally — even if the primary WAN later comes back online.

However, if I subsequently disconnect the secondary WAN interface, the ping drops again, even though the primary WAN is back up.

It seems that OpenMPTCProuter relies too heavily on the network interface that was active at the time of tunnel startup. I would expect MPTCP to provide seamless failover without ping interruptions, but that’s not the behavior I'm observing.

To clarify: ping fails only when the physical WAN interface used during tunnel initiation goes down, even briefly, despite having a secondary WAN path available and configured.

Ysurac, 2025-07-27:
Please use bug template with versions.
I would also need the result of ```ip r``` via SSH on the router.
But yes it's not ideal on the VPN part, it's better on TCP part that is using the proxy. If you have something with enough CPU and memory you can try XRay and enable UDP over Proxy.

marioroox, 2025-07-27:
Thanks Ysurac, I’ll provide you with all the required information tomorrow morning.

Ysurac, 2025-07-28:
It's an update in this case ?
It's in System->OpenMPTCProuter, "Advanced setting" tab at the bottom of the page.

Ysurac, 2025-07-28:
Seems that it was already removed in 0.62. What do you have in ```uci show openmptcprouter.settings``` ?

Ysurac, 2025-07-28:
This is a problem "openmptcprouter.settings.master='static'".
Change it via: ```uci set openmptcprouter.settings.master='balancing' && uci commit openmptcprouter``` and reboot.

marioroox, 2025-07-28:
Hi, 
as you can see in the screenshot, the situation hasn't improved. When I disconnect one of the Ethernet connections, the ping drops—even though I have load balancing configured.

[image]

marioroox, 2025-07-29:
Good morning Ysurac, 
do you happen to have any suggestions on how to solve this?

ccmks, 2025-08-04:
> This is a problem "openmptcprouter.settings.master='static'". Change it via: `uci set openmptcprouter.settings.master='balancing' && uci commit openmptcprouter` and reboot.

Hi, I also realized that newly installation, the master selection is set to static, I had to run the command you gave here in order to make it balancing....that's why I was wondering that if the master is down, nothing works. This is the culprit....

marioroox, 2025-08-04:
Example:

[image]

marioroox, 2025-08-04:
> Hi, I also realized that newly installation, the master selection is set to static, I had to run the command you gave here in order to make it balancing....that's why I was wondering that if the master was down, nothing worked. This is the culprit....

Hi @ccmks 
by any chance, did you manage to solve the problem I'm having?
If so, what settings did you configure or enable in addition to the balancing mode?

ccmks, 2025-08-06:
> Hi [@ccmks](https://github.com/ccmks) by any chance, did you manage to solve the problem I'm having? If so, what settings did you configure or enable in addition to the balancing mode?

No, I didn't know how to configure the OMR Tracker, I just leave it as default settings.  Everything else is default settings aside the master balance mode.

chubx, 2025-08-27:
hi all, I've landed on this thread when looking up for solution to my similar issue - almost the same actually.

My platform is ZBT Z8102AX, i've got two of this. one with LTE/5G module, one without.

Same issue, ping drops when the primary or master MPTCP enabled connection disconnects.

My Firmware:
``` v0.62-6.6 r0+28431-92e020b50f```