Ysurac/openmptcprouter #4019: SFP Modules not working and self-assigning IP's

tuskydev, 2025-07-30:
## Expected Behavior
Plugging in two well known SFP modules and getting the same result.
https://store.ui.com/us/en/category/accessories-modules-fiber/collections/accessories-pro-direct-attach-cables/products/uacc-cm-rj45-mg?variant=uacc-cm-rj45-mg
https://www.fs.com/products/176316.html

## Current Behavior
Self assigned IP's make it impossible to use SFP ports on a Banana Pi R4

## Specifications

  - OpenMPTCProuter version: SD image: [openmptcprouter-v0.62-6.6-r0+28431-92e020b50f-mediatek-filogic-bananapi_bpi-r4-sdcard.img.gz](https://releases.openmptcprouter.com/v0.62-6.6/bpi-r4/targets/mediatek/filogic/openmptcprouter-v0.62-6.6-r0+28431-92e020b50f-mediatek-filogic-bananapi_bpi-r4-sdcard.img.gz) (2025/03/13 - 70.30Mo - [sha256sum](https://releases.openmptcprouter.com/v0.62-6.6/bpi-r4/targets/mediatek/filogic/openmptcprouter-v0.62-6.6-r0+28431-92e020b50f-mediatek-filogic-bananapi_bpi-r4-sdcard.img.gz.sha256sum))
  - OpenMPTCProuter VPS version: NULL
  - OpenMPTCProuter VPS provider: NULL
  - OpenMPTCProuter platform: BPiR4

Ysurac, 2025-07-30:
Did you try latest snapshot on kernel 6.12 ? https://snapshots.openmptcprouter.com/6.12/bpi-r4/targets/mediatek/filogic/
What do you mean with self assigned IP ?

Ysurac, 2025-07-31:
It's an APIPA address (IPv4LL), when address is set as DHCP and there is no DHCP server answer.
Did you try to set it as static on client ?

Only 10G modules seems to work: https://wiki.banana-pi.org/Getting_Started_with_BPI-R4#1.EF.BC.8910G_SFP_Module

tuskydev, 2025-08-01:
I made a post about this issue here on OpenWRT, but I would still love to know if this is a software (OpenWRT/OMR) or hardware (BPiR4) issue forum.openwrt.org/t/need-a-sanity-check-on-bpi-r4-sfp-cages-both-ports-link-but-pull-a-169-254-x-x-self-assigned-every-time/238209

tuskydev, 2025-08-02:
The post I made in OpenWRT was recommended to come back here due to the nature of it not being actual OpenWRT