Ysurac/openmptcprouter #4020: IPv6 with DHCPv6

ravor-org, 2025-08-01:
## Expected Behavior
I would like to leverage both IPv4 and IPv6 routing via OpenMPTCP. At the moment, I have 2 routers (5G and DSL), which both lease their IP via DHCP (DHCPv6 and DHCP). IPv4 is working now, but I'm unsure how to setup the IPv6 path. The VPS does have a /64 and a IPv6 static address assigned. My both ISPs provide full Dual-Stack. How do I get IPv6 and IPv4 running? There's one additional thing to consider: the main routing component is a opnsense, which has a dedicated line connected to the OpenMPTCP router. IPv6 is already activated.

So the question would be:
- how to configure dual stack for the WAN interfaces?
- how to setup the LAN interface to have a IPv6 which can be used between Opnsense and OpenMPTCP?

## Current Behavior
IPv6 is enabled, but neither on WANs nor on the LAN interface there's IPv6 connectivity.

## Specifications

  - OpenMPTCProuter version: v0.62
  - OpenMPTCProuter VPS version: v0.62 (Debian 12, Kernel 6.x)
  - OpenMPTCProuter VPS provider: netcup
  - OpenMPTCProuter platform: x86

ravor-org, 2025-11-17:
This is not stale. Still waiting for questions or a fix.

Ysurac, 2026-01-05:
You tested latest snapshot ?