Ysurac/openmptcprouter #4022: Installation on ZBT Z8109AX V02 128MB board

pbapba512, 2025-08-02:
## Expected Behavior
I tried to install the latest Openmptcp on this board.

## Current Behavior
The latest version was downloaded (as I see well, only sysupgrade.bin version is available). I tried to flash as a new firmware image, but I received "The uploaded image file does not contain a supported format." 

The question is: should I click the force update button, or my device will be bricked?

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.62-6.6-r0+28431-92e020b50f-mediatek-filogic-z8109ax-128m-squashfs-sysupgrade.bin
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: ZBT Z8109AX V02 128MB

Ysurac, 2025-08-02:
Yes. In any case with UBoot and the reset button you can flash and try again.

pbapba512, 2025-08-03:
Thank you, done with success!