Ysurac/openmptcprouter #4024: Hard reset via button is not working on 8102AX-128m

dormancygrace, 2025-08-03:
## Expected Behavior
router resets 

## Current Behavior
nothing happens (checked with serial connection)

## Steps to Reproduce the Problem

  1. press reset button

## Specifications

  - OpenMPTCProuter version: 0.62
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 8102AX-128m

Ysurac, 2025-08-04:
I can't control what a hard reset (when you press more than 5s on button at startup) do but in must case you should have uboot webserver on 192.168.1.1
And it's working correctly on latest snapshot (even it's it's not so easy to press the reset button that is not a real button on some box...)

dormancygrace, 2025-08-04:
I can upload video, that there is nothing in serial console, when you press and hold reset button. Can you point to the correct procedure for resetting with button?

Ysurac, 2025-08-04:
Factory reset from OpenMPTCProuter or from ZBT ?
In case of factory reset from OMR, this should work in snapshot (I really don't use that, I put a new image without keeping settings), for ZBT yes it's via the webserver and upload another image (ZBT image if you have it)

closed 2025-08-05 by dormancygrace