Ysurac/openmptcprouter #4030: pcie modem interface on Quectel RM520N-GL never receives packets, RX: 0.

tiborkiss, 2025-08-07:
Last time I fought with a Lenovo edition of RM520N-GL modem, which was instable and Lenovo locked us out of firmware update possibilities. I had to learn the hard way that that modem is faked, to not be usable as Quectel RM520N-GL modem.

Therefore I ordered a genuine Quectel RM520N-GL RM520NGLAA-M20-SGASA modem.
That had a somewhat older firmware, therefore I updated the firmware to the latest version 
RM520NGLAAR01A08M4G_01.206.01.206 .

## Expected Behavior
For 5G we must use PCIE interface, which presents a `mhi_hwip0` interface .  
There is also an USB modem, which presents a `ppp0` interface.

Both modems should work, especially the PCIE modem.

## Current Behavior
To correctly detect the two modems, indifferent of power of or cold reboot, I still have to 
```
killall ModemManager
/etc/init.d/modemmanager start
```
then I see both the pcie and usb modems. 

I select the pcie modem and the mhi_hwip0 get correct IP, DNS, Gateway and the bearer shows proper connection.
The TX packet counter increasing, but RX counter always stays at 0 and also ping 8.8.8.8 doesn't work.

Is not a firewall filtering issue, because with the same setup, just switch to usb modem, everything works, but slower and only LTE mode.

## Possible Solution

I can still use in USB mode, but after researching the forums, I saw that some peoples experiencing similar issues, resolved by applying some driver patches from Quectel and also compiled a Quectel Modem Manager.

First of all, I would like to hear if anybody else had similar experience and have a hint, or solution.

Second, I can find some guide or hints how to build the Quectel drivers, which one to choose, from where and of course, the Quectel modem manager. I did not built yet the OMR, definitively I need some help.

Since this modem is a well regarded compatible modem, definitively would be nice to apply that patch and shared with the community.

## Steps to Reproduce the Problem

  1. Use the open source version of mhi drivers
  2. Configure pcie mode in AT commands
  3. killall ModemManager && /etc/init.d/modemmanager start  // otherwise only usb modem we see with id 1, leaving the placeholder for pcie modem which has to have id 0.
  4. check TX and RX counter while trying to ping something, also verify possible firewall issue
    RX counter stays at 0 and ping will not work toward internet

## Context (Environment)
I am using on a Banana Pi BPI-R4 with OMR, but essentially without MPTCP features, just like OpenWrt style at this moment, until I sort out these problems.

## Specifications

  - OpenMPTCProuter version: v0.63-rc1
  - OpenMPTCProuter VPS version: not using yet
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: Debian 12
  - Country: CH digitalrepublic

Ysurac, 2025-08-08:
I have this:
```
 ------------------------------------
  Hardware  |            manufacturer: Quectel
            |                   model: RM520N-GL
            |       firmware revision: RM520NGLAAR03A04M4G
            |          carrier config: ROW_Commercial
            | carrier config revision: 0A010809
            |            h/w revision: RM520N-GL
            |               supported: gsm-umts, lte, 5gnr
            |                 current: gsm-umts, lte, 5gnr
            |            equipment id: 868371053175192
  ------------------------------------
  System    |                  device: /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1
            |                 physdev: /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1
            |                 drivers: option1, cdc_mbim
            |                  plugin: quectel
            |            primary port: cdc-wdm0
            |                   ports: cdc-wdm0 (mbim), ttyUSB0 (ignored), ttyUSB1 (gps), 
            |                          wwan0 (net)
  ------------------------------------
  Status    |                    lock: sim-pin2
            |          unlock retries: sim-pin2 (3)
            |                   state: connected
            |             power state: on
            |             access tech: lte
            |          signal quality: 38% (cached)
```
AT commands are working via mmcli
/sys/devices... is set as "device" on the interface with protocol ModemManager. Nothing special to do.

tiborkiss, 2025-08-08:
Very interesting!
Mine is this:

`
root@OpenMPTCProuter:~# mmcli -L
    /org/freedesktop/ModemManager1/Modem/1 [Quectel] RM520N-GL
    /org/freedesktop/ModemManager1/Modem/0 [generic] RM520N-GL

root@OpenMPTCProuter:~# mmcli -m 0
  ------------------------------------
  General   |                    path: /org/freedesktop/ModemManager1/Modem/0
            |               device id: 71ab4bdbdfc559c4a5fcb08023d60f5e75b31c71
  ------------------------------------
  Hardware  |            manufacturer: generic
            |                   model: RM520N-GL
            |       firmware revision: RM520NGLAAR01A08M4G
            |          carrier config: ROW_Commercial
            | carrier config revision: 0A010809
            |            h/w revision: RM520N-GL
            |               supported: gsm-umts, lte, 5gnr
            |                 current: gsm-umts, lte, 5gnr
            |            equipment id: 868371054956004
  ------------------------------------
  System    |                  device: /sys/devices/platform/soc/11280000.pcie/pci0003:00/0003:00:00.0/0003:01:00.0
            |                 physdev: /sys/devices/platform/soc/11280000.pcie/pci0003:00/0003:00:00.0/0003:01:00.0
            |                 drivers: mhi_net, mhi-pci-generic
            |                  plugin: generic
            |            primary port: wwan0mbim0
            |                   ports: mhi_hwip0 (net), mhi_swip0 (net), wwan0mbim0 (mbim),
            |                          wwan0qcdm0 (ignored)
  ------------------------------------
  Status    |                    lock: sim-pin2
            |          unlock retries: sim-pin2 (3)
            |                   state: connected
            |             power state: on
            |             access tech: lte, 5gnr
            |          signal quality: 46% (cached)
  ------------------------------------

root@OpenMPTCProuter:~# mmcli -m 1
  ----------------------------------
  General  |                   path: /org/freedesktop/ModemManager1/Modem/1
           |              device id: f237e9976a8e94b2c83188c237edd49601cff604
  ----------------------------------
  Hardware |           manufacturer: Quectel
           |                  model: RM520N-GL
           |      firmware revision: RM520NGLAAR01A08M4G
           |              supported: gsm-umts, lte
           |                current: gsm-umts, lte
           |           equipment id: 868371054956004
  ----------------------------------
  System   |                 device: /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1
           |                physdev: /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1
           |                drivers: option1
           |                 plugin: quectel
           |           primary port: ttyUSB2
           |                  ports: ttyUSB0 (ignored), ttyUSB1 (gps), ttyUSB2 (at)
  ----------------------------------
  Status   |         unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                  state: connected
           |            power state: on
           |            access tech: lte
           |         signal quality: 100% (cached)

root@OpenMPTCProuter:~# mmcli -m 1 --command="AT"
error: command failed: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Failed: Failed: No AT port available to run command'

root@OpenMPTCProuter:~# mmcli -m 0 --command="AT"
error: command failed: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Failed: Failed: No AT port available to run command'
`
The fact that AT command now doesn't work, that is really stupid. For a while at the beginning I was able to use mmcli and also picocom.

I have two modems, the usb modem doesn't have `cdc-wdm0 (mbim)` port, instead on the pcie modem I have `wwan0mbim0 (mbim)`. 
But that is not listed in `primary ports`, like for you. I just have `ttyUSB2` in my primary port for USB modem.

Moreover, if you notice, my usb modem doesn't have 5g, but pcie modem.

My firmware version is the latest from R01 series. If you check this firmware is really new. I flashed this because my searches indicated that in Europe, this version of the firmware was the rock solid.

Probably now is the time to play with downgrades, if I want 5G on usb modem.

Otherwise, we can expect that in the future more peoples will install this latest version of modem which I currently have. I am still researching the forums, but what I found until now is that some peoples just had to patch and use the Quectel version of the drivers etc. and as a result, they really get the best performances. Some reported 700Mbps. Here in my situation, I am also close to a 5G antenna, definitively the pcie solution it would be the desired one.

I am really undecided yet in which direction to go.. to begin downgrading for usb which does 5g, or better, learn to compile the firmware and spend time with patching and official drivers.

franktfye, 2026-08-16:
I got this fixed on my laptop (Arch Linux) — no kernel patch, no out-of-tree driver. Debugged it with Claude Fable 5, which traced it to the modem dialing the PDN internally rather than any driver bug. Full write-up: https://claude.ai/code/artifact/503fe818-4ed4-4f39-a3d1-2992a675500c

Worth a 30-second check before any driver patching:

I hit the exact same symptom cluster with an RM520N-GL on PCIe/MHI (different platform: ThinkPad P16s Gen 4, Arch Linux, kernel 7.1.8, stock in-tree `mhi-pci-generic` + `mhi_wwan_mbim`, ModemManager 1.24): registered, real carrier IP on the interface, TX counting up, RX permanently 0, 100% ping loss. I also went down the suspected-driver-bug road first — Quectel's out-of-tree `quectel_pcie_mhi` driver, patched `quectel-CM` — and none of it helped, because the driver was never the problem.

**Root cause (in my case):** the modem's internal **QMAP auto-connect** was enabled, so the modem dials the PDN *itself* at radio-up and binds the downlink to its embedded client. Every host-side dial is then rejected — QMI `START_NETWORK` fails with verbose call-end reason **241**, which libqmi decodes as *"interface in use, config match"* (i.e. the modem saying it already holds this exact connection), and MBIM connect fails with `ContextNotActivated`. Host TX goes out, but downlink never binds to the host interface → RX 0.

**Check (on the AT port, e.g. `/dev/wwan0at0` / `/dev/mhi_DUN`, with ModemManager/connection scripts stopped):**

```
AT+QMAP="auto_connect"
+QMAP: "auto_connect",0,1      <- instance 0 = 1 means this is your problem

AT+QMAP="MPDN_status"
+QMAP: "MPDN_status",0,1,0,1   <- internal call is up
```

**Fix (one-time — persists in modem NV, survives reboots):**

```
AT+QMAP="auto_connect",0,0
AT+QMAP="connect",0,0
AT+QMAP="MPDN_rule",0
AT+CGACT=0,1
AT+CGPADDR=1                   <- should now show 0.0.0.0 and STAY down
```

Then let the host own the call normally (ModemManager/MBIM in my case). After this, the stock in-tree drivers pass traffic in both directions — verified with `ping -I wwan0`, ~200 MB downlink in the first session.

A telltale sign you can check in your logs: if `AT+CGACT=0,1` returns OK but the context is instantly active again on the next query, that's the internal auto-connect redialing.

Caveats: my firmware is `RM520NGLAAR03A03M4G` (issue reports R01A08), and I'm on ModemManager/NetworkManager rather than OpenMPTCProuter's stack, so I can't promise it's the same cause on the BPI-R4 — but the check costs two AT commands and rules it in or out definitively.