Ysurac/openmptcprouter #4031: Redundant Scheduler not working

ccmks, 2025-08-08:
## Expected Behavior
Get the multipath TCP Scheduler to work as "mptcp_bpf_red.o" working. My goal is to send all packet for all available WAN in order to achieve highest resilience for real-time UDP traffic, such as VoIP, game, and video conference.

## Current Behavior
When I selected "mptcp_bpf_red.o" and select use UDP for V2RAY/XRAY, I ran VoIP test call and start to disconnect master WAN, the call dropped instead of continue working as expected.

## Specifications

  - OpenMPTCProuter version: 0.62-6.6
  - OpenMPTCProuter VPS version: 0.1032 6.12.15-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: BiznetGIO Cloud
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-08-08:
It's correctly sending data over all connections or not ?
You can check in Network->MPTCP, tab "Bandwidth"

ccmks, 2025-08-08:
No, only master WAN so far. the other one just idle....

ccmks, 2025-08-13:
Hi, any news or update so far? Or is it a bug?

ccmks, 2025-08-16:
I ended up reverting back to v0.61 for redundant scheduler....I think there is bug on v0.62

vempire-ghost, 2025-08-17:
> I ended up reverting back to v0.61 for redundant scheduler....I think there is bug on v0.62

Did you manage to get the redundant scheduler working with the kernel greater than 5.4?