Ysurac/openmptcprouter #4033: VPS Server

Yessine1200, 2025-08-08:
## Expected Behavior
OMR can fetch keys from VPS.

## Current Behavior
Hello,

I have OMR installed on a ZBT router.
Regarding the VPS — before installing the script, I had 4 interfaces:
- Management interface
- Public IN interface
- Output interface
- Private IN interface

Before installing the script, I could access my VPS via SSH using the management interface. However, after running the script, I lost SSH access, I can’t ping my management gateway, and OMR is unable to fetch keys from the VPS.

My VPS has two public IP addresses.

[image]

Below is my network architecture.

Any help resolving this issue would be greatly appreciated.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.62-6.6 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: ZBT Z8102AX 128M
--
 

<br class="Apple-interchange-newline">

Yessine1200, 2025-08-08:
Yes, I used port 65222 but not working.(I can't ping the gateway)
I checked /etc/shorewall/params.net, I found `NET_IFACE=ens21`, so it use the correct interface.

I have to add rules related to my config in Shorewall rules ?

Ysurac, 2025-08-08:
Try to disable Shorewall and check if it's working again.

Yessine1200, 2025-08-08:
I disabled Shorewall, now I can ping the gateway but not the interface, it's really weird this behaviour.

Ysurac, 2025-08-19:
```ip r``` ony show routes, not subflows.
If everything is green in System->OpenMPTCProuter, "Status" tab, check with ```ip mptcp endpoint``` to check that endpoint are correct.

Yessine1200, 2025-08-19:
Yes, the subflows look ok. 
The issue might be on the server side. My VPS has two interfaces not one(one for the private link and one for the public link). For example, if I want OMR to reach the VPS over the public connection, I enter the public IP in OpenMPTCProuter → Settings Wizard. If I want it to reach the VPS over the private connection, I enter the other interface’s IP.

Ysurac, 2025-08-19:
Yes. What is the result of ```ip mptcp endpoint``` on the VPS ?

Ysurac, 2025-08-19:
Only IPs defined as signal endpoint are working. I think some changes would be needed in omr-service script at least and also some on router side...

Yessine1200, 2025-08-19:
on router side, it's ok I think, the result of `ip mptcp endpoint `:
    79.132.158.214 id 1 subflow fullmesh dev wwan1 
    10.200.6.142 id 2 subflow fullmesh dev wwan0

And if I use a single interface, will I not have this problem?

skullface101, 2025-09-18:
Regarding OMR vps, can I have the same vps host OMR ,X-UI for XRay proxies, some wireguard/openvpn VPNs and a control panel like webmin?