Ysurac/openmptcprouter #4035: Different versions of XRAY on vps and router

Slikpro, 2025-08-08:
Good afternoon! Your product has become my main hobby for the last six months. I discovered that different versions of XRAY are installed on the VPS and router.
VPS  - Xray 24.11.5  Router - Xray 25.6.8 Is it possible to update the Xray module to the latest version so that I don't have to restore and rewrite all the configuration files?
Thank you!

[image]
[image]

Ysurac, 2025-08-08:
Update VPS with snapshot script: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

Ysurac, 2025-08-11:
Quic was never enabled on XRay in the config for OpenMPTCProuter. Sing-box is also not supported for now.

zerojj-cyber, 2025-09-18:
We are really looking forward to proper QUIC support through XRay UDP in OpenMPTCPRouter. At the moment, without full QUIC support, Apple Music on iPhone cannot play music. On macOS it works (falls back to TCP/HTTP2), but iOS strictly requires QUIC/HTTP3 with correct ALPN negotiation.

Could you please add or enable full QUIC support in XRay UDP transport in OMR? This feature is very important for Apple services and modern apps.

Thank you!

zerojj-cyber, 2025-09-18:
> You should open a new issue. But you can try to disable the "Block Quic *" firewall rules.

Unfortunately, it does not work( 

I’ve been troubleshooting Apple Music playback on iPhone when using OpenMPTCPRouter. Here’s what I found:
	•	With Shadowsocks-go + Glorytun (UDP/TCP) Apple Music works fine on the phone.
	•	But with V2Ray or XRay in VLESS / VLESS Reality mode, Apple Music on iPhone fails with a TLS error. On macOS it still works, but iOS seems much stricter.

From my testing, I am almost certain that the root cause is related to missing HTTP/3 + QUIC support in XRay on the server side. iOS Apple Music requires QUIC/ALPN negotiation (h3), and without it the TLS handshake fails.