Ysurac/openmptcprouter #4039: Single cpu core getting 100% load

shabzz98, 2025-08-12:
## Expected Behavior

Higher bandwidth with a bonded connection

## Current Behavior

I have reinstalled OMR 3 time now in hopes that it will get rid of the issue but i keep running into the same thing over and over.

I have two wan one is a 5g connection and the other is 4g lte connection. The 5g alone delivers 400mbps on speedtest alone but with the bonded setup i am getting 150mbps.

Htop on vps shows healthy cpu utilisation max 30-40%. But it is a single core on my vm that is hitting 100% during speedtest (only during speedtest not while it is idle), i have tried upto 5 cores at this point but i always keep getting a single core hitting 100%.

I have tried both chacha20 and aes encrytion with all different vpn options but nothing seems to work for me.

## Possible Solution

- Encryption change
- Hardware / resource boost
- manual load distribution

## Steps to Reproduce the Problem

Simply setting up the openmptcp solution 

## Context (Environment)

I have the latest release of omr running inside vmware fusion on an intel mac mini late 2012 and a vultur vps with 2 cores on the other end. The VM has 4 cpu cores and 786mb ram with 8gb storage assigned.

## Specifications

  - OpenMPTCProuter version: v0.62-6.6
  - OpenMPTCProuter VPS version: unsure
  - OpenMPTCProuter VPS provider: Vultur
  - OpenMPTCProuter platform: x86_64
  - Country: Australia

Ysurac, 2025-08-12:
Can you take a screenshot of System->OpenMPTCProuter, "Status" tab ?
Can you test speed using ```omr-test-speed``` via SSH on the router and put result here ?

shabzz98, 2025-08-12:
[image]

[image]

The speedtest stopped at 1 minutesand the max value was 20.1M under speed

shabzz98, 2025-08-12:
WAN1:

[image]

WAN2:

[image]

I was getting upto 1mpbs last night seems slower today. still now what the combined bandwidth should be.

Here is what i see with htop (it was the same when i had 5 cores on the VM too, this is a fresh install i did last night)

[image]

shabzz98, 2025-08-12:
[image]

Here is the shot from VPS and this speed is not good. Even though the speed is slow right now the Telstra WAN standalone is still reporting 250mbps+ on ookla. I just ran a speedtest on my phone with the non bonded connection.

Ysurac, 2025-08-13:
You should try to make tests always from same server, for Telstra it's using Telstra server, so it's in same network and this explain why it's fast.
You can also use, via SSH on the router, ```omr-iperf``` to try speed on VPS iperf3 server (```omr-iperf -R --bind-dev wan1```, ```omr-iperf -R --bind-dev wan2``` and ```omr-iperf -R``` for aggregated download speed)

Ysurac, 2025-08-15:
Nothing to worry. 
Load average is a quite good indicator : You have 2 cores on your CPU so should be always under 2.