Ysurac/openmptcprouter #4040: ERR_NETWORK_CHANGED - I keep on getting this on my pc connected to OMR #4027

chubx, 2025-08-13:
## Expected Behavior

Smooth and reliable internet

## Current Behavior

In Chrome, I kept on getting `ERR_NETWORK_CHANGED `
## Possible Solution

## Steps to Reproduce the Problem

It's random

## Context (Environment)

As stated in:
https://github.com/Ysurac/openmptcprouter/discussions/4027#discussioncomment-14005910

I kept on getting random error, using Chrome.

It's the browser I often use, and it is on which I always encounter the issue.

## Specifications

  - OpenMPTCProuter version:  v0.62-6.6 r0+28431-92e020b50f
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1031
  - OpenMPTCProuter VPS provider: Lightnode
  - OpenMPTCProuter platform:  x86_64
  - Country:  Philippines

## Logs After The Error : 20 Secs before/after the problem

```
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: using nameserver 8.8.8.8#53
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Wed Aug 13 14:33:41 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Wed Aug 13 14:34:02 2025 daemon.warn dnsmasq[2]: possible DNS-rebind attack detected: 192-168-111-2.<redacted-this-is-a-server-on-my-network>
Wed Aug 13 14:34:12 2025 daemon.warn dnsmasq[3]: possible DNS-rebind attack detected: 192-168-111-2.<redacted-this-is-a-server-on-my-network>
Wed Aug 13 14:34:37 2025 daemon.warn dnsmasq[4]: possible DNS-rebind attack detected: 192-168-111-2.<redacted-this-is-a-server-on-my-network>
Wed Aug 13 14:35:23 2025 daemon.warn dnsmasq[5]: possible DNS-rebind attack detected: 192-168-111-2.<redacted-this-is-a-server-on-my-network>
Wed Aug 13 14:35:53 2025 daemon.warn dnsmasq[6]: possible DNS-rebind attack detected: 192-168-111-2.<redacted-this-is-a-server-on-my-network>
Wed Aug 13 14:36:53 2025 daemon.warn dnsmasq[7]: possible DNS-rebind attack detected: 192-168-111-2.<redacted-this-is-a-server-on-my-network>
```

Ysurac, 2025-08-13:
I think it's due to DNS-over-HTTPS.
Select the three-dot menu in your browser > Settings.
Select Privacy and security > Security.
Scroll down and Disable "Use secure DNS"

chubx, 2025-08-13:
> I think it's due to DNS-over-HTTPS. Select the three-dot menu in your browser > Settings. Select Privacy and security > Security. Scroll down and Disable "Use secure DNS"

will do, meanwhile, i've updated my post above, i've added logs.

chubx, 2025-08-13:
> I think it's due to DNS-over-HTTPS. Select the three-dot menu in your browser > Settings. Select Privacy and security > Security. Scroll down and Disable "Use secure DNS"

This is done on my main chrome profile instance. however, I have other chrome profile and instances running, and I noticed that the other chrome instance is already using "Use Secure DNS" - disabled. And yet, I got the same ERR_NETWORK_CHANGED error.

Will continue to test and observe.

thank you!

chubx, 2025-08-19:
@Ysurac i'm still having issues even with the **Use Secure DNS** in Chrome disabled.

chubx, 2025-08-19:
> I think it's due to DNS-over-HTTPS. Select the three-dot menu in your browser > Settings. Select Privacy and security > Security. Scroll down and Disable "Use secure DNS"

@Ysurac , so it kept on happening, and i got some new logs, please see below. I've redacted my IP addresses. Also, I've some notes below.

```
Tue Aug 19 22:11:56 2025 user.notice post-tracking-020-status: Check API configuration... Done
Tue Aug 19 23:04:41 2025 daemon.info unbound: [26637:0] info: generate keytag query _ta-4f66-9728. NULL IN
Tue Aug 19 23:22:44 2025 user.notice post-tracking-020-status: Check API configuration...
Tue Aug 19 23:22:44 2025 user.notice post-tracking-020-status: Check API configuration... Done
Wed Aug 20 00:00:00 2025 user.info vnstat_backup: backing up database
Wed Aug 20 00:26:10 2025 user.notice post-tracking-020-status: Check API configuration...
Wed Aug 20 00:26:10 2025 user.notice post-tracking-020-status: Check API configuration... Done
Wed Aug 20 02:00:00 2025 user.info vnstat_backup: backing up database
Wed Aug 20 02:09:50 2025 user.notice post-tracking-020-status: Check API configuration...
Wed Aug 20 02:09:50 2025 user.notice post-tracking-020-status: Check API configuration... Done
Wed Aug 20 02:22:17 2025 user.notice post-tracking-002-error: omrvpn down because IPv4 gateway down
Wed Aug 20 02:22:17 2025 user.notice post-tracking-002-error: Default gateway disabled, no default gateway set
Wed Aug 20 02:22:17 2025 user.notice post-tracking-002-error: DSVPN down, restart it
Wed Aug 20 02:22:17 2025 daemon.info mptcpize[7339]: Done.
Wed Aug 20 02:22:17 2025 daemon.notice ttyd[23092]: [2025/08/20 02:22:17:2210] N: rops_handle_POLLIN_netlink: DELADDR
Wed Aug 20 02:22:17 2025 daemon.notice ttyd[23092]: [2025/08/20 02:22:17:2210] N: rops_handle_POLLIN_netlink: DELADDR
Wed Aug 20 02:22:17 2025 daemon.notice ttyd[23092]: [2025/08/20 02:22:17:2215] N: rops_handle_POLLIN_netlink: DELADDR
Wed Aug 20 02:22:17 2025 user.notice NET: hotplug (iface): action='remove' interface='tun0'
Wed Aug 20 02:22:17 2025 daemon.info ModemManager[21112]: hotplug: remove network interface tun0: event processed
Wed Aug 20 02:22:17 2025 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Aug 20 02:22:17 2025 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Aug 20 02:22:17 2025 daemon.notice netifd: Network device 'tun0' link is down
Wed Aug 20 02:22:17 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Aug 20 02:22:19 2025 daemon.info dsvpn: starting dsvpn vpn instance vpn
Wed Aug 20 02:22:19 2025 daemon.err mptcpize[22562]: Command [ip route show default 2>/dev/null|awk '/default/{print $5}'] failed]
Wed Aug 20 02:22:19 2025 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Aug 20 02:22:19 2025 daemon.notice netifd: Network device 'tun0' link is up
Wed Aug 20 02:22:19 2025 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Aug 20 02:22:19 2025 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Aug 20 02:22:19 2025 daemon.info mptcpize[22562]: Interface: [tun0]
Wed Aug 20 02:22:19 2025 daemon.info mptcpize[22562]: Trying to reconnect
Wed Aug 20 02:22:19 2025 daemon.info mptcpize[22562]: Connecting to <redacted.vps.ip.address>:65401...
Wed Aug 20 02:22:19 2025 daemon.notice netifd: Interface 'omrvpn' is now up
Wed Aug 20 02:22:19 2025 user.notice NET: hotplug (iface): action='add' interface='tun0'
Wed Aug 20 02:22:19 2025 daemon.info mptcpize[22562]: Connected
Wed Aug 20 02:22:19 2025 daemon.info ModemManager[22694]: hotplug: add network interface tun0: event processed
Wed Aug 20 02:22:19 2025 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Wed Aug 20 02:22:19 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Wed Aug 20 02:22:20 2025 user.notice OMR-Tracker: Reloading OMR-Tracker due to ifup of omrvpn (tun0)
Wed Aug 20 02:22:22 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.251.1 dev tun0 (was 192.168.100.1)
Wed Aug 20 03:20:45 2025 user.notice post-tracking-020-status: New public ip detected for wan3 (eth0.7): <redacted.ip.address.new> (previous: <redacted.ip.address.previous>)
Wed Aug 20 03:20:47 2025 user.notice post-tracking-020-status: Check API configuration...
Wed Aug 20 03:20:47 2025 user.notice post-tracking-020-status: Check API configuration... Done
Wed Aug 20 03:31:40 2025 user.notice post-tracking-002-error: wan2 (eth0.6) switched off because check error and ping from 192.168.100.2 error (9.9.9.9,1.0.0.1,114.114.115.115)
Wed Aug 20 03:31:40 2025 user.notice post-tracking-002-error: Delete default route to <redacted.vps.ip.address> dev eth0.6
Wed Aug 20 03:32:05 2025 user.notice post-tracking-003-up: wan2 (eth0.6) switched up
Wed Aug 20 03:32:05 2025 user.notice post-tracking-003-up: Status change, reload MPTCP config for eth0.6
Wed Aug 20 04:00:00 2025 user.info vnstat_backup: backing up database
Wed Aug 20 04:21:21 2025 user.notice post-tracking-020-status: Check API configuration...
Wed Aug 20 04:21:21 2025 user.notice post-tracking-020-status: Check API configuration... Done
Wed Aug 20 05:31:05 2025 user.notice post-tracking-020-status: Check API configuration...
Wed Aug 20 05:31:05 2025 user.notice post-tracking-020-status: Check API configuration... Done
Wed Aug 20 06:00:00 2025 user.info vnstat_backup: backing up database
Wed Aug 20 06:06:44 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: 192-168-111-2.host.myunraid.net
Wed Aug 20 06:06:49 2025 daemon.err uhttpd[6867]: [info] luci: accepted login on /admin/status/logs for root from 192.168.111.111

```

**NOTES**:  I see logs for Public DNS IP Addresses 9.9.9.9,1.0.0.1,114.114.115.115

Ysurac, 2025-08-20:
DSVPN doesn't seems to be really stable here...
For the DNS IP, they are used to check if link is down or not by OMR-Tracker. Maybe a problem if you change DNS servers ?

chubx, 2025-08-20:
> DSVPN doesn't seems to be really stable here... For the DNS IP, they are used to check if link is down or not by OMR-Tracker. Maybe a problem if you change DNS servers ?

I will try your suggestion @Ysurac , however, i'm not sure where to do that, can you please guide me using:
- WebUI
- Configuration Files via SSH

Ysurac, 2025-08-20:
What do you have in Network->DHCP&DNS, "General" tab ?

chubx, 2025-08-29:
@Ysurac i'm still having this issue with my ZBT Z8102AX. See logs below.

```
Fri Aug 29 11:35:23 2025 user.notice post-tracking-002-error: wan3 (eth0.7) switched off because check error and ping from 192.168.1.2 error (114.114.115.115,1.2.4.8,80.67.169.40)
Fri Aug 29 11:35:23 2025 user.notice post-tracking-002-error: Delete default route to <ip addr redacted> dev eth0.7
Fri Aug 29 11:35:47 2025 user.notice post-tracking-003-up: wan3 (eth0.7) switched up
Fri Aug 29 11:35:47 2025 user.notice post-tracking-003-up: Status change, reload MPTCP config for eth0.7
Fri Aug 29 11:59:05 2025 user.notice post-tracking-002-error: wan3 (eth0.7) switched off because check error and ping from 192.168.1.2 error (114.114.115.115,1.2.4.8,80.67.169.40)
Fri Aug 29 11:59:05 2025 user.notice post-tracking-002-error: Delete default route to <ip addr redacted> dev eth0.7
Fri Aug 29 11:59:29 2025 user.notice post-tracking-003-up: wan3 (eth0.7) switched up
Fri Aug 29 11:59:29 2025 user.notice post-tracking-003-up: Status change, reload MPTCP config for eth0.7
```

chubx, 2025-08-29:
> Your connection is not stable. You can change parameters in Services->OMR-Tracker Manager.

@Ysurac , thank you for the response, however - OMR is supposed to help with **unstable connection**, especially in Bonding mode?

Also, i would appreciate if you can guide us using the OMR Tracker. it will benefit us all. Let me know, then I'll share the knowledge to the community

chubx, 2025-09-04:
> OMR aggregate connections, I can do nothing on bad connections. You can try to be more tolerant to connections problems by increasing "Interface down" value as it seems to be quite short cut. I would also need a screenshot of status page to check if you need to increase ping timeout (but default should be ok in most case)

I increased ping time out to 8, for Interface down, and 3 for interface up.

@Ysurac  The test setup is 1 fiber connection + starlink.

Also, i updated my tracking host name or Ip setup to cloudflare and google, for the time being.

Just a quick feedback/question: I thought OMR can stay up or online even when a single is own?

closed 2025-12-09 by github-actions[bot]