Ysurac/openmptcprouter #4044: Whats the best config for max speed, min latency/ping/jitter and even distribution of bandwidth consumption?

skullface101, 2025-08-14:
Whats the best config for max speed, min latency/ping/jitter and even distribution of bandwidth consumption?

Note: Its planned that all lines are of the same type and have same speed, but due to line quality and availability, they are currently not even.

Ysurac, 2025-08-15:
A x86_64 CPU with hardware AES with a good speed by cores (better less cores and higher frequency)

skullface101, 2025-08-16:
I am already using X86_64 with hardware AES, I am asking about the router configuration.

h45h3ncryp73d, 2025-08-30:
I'd also like to know if there would be any benefits from moving OMR from RPi 4b to x32-64 platform ...
During speedtest htop doesn't show any of RPi's 4 cores utilization past 70% so I guess there would actually be no benefit in migrating OMR router (not VPS part) to x32-64 platform. But I might be wrong.

h45h3ncryp73d, 2025-08-30:
Actually, I've been monitoring RPi load a bit closer ... and I'll definitely migrate to x32-64 ...

skullface101, 2025-09-18:
I currently have 4 similar connection and I am getting only the speed of two!
Whats causing this?
and also the bandwidth usage is not distributed

skullface101, 2025-09-18:
`root@OpenMPTCProuter:~# omr-test-speed pppoe-wan2
Select best test server...
Best server is http://appliwave.testdebit.info/10G/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 32.9M    0     0  3460k      0  0:47:02  0:00:09  0:46:53 3590k^C
`
`root@OpenMPTCProuter:~# omr-test-speed pppoe-wan9
Select best test server...
Best server is http://appliwave.testdebit.info/10G/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 49.5M    0     0  3517k      0  0:46:16  0:00:14  0:46:02 3411k^C
`
`root@OpenMPTCProuter:~#` omr-test-speed pppoe-wan7
Select best test server...
Best server is http://appliwave.testdebit.info/10G/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 23.9M    0     0  3525k      0  0:46:10  0:00:06  0:46:04 3623k^C
`
`root@OpenMPTCProuter:~# omr-test-speed eth0
Select best test server...
Best server is http://appliwave.testdebit.info/10G/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 76.7M    0     0  3441k      0  0:47:17  0:00:22  0:46:55 3283k^C
`
`root@OpenMPTCProuter:~# omr-test-speed
Select best test server...
Best server is http://appliwave.testdebit.info/10G/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  3 9536M    3  299M    0     0  9333k      0  0:17:26  0:00:32  0:16:54 9825k^C
`

Ysurac, 2025-09-18:
All connection are using same server, strange. Can you put a screenshot of System->OpenMPTCProuter status ?
For now there is no options to distribute on all connections.
Burst: when a connection is full it use next connection with higher rtt and so on.
Round robin is one after one.
For congestion control, BBR is the better in most case, but you can try.
XRay Vless with UDP enabled is the better with UDP I think, Shadowsocks-Rust give a better TCP aggregation.

skullface101, 2025-09-18:
[image]

Ysurac, 2025-09-18:
Lot of red on screenshot: there is an anonymize checkbox on this page.
Can you try to run omr-test-speed on all interfaces at the same time to check if there is no congestion somewhere

On VPS it's not one of provided kernel. And should be at least a 6.12 kernel.

You need to set the interface the more stable. Master is used for initial connection, it's not a real issue if down?
You can set what you want in Services->OMR-Tracker Manager. There is a "Link quality" checkbox.

VPN is only used for UDP, ICMP,... Proxy is used for TCP (and UDP for *Ray). Fastest TCP proxy is shadowsocks-Rust, for VPN you have to try what is best in your case.

For providers, I don't know. It was data from each website when wiki page was made, maybe it was updated... It's difficult to find a VPS provider that provide a real high bandwidth, maybe easier with a dedicated server.

skullface101, 2025-12-19:
I am running wire guard on another outer behind OMR, the speed becomes lower by 50%, how can I solve this?
Using Glory TUN TCP or UDP improves the speed some times.
Whats the best config?
Using V2RAY/XRAY for UDP slows things down